esphome/issues #2040: RGBCCT Bulb issue

cyber-junkie9, 2021-05-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

1.17.2 

**Affected component:**

RGBWW

**Description of problem:**
Color does not apply autocratically , have to set white brightness value to 0 to make it work
Color brightness also doesn't work
is there any platform available for RGBCCT bulb ?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: smitch_1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "ssid"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "smitch_1 Fallback Hotspot"
    password: "YA50s8A0ekW5"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
    max_power: 100%
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
    max_power: 100%
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
    max_power: 100%
  - platform: esp8266_pwm
    id: output_warm_white
    pin: GPIO13
    max_power: 100%
  - platform: esp8266_pwm
    id: output_cold_white
    pin: GPIO05
    max_power: 100%

# Define a light entity
light:
  - platform: rgbww
    name: smitch
    id: bulb
    red: output_red
    green: output_green
    blue: output_blue
    warm_white: output_warm_white
    cold_white: output_cold_white
    cold_white_color_temperature: 6200 K
    warm_white_color_temperature: 2800 K
    default_transition_length: 3s
    color_interlock: true # avoids simultaneous RGB and W/W

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

michaelpayne02, 2021-06-03:
I'm having the exact same issue. It seems like homeassistant is changing the way that different types of lights work. Instead of the UI looking at the supported features, the whole system is moving forward with color_mode and supported_color_modes. If I check the state of my RGBWW bulb in the developer tools, it is reporting that the bulb is in RGBW mode, when instead it should be in RGBWW mode. 
[image]

This can be changed in customizations but doesn't fix anything. I've tried many cobinations of \color_mode and supported_color_modes but they don't fix the issue. ESPHome needs to update its code to support the new color_mode in the API, Integration, and Discovery platform somehow.

oxan, 2021-07-23:
Could someone try if this is still an issue with v1.20? The white channel changes might've resolved this.

oxan, 2021-09-02:
I assume this is resolved.

closed 2021-09-02 by oxan