esphome/issues #2042: ESP32 + MCP23017 warrnings

krzysiek2788, 2021-05-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSIO 5.13
supervisor-2021.04.3
core-2021.4.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, Board: LILYGO® TTGO T-Call V1.3 V1.4 ESP32
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.17.2 (it was the same in previous versions)
**Affected component:**

https://esphome.io/components/mcp230xx.html

**Description of problem:**
I have connected MCP23017 to ESP32 and configure it all connection are ok, VCC -> VCC, GND-> GND, I2C -> I2C. The module is available and OUTPUT pins are fine, only INPUT pins are making me crazy. In to the INPUT pin I connected momentary button to turn on/off lights in the rooms and it is giving me false positive signals. I added pull-up resistor even changed INPUT to INPUT_PULLUP which didn't work only setup filter + delay 100ms work but below warning still remains:
`[W][i2c:086]: Requesting 1 bytes from 0x20 failed!`

My code is quite big and there are part of code which will be used to read energy meter SDM630 over modbus.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: home_lights_notify
  friendly_name: ttgo_home_lights_notify
  ip_address: !secret home_lights_notify

esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit
#  includes:
#    - custom_components/sdm/SDM_Config_User.h
#    - custom_components/sdm/SDM.h
#    - custom_components/sdm/sdm_esphome.h

wifi:
  ssid: !secret wifi_id
  password: !secret wifi_pwd
  manual_ip:
    static_ip: ${ip_address}
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  #ap:
   # ssid: "Home Lights Notify"
    #password: "YiwdlGqxsCgP"

#captive_portal:

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
  # level: VERY_VERBOSE

# Enable Home Assistant API
api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'
        
[... 311 lines cut ...]
[08:13:04][V][sim800l:044]: S: AT - 1
[08:13:14][W][i2c:086]: Requesting 1 bytes from 0x20 failed!
[08:13:19][V][sim800l:044]: S: AT - 1
[08:13:34][V][sim800l:044]: S: AT - 1
[08:13:39][W][i2c:086]: Requesting 1 bytes from 0x20 failed!
[08:13:39][W][i2c:086]: Requesting 1 bytes from 0x20 failed!
[08:13:39][D][esp32_ble_tracker:148]: Starting scan...
[08:13:39][D][binary_sensor:036]: 'Galaxy Watch': Sending state OFF
[08:13:41][D][esp32_ble_tracker:544]: Found device 8C:79:F5:A3:49:D3 RSSI=-83
[08:13:41][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[08:13:41][D][esp32_ble_tracker:544]: Found device B8:BC:5B:AA:DD:FB RSSI=-90
[08:13:41][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[08:13:47][D][esp32_ble_tracker:544]: Found device 5E:26:BA:B0:0E:34 RSSI=-67
[08:13:48][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[08:13:49][W][i2c:086]: Requesting 1 bytes from 0x20 failed!
[08:13:49][V][sim800l:044]: S: AT - 1
[08:13:54][D][esp32_ble_tracker:544]: Found device 41:BE:11:3E:3C:51 RSSI=-86
[08:13:54][D][esp32_ble_tracker:565]:   Address Type: RANDOM

```

**Additional information and things you've tried:**
I tried different VCC 3,3V and 5V, pullup resistors for I2C (which isn't necessary as MCP module I have, has all necessary resistors pullup
[image]
Now it is connected to 3,3V I don't know why I receive this false positive when I connect those input pins to direct pins on ESP8266 (Wemos D1 mini) I don't have that issue.

ssieb, 2021-05-13:
The `Requesting 1 bytes from 0x20 failed!` message suggests an issue with your i2c wiring somewhere.
How do you have the buttons wired?  What is each side of the button connected to?

krzysiek2788, 2021-05-13:
This is how I connected it:
[image]

andriej, 2021-07-22:
Same here, wire between esp32 and mcp is like 4cm max - warnings all over the place

closed 2021-12-04 by github-actions[bot]