esphome/issues #2044: remote.rc_switch supports IR receiving but not IR transmitting

lanrat, 2021-05-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.17.1 

**Affected component:**

* https://esphome.io/components/remote_transmitter.html
* https://esphome.io/components/remote_receiver.html

**Description of problem:**

I have a few IR remotes that ESPHome's IR decoder decodes as `RCSwitch Raw`.
For example, here are a few of the decodes from my Lasko heater's IR remote:
```
RCSwitch Raw: protocol=4 data='11011000000'
RCSwitch Raw: protocol=4 data='11011000010'
RCSwitch Raw: protocol=4 data='11011000100'
RCSwitch Raw: protocol=4 data='11011000001'
RCSwitch Raw: protocol=4 data='11011001000'
RCSwitch Raw: protocol=4 data='11011010000
```

However, the RCSwitch Raw service `remote_transmitter.transmit_rc_switch_raw` does not support sending IR commands. All calls to `remote_transmitter.transmit_rc_switch_raw` with an IR transmitter appear to do nothing. Additionally the `RCSwitchBase::transmit` function appears to set the frequency to 0 for every transmission:
https://github.com/esphome/esphome/blob/dev/esphome/components/remote_base/rc_switch_protocol.cpp#L57

1.  If the `RCSwitch` remote type is meant to be used for RF only, it should not be used when decoding IR signals.
2. If the `RCSwitch` remote type is meant to support IR, then the `remote_transmitter.transmit_rc_switch_raw` function should work, and not have the frequency set to 0.

Also, when using `dump: all` in `remote_receiver` raw codes are not printed if another decoder is able to successfully decode the signal. Since my codes are being decoded by `RCSwitch`, the raw codes are not printed. This can be worked around by explicitly setting `dump: raw` to forcefully show all raw codes.

Sending any other IR codes using `remote_transmitter.transmit_raw` or `remote_transmitter.transmit_nec` on this node works as expected.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  node_name: remote

packages:
  device_base: !include common/device_base.yaml

esphome:
  name: ${node_name}
  platform: ESP32
  board: nodemcu-32s

remote_receiver:
  pin:
    number: GPIO34
    inverted: True
    mode: INPUT_PULLUP
  dump: all

remote_transmitter:
  pin: GPIO32
  carrier_duty_percent: 50%
  id: ir_trans

  services:
    - service: remote_send_raw_command
      variables:
        command: int[]
      then:
        - remote_transmitter.transmit_raw:
            carrier_frequency: 38kHz
            code: !lambda 'return command;'
    - service: remote_send_nec_command
      variables:
        address: int
        command: int
      then:
        - remote_transmitter.transmit_nec:
            address: !lambda 'return address;'
            command: !lambda 'return command;'
 
binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO26
      mode: INPUT_PULLUP
      inverted: True
    id: tv_power
    internal: true # don't expose to HA
    filters:
      - delayed_off: 1s # debounce, when powering on, the TV turns off briefly
      - delayed_on: 100ms

switch:
  - platform: template
    name: "TV Power"
    lambda: |-
      if (id(tv_power).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0x20DF
          command: 0x54AB
    turn_off_action:
      - remote_transmitter.transmit_nec:
          address: 0x20DF
          command: 0xA45B

  - platform: template
    name: "test switch"
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          protocol: 4
          code: '11011000000'
          transmitter_id: ir_trans
          repeat: 5
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I've tried decoding the Raw IR from these remotes in [IRScrutinizer](https://github.com/bengtmartensson/IrScrutinizer) which detects the protocol as `F12_relaxed`. If this is correct (I'm not 100% sure), then it would imply that this signal is being incorrectly decoded as RCSwitch and should be RAW since ESPHome's IR libraries do not natively support `F12_relaxed`.

erkme73, 2021-05-13:
I wanted to say "me too" - though mine is on a Sonoff 433mhz RF bridge (built on the ESP8266).  If I compile the yaml with 1.15.3 or older (add on to home assistant, or standalone),  both transmit and receive work 100%.  On builds newer than 1.15.3 (or thereabouts), only the receive works.  Confirmed on multiple machines, and with multiple RF bridges.   FWIW, a staffer has confirmed the problem, and is researching it.

lanrat, 2021-05-14:
I just tested again on the latest version of ESPHome and on 1.15.3, and I have the same problem on both versions with IR.

erkme73, 2021-05-14:
Set your wait_time to 0 milliseconds

<sent from my non-iFruit device>

saue0, 2022-01-28:
Have the same problem... corrected by changing this line :
void RCSwitchBase::transmit(RemoteTransmitData *dst, uint64_t code, uint8_t len) const {
  dst->set_carrier_frequency(38000);   #<==== forcing the frequency 
  this->sync(dst);
  for (int16_t i = len - 1; i >= 0; i--) {
    if (code & ((uint64_t) 1 << i))
      this->one(dst);
    else
      this->zero(dst);
  }
}

Now this command work:
  - platform: template
    name: Foyer
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '11011000100'
          protocol: 4

michd, 2023-09-20:
> Have the same problem... corrected by changing this line : void RCSwitchBase::transmit(RemoteTransmitData *dst, uint64_t code, uint8_t len) const { dst->set_carrier_frequency(38000); #<==== forcing [...]

I ran into the similar thing and tried this "fix" in my esphome docker container, and it indeed works. I'd like to fix this properly though, and contribute a PR.

From what I gather, all the supported protocols set their own carrier frequency. Typically, it looks like a protocol is either designed for IR or RF, but not both, so that approach works. In this case, rc_switch is used on both RF and IR, but the implementation sets the carrier frequency to 0, for RF, causing it to be broken for IR.

## Proposed solution

It's bit of a breaking change for RF users who have left `carrier_duty_percent` on `50%`, but allows it to work for both RF and IR:

1. ESPHome change: Set the carrier frequency to 38kHz in `rc_switch_protocol.cpp`
2. User change: Users of transmit_rc_switch_* must set `carrier_duty_percent` to `100%` to cancel out the carrier frequency; IR users leave it `50%` to enable it

I'll go ahead and put together a pull request for this. I'll link here.

closed 2024-07-10 by jesserockz