esphome/issues #2045: Enabling BLE on a ST7789V TFT LCD TTGO T-Display module crashes the device until it starts in safe mode

xbmcnut, 2021-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
 ESP32

**ESPHome version (latest production, beta, dev branch)**
 1.17.2

**Affected component:**
 https://esphome.io/components/display/st7789v.html

**Description of problem:**
If I add ```esp32_ble_tracker:``` to the working code below, the device crashes mutiple times until it starts in safe mode. Running the same board with only display works fine and another board with only BLE Tracker works fine but the two don't play nicely. When adding the BLE Tracker, the code was uploaded via USB as advised in the docs.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_tdisplay
  platform: ESP32
  board: featheresp32

wifi:
  ssid: !secret wifi_rumpus_ssid
  password: !secret wifi_pass

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp_tdisplay"
    password: "somepassword"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "somepassword"

ota:
  password: "somepassword"

# Example minimal configuration entry
spi:
  clk_pin: GPIO18
  mosi_pin: GPIO19

display:
  - platform: st7789v
    backlight_pin: GPIO4
    cs_pin: GPIO5
    dc_pin: GPIO16
    reset_pin: GPIO23

```

**Logs (if applicable):**

```
Using 'COM6' as serial port.
Showing logs:
[21:38:40]ets Jul 29 2019 12:21:46
[21:38:40]
[21:38:40]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:38:40]configsip: 0, SPIWP:0xee
[21:38:40]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:38:40]mode:DIO, clock div:2
[21:38:40]load:0x3fff0018,len:4
[21:38:40]load:0x3fff001c,len:1044
[21:38:40]load:0x40078000,len:8896
[... 287 lines cut ...]
[21:39:00]
[21:39:00]Backtrace: 0x4010425c:0x3fff7ca0 0x40105d51:0x3fff7d10 0x40109f26:0x3fff7d80 0x4010dad0:0x3fff7dd0 0x40109d26:0x3fff7e00 0x400955f3:0x3fff7e20 0x4008ebed:0x3fff7e60
[21:39:00]
[21:39:00]Rebooting...
[21:39:00]ets Jul 29 2019 12:21:46
[21:39:00]
[21:39:00]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:39:00]configsip: 0, SPIWP:0xee
[21:39:00]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:39:00]mode:DIO, clock div:2
[21:39:00]load:0x3fff0018,len:4
[21:39:00]load:0x3fff001c,len:1044
[21:39:00]load:0x40078000,len:8896
[21:39:00]load:0x40080400,len:5828
[21:39:00]entry 0x400806ac
[21:39:01][I][logger:166]: Log initialized
[21:39:01][C][ota:366]: There have been 10 suspected unsuccessful boot attempts.
[21:39:01][E][ota:371]: Boot loop detected. Proceeding to safe mode.
[21:39:01][I][app:029]: Running through setup()...
[21:39:01][C][wifi:033]: Setting up WiFi...
[21:39:01][D][wifi:324]: Starting scan...

```

**Additional information and things you've tried:**

buxtronix, 2021-05-19:
I think there simply isn't enough memory on the device. BLE takes up most of the available ram/heap space leaving not enough for the display.

xbmcnut, 2021-05-19:
Thanks. Is there a way to prove that or a better option for a BLE monitor with a display? I wanted to be able to display some connection data on the device if possible.

lanrat, 2021-07-09:
I wonder if this is related to my issue in #2112.

closed 2021-12-03 by github-actions[bot]