esphome/issues #2046: HA 2021.5 colormodes issues with color_interlock

BoukeHaarsma23, 2021-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

1.18.0b3
**Affected component:**

RGBWW

**Description of problem:**
Using the RGBWW color with color_interlock set to true I get some strange behavior in HA 2021.5.

For example:
HA 2021.5: Bulb currently in white mode --> setting color --> nothing happens --> setting white to 0 --> see color
I can see in the logs of the esp that the white value is not set to 0 when setting the color
```
[11:52:09][D][light:265]: 'overhead1' Setting:
[11:52:09][D][light:287]:   Red=100%, Green=0%, Blue=0%
[11:52:09][D][light:290]:   White Value: 100%
[11:52:09][D][light:304]:   Transition Length: 0.2s
```

In HA 2021.4.5 the white channel is correctly set to 0

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: rgbww
    name: "${name}"
    cold_white: output_white
    warm_white: output_warmwhite
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    default_transition_length: 0.2s
    red: output_red
    green: output_green
    blue: output_blue
    color_interlock: true

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

malteger, 2021-05-25:
I have the same issue, there is no automatic change in the white brightness. I would also expect only one brightness slider instead of three for lights with interlocked colors (mine also has `constant_brightness: true`)

I am not even sure if this is a problem in esphome or home-assistant...

oxan, 2021-06-04:
See also #1475

sparkydave1981, 2021-06-05:
> I have the same issue, there is no automatic change in the white brightness. I would also expect only one brightness slider instead of three for lights with interlocked colors (mine also has `constant_brightness: true`)
> 
> I am not even sure if this is a problem in esphome or home-assistant...

To me it seems an issue in ESPhome since my lights are ignoring the interlock: true.  Saying that, they were fine before the HA update that brought in the white channel control...

pmannk, 2021-06-05:
I'm experiencing the same issue too. HA 2021.6 has not resolved it either.
I don't think color_interlock is doing anything wrong - my lights still had the same settings flashed for a few months. The only "change" was the update of HA to 2021.5. I re-flashed the lights this morning but it didn't change the behaviour.

My lights tend to get an uncomfortable hue if the RGB leds are on at the same time as the WW. This is definitely not happening right now, so I'm confident color_interlock is behaving as it should.
It seems more like the colour wheel/selector in HA should be setting white to 0% for interlocked lights when you select a colour, but it isn't.

Eg, when I pick a colour the white value remains at 100%. In earlier versions of HA it would change to 0%:
```
[14:34:55][D][light:265]: 'Light' Setting:
[14:34:55][D][light:287]:   Red=100%, Green=0%, Blue=0%
[14:34:55][D][light:290]:   White Value: 100%
[14:34:55][D][light:304]:   Transition Length: 1.5s
```

sparkydave1981, 2021-06-08:
As far as I'm concerned, for my LED's the `color_interlock` is NOT working, otherwise the lights would not have the white LED on at the same time as the colour ones.

> It seems more like the colour wheel/selector in HA should be setting white to 0% for interlocked lights when you select a colour, but it isn't.

The interlock is a feature of ESPhome, not HA, so ESPhome should be turning off the white channel when a colour is selected.

tungmeister, 2021-06-27:
I'm also seeing this issue - interlock is working but only when changing the white brightness slider down to 0, at which point the colour will show. I believe the intended behavior should be as soon as a colour is selected/set the white brightness should be set to 0 and when selecting a colour temp from the white channels this should restore the white brightness.

oxan, 2021-07-10:
This should be fixed with the merge of esphome/esphome#1895, available in the upcoming v1.20 release.

closed 2021-07-10 by oxan

JohanBraeken, 2021-08-16:
This still seems to be broken after the upgrade to HA 2021.5
I've upgraded to ESPHome 1.20.4 now and when selecting any RGB color with "color_interlock" enabled, it still doesn't switch to the color when the light is already on and "white".
In the ESPHome, the following message is logged: "Cannot enable color and white channel simultaneously with interlock!"

If I turn off "white" manually, only then I can select a color, but the old behaviour was that this "turn off" was done automatically previously when in the HA user interface a manual selection between "white" or "RGB" was made.

Is there any way to get this old behaviour back? Or do I need to make changes to my existing ESPHome light configurations?

malteger, 2021-08-16:
For me this works as expected again with HA 2021.8 and the beta version of esphome 1.21, with options to set either the color or the temperature, which will turn on only the rgb leds or the cwww leds. Previous combinations broke the old behaviour, starting with HA changing the color model I think.

JohanBraeken, 2021-08-16:
OK, I'll wait then until 1.21 is released.
The combination of the currently latest non-beta versions (HA 2021.8.7 and ESPHome 1.20.4) still seem to have this broken behaviour.

sparkydave1981, 2021-08-20:
I can confirm that my ESPhome LED downlights now work perfectly with ESPhome verison 2021.8 and Home Assistant version 2021.8 so I'm very happy

dylan09, 2021-08-21:
@sparkydave1981 Did you changed your ESPHome configuration to one of the new color_temperature light?

JohanBraeken, 2021-08-21:
In HA 2021.8.7, I can select the colors now without turning off "white" first.

However, when choosing white from the color pallette (central area), it does it through the RGB output, resulting in an ugly semi-white.
Previously, with "color_interlock" enabled,  it would swith automatically to the dedicated white output channel when the white color was selected.
Is this an intentional change or still a bug?

oxan, 2021-08-21:
Yeah, that was a more or less intentional change to be consistent with how color modes are supposed to work in HA. If you want to use the white channel, you can set it by using the White or Color Temperature tab.

sparkydave1981, 2021-08-21:
> @sparkydave1981 Did you changed your ESPHome configuration to one of the new color_temperature light?

Hi Dylan, no I didn't change anything in my ESPhome device configuration at all. 
It's setup as a rgbw light with `color_interlock: true`