esphome/issues #2047: Setting static IP error when uploading firmware

jcastro, 2021-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome version: 1.17.2 - Official Hassio Add-on

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 or ESP8266, happens on both

**ESPHome version (latest production, beta, dev branch)**

1.17.2

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
1. Create a new device on ESPHome, with Wifi enabled but without setting an static IP in the Yaml configuration file
2. Once it's setup and working, try to add a `manual_ip configuration. For example:
```manual_ip:
    static_ip: 10.0.10.230
    gateway: 10.0.10.1
    subnet: 255.255.255.0```
3. Once you try to upload the changes, it will try to connect to the NEWLY setup IP address which of course is still not active, instead of trying to use the current one

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

markusg1234, 2021-05-14:
Hi,
To change the IP address you will need to use the use_address: command for the first reprogram that will set the current IP address.

https://esphome.io/components/esphome.html#esphome-changing-node-name

rradar, 2021-05-18:
> Once you try to upload the changes, it will try to connect to the NEWLY setup IP address which of course is still not active, instead of trying to use the current one

This is the expected behavior. See the link from @markusg1234 to solve your problem.

closed 2021-05-18 by rradar