esphome/issues #2051: too many component id candidates

Giuseppe-P, 2021-05-14:
**Docker esphome**

**ESP8266 - NEXTION TOUCH**

**1.17.2**

**N
[image]
extion touch pannel**

**I don't understand this change esphome/esphome#1570 I had a working firmware now it detects errors.:**

**Problem-relevant YAML-configuration entries:**
```yaml
      
binary_sensor:    

    
  - platform: nextion
    page_id: 0
    component_id: 6
    id: set_up #custom sensor for the power button on the display
    name: "set up"
    on_press:
        lambda: |-
         id(set_temp) + 0.5;
         delay(10);
         if (id(set_temp) == 31){
          id(set_temp) = 30.0;
          }
     
  - platform: nextion
    page_id: 0
    component_id: 7
    id: set_down #custom sensor for the power button on the display
    name: "set down"
    on_press:
      lambda: |-
       id(set_temp) - 0.5; 
       delay(10);
        if (id(set_temp) == 14) {
         id(set_temp) = 15.0;
         }
         
  - platform: nextion
    page_id: 1
    component_id: 6
    id: set_up_local #custom sensor for the power button on the display
    name: "set up local"
    internal: true
    on_press:
      lambda: |-
         id(set_temp_local) = id(set_temp_local) + 0.5;
         if (id(set_temp_local) == 31){
          id(set_temp_local) = 30.0;
          }

  - platform: nextion
    page_id: 1
    component_id: 7
    id: set_down_local #custom sensor for the power button on the display
    name: "set down local"
    internal: true
    on_press:
      lambda: |-
       id(set_temp_local) = id(set_temp_local) - 0.5; 
        if (id(set_temp_local) == 14) {
         id(set_temp_local) = 15.0;
         }
         
  - platform: nextion
    page_id: 0
    component_id: 4
    id: heater_off #custom sensor for the power button on the display
    name: "Turn Heater Off"

        
  - platform: nextion
    page_id: 0
    component_id: 5
    id: heater_on #custom sensor for the power button on the display
    name: "Turn Heater On"

  - platform: nextion
    page_id: 1
    component_id: 4
    name: "Turn Heater Off local"
    internal: true
    on_press:
    - then:
        lambda: |-
          id(on_off_manual) = 0;

  - platform: nextion
    page_id: 1
    component_id: 5
    name: "Turn Heater On local"
    internal: true
    on_press:
    - then:
        lambda: |-
          id(on_off_manual) = 1;

```

**Logs (if applicable):**

```
INFO Reading configuration /config/termostato_nextion.yaml...
INFO Detected timezone 'UTC' with UTC offset 0
Failed config

binary_sensor.nextion: [source /config/termostato_nextion.yaml:296]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 0
  component_id: 6
  id: set_up
  name: set up
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(set_temp) + 0.5;
            delay(10);
            if (id(set_temp) == 31){
             id(set_temp) = 30.0;
             }
binary_sensor.nextion: [source /config/termostato_nextion.yaml:309]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 0
  component_id: 7
  id: set_down
  name: set down
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(set_temp) - 0.5; 
            delay(10);
             if (id(set_temp) == 14) {
              id(set_temp) = 15.0;
              }
              
binary_sensor.nextion: [source /config/termostato_nextion.yaml:322]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 1
  component_id: 6
  id: set_up_local
  name: set up local
  internal: True
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(set_temp_local) = id(set_temp_local) + 0.5;
            if (id(set_temp_local) == 31){
             id(set_temp_local) = 30.0;
             }
binary_sensor.nextion: [source /config/termostato_nextion.yaml:335]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 1
  component_id: 7
  id: set_down_local
  name: set down local
  internal: True
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(set_temp_local) = id(set_temp_local) - 0.5; 
             if (id(set_temp_local) == 14) {
              id(set_temp_local) = 15.0;
              }
              
binary_sensor.nextion: [source /config/termostato_nextion.yaml:348]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 0
  component_id: 4
  id: heater_off
  name: Turn Heater Off
binary_sensor.nextion: [source /config/termostato_nextion.yaml:355]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 0
  component_id: 5
  id: heater_on
  name: Turn Heater On
binary_sensor.nextion: [source /config/termostato_nextion.yaml:361]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 1
  component_id: 4
  name: Turn Heater Off local
  internal: True
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(on_off_manual) = 0;
binary_sensor.nextion: [source /config/termostato_nextion.yaml:371]
  
  Too many candidates found for 'nextion_id' type 'nextion::Nextion' Some are 'agg_set', 'clima_nextion', 'nextion_stamp_ora', 'nextion_brightness', 'nextion_page'.
  platform: nextion
  page_id: 1
  component_id: 5
  name: Turn Heater On local
  internal: True
  on_press: 
    - then: 
        - lambda: !lambda |-
            id(on_off_manual) = 1;

```

**Additional information and things you've tried:**

Giuseppe-P, 2021-05-14:
actually the errors related to too many calls in the nextion display entry.  But this update introduced a problem.  The repeated calls under the display item allow you to send data to nextion at different intervals so as not to flood the esp chip.  I had to put all the lambda calls I had under a single candidate but this is not good because the chip has to perform many functions under a single interval and then reboots continuously.

glmnet, 2021-05-15:
to have exactly the previous behavior modify each 
```yaml
binary_sensor:    

    
  - platform: nextion
    page_id: 0
    component_id: 6
    id: set_up #custom sensor for the power button on the display
    name: "set up"
    on_press:
        lambda: |-
         id(set_temp) + 0.5;
         delay(10);
         if (id(set_temp) == 31){
          id(set_temp) = 30.0;
          }
````

to

```yaml
binary_sensor:    
  - platform: nextion
    nextion_id: agg_set # *** ADD THIS LINE TO EACH SENSOR ***
    page_id: 0
    component_id: 6
    id: set_up #custom sensor for the power button on the display
    name: "set up"
    on_press:
        lambda: |-
         id(set_temp) + 0.5;
         delay(10);
         if (id(set_temp) == 31){
          id(set_temp) = 30.0;
          }
```` 

You failed to provide the full file so I might be missing something as I've never used nextion.

But given the description: if you want to have different updates intervals, I'd recommend you use a single:

```yaml
display:
  - platform: nextion
```
  
component and also use the `interval:` component and send the updates from there (at different intervals), then you won't need to specify the `nextion_id` as there is only one.

BTW, having a config like yours (5 nextion displays) suggest you have actually those many physical displays. The error is there because ESPHome wants to be sure to which display those sensors belongs.

Giuseppe-P, 2021-05-15:
Thanks for your help, I just needed to assign nextion_id to the binary sensors and link them to the ids of the nextion display. I fixed the code that I am attaching below. Thanks again

-----------------------------------------------------------------------------
TERMOSTATO ESPHOME NEXTION ESP8266

substitutions:
  devicename: termostato_nextion
  friendly_name: Termostato Nextion
  device_description: WEMOS Mini - Termostato Nextion riscaldamenti
  
esphome:
  name: ${devicename}
  comment: ${device_description}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "TP-xxxxxxxx"
  password: "xxxxxxx"
  reboot_timeout: 900s
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Termostato Nextion"
    password: "xxxxxxxxxx"

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
 reboot_timeout: 1200s
 
ota:

dallas:
  - pin: GPIO5
  #0x0E041623899DFF28

  
switch:
  - platform: gpio
    name: "Heating"
    id: "Heating_D3"
    pin: GPIO4    
    
  - platform: restart
    name: Restart ${device_description}
    
sensor:
  - platform: dallas
    address: 0x0E041623899DFF28
    name: ${friendly_name} Temperatura interna
    id: temp_dallas_1 
    
  - platform: homeassistant #configuration.yaml valori ricevuti da HA
    entity_id: sensor.set_temp
    id: set_temperature
    
  - platform: template #invia il valore del setpoint ad HA
    name: "Set Temperature"
    lambda: |-
        id(set_temp) = id(set_temperature).state;
        return id(set_temp);
    update_interval: 500ms
      
  # Segnala per quanto tempo il dispositivo è stato alimentato (in minuti)
  - platform: uptime
[... 356 lines cut ...]
    restore_value: no
  - id: on_off_manual            
    type: int
    initial_value: '0'
    restore_value: no
  - id: blocco_stato_local            
    type: int
    initial_value: '0'
    restore_value: no
  - id: brightness           
    type: int
    initial_value: '0'
    restore_value: yes
  - id: blocco_manual             
    type: int
    initial_value: '0'
    restore_value: no
  - id: blocco_temp             
    type: float
    initial_value: '0.0'
    restore_value: no
  - id: blocco_set             
    type: float
    initial_value: '0.0'
    restore_value: no

closed 2021-05-15 by Giuseppe-P