esphome/issues #2055: ili9341: Blank Screen after Initial Frame Is Painted

sebasgo, 2021-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I am using ESPHome on Linux installed via pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 on [az-delivery-devkit-v4](https://www.az-delivery.de/en/products/esp32-dev-kit-c-v4-unverlotet?variant=32437206548576)

**ESPHome version (latest production, beta, dev branch)**

ESPHome dev version (esphome/esphome@7e3b8fd346b3cf91d7bf9b0be73ec4584e9a3f60)

**Affected component:**

[ILI9341 TFT LCD](https://esphome.io/components/display/ili9341.html)

**Description of problem:**

I have tried to setup the display of the [AZ-Touch wall enclosure with 2.8 inch touchscreen](https://www.az-delivery.de/en/products/az-touch-wandgehauseset-mit-2-8-zoll-touchscreen-fur-esp8266-und-esp32) with ESPHome.  I got so far that the display (and its backlight) turned on when the board was powered up. I could see that my painting code was working. The "Hello world" message was visible. After a few msecs the screen turns off completely, though. I have captured a [slowmo video](https://photos.app.goo.gl/oqxoxujNrmpotrqm6) to illustrate the problem.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ili9341
  platform: ESP32
  board: esp32dev
  platformio_options:
    platform: espressif32
    board: az-delivery-devkit-v4
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L

logger:

spi:
  clk_pin: 18
  mosi_pin: 23
  miso_pin: 19

font:
  - file: fonts/FiraSans-Medium.ttf
    id: fira_medium
    size: 20

display:
  - platform: ili9341
    model: TFT 2.4
    cs_pin: 5
    dc_pin: 4
    reset_pin: 22
    lambda: |-
      it.print(0, 0, id(fira_medium), COLOR_WHITE, TextAlign::TOP_LEFT, "Hello World!");
    update_interval: 5s

output:
  - platform: ledc
    id: display_backlight_output
    pin: 15

light:
  - platform: monochromatic
    id: backlight
    output: display_backlight_output
    restore_mode: ALWAYS_ON
```

**Logs (if applicable):**

```
[21:52:52][I][app:059]: setup() finished successfully!
[21:52:52][I][app:105]: ESPHome version 1.19.0-dev compiled on May 15 2021, 21:52:28
[21:52:52][C][spi:081]: SPI bus:
[21:52:52][C][spi:082]:   CLK Pin: GPIO18 (Mode: OUTPUT)
[21:52:52][C][spi:083]:   MISO Pin: GPIO19 (Mode: INPUT)
[21:52:52][C][spi:084]:   MOSI Pin: GPIO23 (Mode: OUTPUT)
[21:52:52][C][spi:085]:   Using HW SPI: YES
[21:52:52][C][ledc.output:032]: LEDC Output:
[21:52:52][C][ledc.output:033]:   Pin GPIO15 (Mode: OUTPUT)
[21:52:52][C][ledc.output:034]:   LEDC Channel: 0
[21:52:52][C][ledc.output:035]:   Frequency: 1000.0 Hz
[21:52:52][C][logger:189]: Logger:
[21:52:52][C][logger:190]:   Level: DEBUG
[21:52:52][C][logger:191]:   Log Baud Rate: 115200
[21:52:52][C][logger:192]:   Hardware UART: UART0
[21:52:52][C][light:178]: Light 'backlight'
[21:52:52][C][light:180]:   Default Transition Length: 1.0s
[21:52:52][C][light:181]:   Gamma Correct: 2.80
[21:52:52][C][ili9341:029]: ili9341
[21:52:53][C][ili9341:029]:   Rotations: 0 °
[21:52:53][C][ili9341:029]:   Dimensions: 240px x 320px
[21:52:53][C][ili9341:030]:   Width: 240, Height: 320,  Rotation: 0
[21:52:53][C][ili9341:031]:   Reset Pin: GPIO22 (Mode: OUTPUT)
[21:52:53][C][ili9341:032]:   DC Pin: GPIO4 (Mode: OUTPUT)
[21:52:53][C][ili9341:035]:   Update Interval: 5.0s
```

**Additional information and things you've tried:**
I have tested the hardware setup with one of demos supplied by AZ Delivery. With the demo, the screen works flawlessly. So I am pretty sure this is a software issue. From this demo, I took the IO pin configuration, so that should be correct, too.

Daniel0984, 2021-06-01:
Hi,

I've got the same display module and found that i needed to invert the display pin:

```
output:
  - platform: ledc
    pin: GPIO15
    inverted: yes
    id: gpio_32_backlight_pwm
```

maybe give this a try.

sebasgo, 2021-06-02:
Hi Daniel,

inverting the backlight pin worked indeed. Thanks for the suggestion!

closed 2021-06-02 by sebasgo