esphome/issues #2066: [Question] Pulse counter, update interval and filters

rufik, 2021-05-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266 D1 Mini

**ESPHome version (latest production, beta, dev branch):** v1.17.2

**Affected component:** pulse_counter

**Description of problem:** Just a question.

**Problem-relevant YAML-configuration entries:**
```yaml
See below.
```

**Logs (if applicable):**

```
None.
```

**Additional information and things you've tried:**

I'd like to get some clarification about pulse counter, it's update interval and templates with aggregation (min, max, avg).
Let's assume I have simple anemometer that closes contacts when rotating (1 click per second = 2.4 km/h). My esphome config is just like that:
```
sensor:
  - platform: pulse_counter
    pin:
      number: D1
      mode: INPUT
    id: windspeed
    name: "Wind speed"
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    update_interval: 5s
    accuracy_decimals: 1
    unit_of_measurement: 'km/h'
    filters:
      - lambda: return (x / 60.0) * 2.4;    # X is per 1min, so clicks/60sec * 2.4

  - platform: template
    name: "Wind speed avg"
    update_interval: 60s
    lambda: |-
      return id(windspeed).state;
    filters:
      - sliding_window_moving_average:
          window_size: 2
          send_every: 2
          send_first_at: 2

  - platform: template
    id: windspeed_max
    name: "Wind speed max"
    update_interval: 60s
    lambda: |-
      return id(windspeed).state;
    filters:
      - max:
          window_size: 5
          send_every: 5
          send_first_at: 2
```

So with `Wind speed` sensor i just pulse counter that gives me _pulses per 1 min_ as measurement, which I recalculate to wind speed (1pulse/sec = 2.4 km/h) using lamdba expression. Result is being pushed to HA every **5 seconds**.
Next want two additional sensors:
- average of 2 minutes as `Wind speed avg`,
- max of 5 minutes as `Wind speed max`

So I simply defined template sensors and used lambda to get values (**already recalculated** I hope) of current wind speed, and applied avg and max filters. Looks fine so far :)
Then I started to watch values coming from esphome using Graphana. The max values doesn't look good:
[image]

I suppose something is not really right configured within max filter. I started to read doc about filters options and question started to come up:

1. `update_interval` option is just a interval that value is pushed to HA, right? Sensor value is still given as `pulses per 1 min`, right?
2. So **how frequent** template sensor `Wind speed max` is being given the value into lambda `return id(windspeed).state;`? Every 1 minute or every 5 sec (`update_interval` defined in original sensor)?
3. Is `window_size` working on last X values (seems logical)? If template sensor is updated every 5 sec, window size is 5, so it's **max of last 25 sec**?
4. Does `send_every` option work the same way? If template sensor is updated every 5 sec, window size is 5, so it's being pushed to HA **every 25 sec**? My data graph negates it - I'm getting values every 5 min...

OttoWinter, 2021-05-24:
It's because your template sensors have a lower update rate than the base sensor. The `update_interval` option means how often the given `lambda` is recalculated, and not how often it's published (You can control that via `send_every`)

You can fix it by changing `update_interval` tos `5s` as in the pulse counter, and `send_every` to 20 (so in effect send every `20 * 5s = 60s`)

rufik, 2021-05-24:
Thanks a lot Otto, that's it! I haven't read this section carefully enough or maybe it should be explained better.

closed 2021-05-24 by rufik