esphome/issues #2068: error: 'BLACK' is not a member of 'esphome::Color'

jagannew, 2021-05-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-s

**ESPHome version (latest production, beta, dev branch)**

version: 1.18.0 

**Affected component:**

light > effects > addressable_lambda

**Description of problem:**
Few days before I updated (version: 1.18.0 ) esphome in home Assistant, 
when I try to add some effects for my LED, I came across this problem.   "error: 'BLACK' is not a member of 'esphome::Color' "
when I try to upload the code to the board. it is showing some ERROR
when I VALIDATE the code it cont find any problem. but in my other device, it is running in the same code but it was uploaded months before (in the old version)

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: esp_led_unit
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "xxxx"
  password: "xxxx"
  manual_ip:
    static_ip: 192.168.10.94
    gateway: 192.168.10.1
    subnet: 255.255.255.0
    dns1: 192.168.10.1
    dns2: 192.168.10.1
  ap:
    ssid: "Esp-Buzzer Unit Fallback Hotspot"
    password: "b122PIdkhceR"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
ota:
  password: "xxxxxx"

status_led:
  pin: GPIO2

light:

  - platform: fastled_clockless
    chipset: WS2812b
    pin: 19
    num_leds: 100
    rgb_order: GRB
    name: "LED Notification"
    effects:
      - addressable_lambda:
          name: "My Custom Effect"
          update_interval: 16ms
          lambda: |-
            static uint16_t progress = 0;
            uint16_t changes = 0;
            if (initial_run) {
              progress = 0;
              it.all() = Color::BLACK;
              return;
            }

```

**Logs (if applicable):**

```
INFO 200 GET /edit?configuration=temp.yaml (0.0.0.0) 0.47ms
INFO 101 GET /upload (0.0.0.0) 0.38ms
INFO 200 POST /edit?configuration=temp.yaml (0.0.0.0) 0.59ms
INFO Running command 'esphome --dashboard /config/esphome/temp.yaml run --upload-port OTA'
INFO Process exited with return code 1
```

**Additional information and things you've tried:**
Compiling & Uploading: 
```
INFO Reading configuration /config/esphome/temp.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/esp_led_unit
Processing esp_led_unit (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32@3.0.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.210126 (1.0.4) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.2.2
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/esp_led_unit/.pioenvs/esp_led_unit/src/main.cpp.o
Archiving /data/esp_led_unit/.pioenvs/esp_led_unit/lib150/libAsyncTCP-esphome.a
Compiling /data/esp_led_unit/.pioenvs/esp_led_unit/lib790/FS/FS.cpp.o
/config/esphome/temp.yaml: In lambda function:
/config/esphome/temp.yaml:51:20: error: 'BLACK' is not a member of 'esphome::Color'
               it.all() = Color::BLACK;
                    ^
Compiling /data/esp_led_unit/.pioenvs/esp_led_unit/lib790/FS/vfs_api.cpp.o
*** [/data/esp_led_unit/.pioenvs/esp_led_unit/src/main.cpp.o] Error 1
========================== [FAILED] Took 1.88 seconds ==========================
```
I got the light effect in ESPhome  website only  https://esphome.io/components/light/index.html
I'm not an expert just follow the steps. if I made any mistakes I'm sorry for that. thank you for helping.

OttoWinter, 2021-05-24:
Looks like those constants [here](https://github.com/esphome/esphome/pull/1551/files#diff-af137927711b612430cc357d4401954f0caaaf04adcddec1d29b00459ae0d017L157-L158) got accidentally removed in the linked PR. CC @SenexCrenshaw

jagannew, 2021-06-17:
When will it be fixed for everyone?
now I updated to latest one (1.19.0) still not yet updated

SenexCrenshaw, 2021-06-18:
Not at a place I can test but should 
``Color::BLACK`` be ``Color::COLOR_BLACK`` ?

https://github.com/esphome/esphome/blob/master/esphome/core/color.h#L148

Validation in the dashboard wont catch lambda code errors. That will come out at compile time.

OttoWinter, 2021-06-18:
@SenexCrenshaw Currently the symbol is `COLOR_BLACK` (without `Color::` as it's not in a Color struct member). In previous versions, `Color::BLACK` worked too (and IMHO it makes more sense as a static class member)

If [these two lines](https://github.com/esphome/esphome/pull/1551/files#diff-af137927711b612430cc357d4401954f0caaaf04adcddec1d29b00459ae0d017L157-L158) are re-added in the new `Color` class, the old way would work too. Until then you can use `COLOR_BLACK` as @SenexCrenshaw mentioned.

SenexCrenshaw, 2021-06-18:
@OttoWinter  Good catch. Please use ``COLOR_BLACK`` . We merged some color structures and I dont remember exactly why it was removed/renamed. I had/have a large display buffer PR and most likely it was aligning names. Sorry for the confusion.

varadujagan, 2021-06-27:
@OttoWinter  thanks for the comment. now instead of `Color::BLACK` I'm using `ESPColor(0,0,0)`  works now . if you are permanently changed to `COLOR_BLACK` please update it in the example files website [https://esphome.io/components/light/index.html?highlight=effects#light-effects](url) So that new fellow like me can understand.
Thank you

closed 2021-08-03 by jesserockz