esphome/issues #2071: ESP8266 I2C bus gets stuck

mikosoft83, 2021-05-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 mini (ESP8266)

**ESPHome version (latest production, beta, dev branch)**
Currently 0.18.0 but happening since 0.17.0

**Affected component:**
most probably I2C bus
https://esphome.io/components/i2c.html

**Description of problem:**
I have SH1106 screen and SHT3 temp/hum sensor connected over I2C bus. I have a very light code running at present with the screen only displaying time and SHT sensors set up + HA API as well. At some point (a random one) the I2C bus starts throwing "Received NACK on transmit of address 0x3C" (that's the screen), after which the SHT3 also stops responding and starts throwing the same (just a different address). After some time the ESP reboots but the I2C bus stays stuck until I completely power off the device, let it sit for a minute or two and power up again.
I have two such devices and they both started doing it since ESPHome 0.17.0.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: ${i2cData}
  scl: ${i2cClock}
  frequency: 200kHz
  scan: True
  id: bus_a

display:
  - platform: ssd1306_i2c
    model: "SH1106 128x64"
    address: 0x3C
    update_interval: 0.1s
    id: idisplay
    lambda: |-

        // Bouncy time
        it.strftime(id(ss_x), id(ss_y), id(big_font), "%H:%M", id(ha_time).now());

sensor:
  - platform: sht3xd
    temperature:
      name: "${friendlyName} Temperature"
      id: ${unitName}_temperature
      filters:
        - offset: -4
    humidity:
      name: "${friendlyName} Humidity"
      id: ${unitName}_humidity
    address: 0x44
    update_interval: 15s

```

**Logs (if applicable):**

```
[22:12:26][D][sht3xd:065]: Got temperature=27.32°C humidity=40.37%
[22:12:26][D][sensor:099]: 'Pithy Screen Milan Temperature': Sending state 23.32052 °C with 1 decimals of accuracy
[22:12:26][D][sensor:099]: 'Pithy Screen Milan Humidity': Sending state 40.37384 % with 1 decimals of accuracy
[22:12:28][W][i2c:070]: Received NACK on transmit of address 0x3C
[22:12:30][W][i2c:070]: Received NACK on transmit of address 0x3C
[22:12:31][W][i2c:070]: Received NACK on transmit of address 0x3C
...
[22:15:45][W][i2c:070]: Received NACK on transmit of address 0x44
...
[22:29:16][E][api:084]: No client connected to API. Rebooting...
[22:29:16][I][app:132]: Forcing a reboot...
[22:29:16][W][wifi_esp8266:444]: Event: Disconnected ssid='redacted' bssid=38:43:7D:BA:E1:6B reason='Association Leave'
[22:29:16]
[22:29:16] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[22:29:16]
[22:29:16]load 0x4010f000, len 3584, room 16 
[22:29:16]tail 0
[22:29:16]chksum 0xb0
[22:29:16]csum 0xb0
[22:29:16]v2843a5ac
[22:29:16]~ld
[22:29:16][I][logger:170]: Log initialized
[22:29:16][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[22:29:16][I][app:029]: Running through setup()...
[22:29:16][C][sht3xd:019]: Setting up SHT3xD...
[22:29:18][W][i2c:070]: Received NACK on transmit of address 0x44
[22:29:18][E][component:092]: Component was marked as failed.
[22:29:18][C][ssd1306_i2c:010]: Setting up I2C SSD1306...
[22:29:19][W][i2c:070]: Received NACK on transmit of address 0x3C
[22:29:19][E][component:092]: Component was marked as failed.
...
[22:29:28][C][i2c:028]: I2C Bus:
[22:29:28][C][i2c:029]:   SDA Pin: GPIO5
[22:29:28][C][i2c:030]:   SCL Pin: GPIO4
[22:29:28][C][i2c:031]:   Frequency: 200000 Hz
[22:29:28][I][i2c:033]: Scanning i2c bus for active devices...

```
Here the log ends, nothing gets found on the I2C bus.

**Additional information and things you've tried:**
I tried stripping the config to bare minimum, I want it connected to HA so I didn't strip that, without that the device doesn't make sense. I already removed captive portal and AP.

LeoMaguire, 2021-07-11:
Struggling with same type of issue - when the ESP8266 reboots the i2c bus is dead until pushing a sketch changing i2c pins back and forth.

Doesn't happen with arduino IDE using code directly from atlas-scientific.

mmakaay, 2021-11-13:
I haven't seen any new issues with this since implementing the I2C bus reset code, so I think that this issue has been fixed. Closing the issue now.

mikosoft83, 2021-11-13:
Well unfortunately I just tested my device with the new codebase and it's still the same.

mmakaay, 2021-11-13:
Then take a look at the log output of the device to see what it reports about I2C bus recovery after a cold boot and a device reset. I'm curious to see if the setup dump log shows a successful recovery here.
When the bus locks up from a clean state or after a successful reset recovery, then it might also be a device misbehaving on the bus, or the wiring might be causing issues (but that is especially true for long wires). In that case, you'd need to do some troubleshooting to see what device is the culprit, by starting with a clean device and adding the I2C components one by one.

mikosoft83, 2021-11-14:
Yeah, I had the device freeze on me after some 2 days of running so I'll try to leave it again and if it repeats then I'll put it on USB logging so that I have something. Should I then add the information here or create a new issue?

mmakaay, 2021-11-14:
Since you started this issue, I think it's fine if you continue in here.
I'll re-open the ticket as well for you.

mikosoft83, 2021-11-14:
Thanks, I'll get back in a couple of days when I have more information.

mikosoft83, 2021-11-17:
As it is now the device is working fine so I guess the previous one getting stuck was either a fluke or maybe I was just hallucinating. I will close this issue so it doesn't hang on and if anything occurs in the meantime I'll open a new one.

closed 2021-11-17 by mikosoft83

mmakaay, 2021-11-17:
Thanks, I hope for hallucination here ;-)

Svrbinek, 2022-01-11:
Hi, I have a similar problem. Sometimes the bus cuts and then the sensors don't go. Only a restart will help. I have a 24-hour restart interval, but I would like someone else to detect the bus error and reset itself as you write about it. However, I did not find anywhere on the internet a description of how this is solved, by what code. Can you write it here? Thank you