esphome/issues #2076: ESPHome device always "unavailable" after Home Assistant server restart

GazTheGeek, 2021-05-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Homeassistant core-2021.5.5 / Raspberry Pi

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff TH16 Board: esp01_1m

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.18.0
**Affected component:**

https://esphome.io/devices/sonoff.html

**Description of problem:**

After every Home Assistant server reboot the entities for this device come back "unavailable". To fix the problem I have to delete the integration and re add it. Then it work fine until the next reboot.

I have other Sonoff TH16s installed and they do not have the same issue.

I also have a Wemos D1 Mini that sometime has the same issue.

In the ESPHome UI The Sonoff shows green and the logs will some the component working fine. The issue is with Home assistant showing the entity as "unavailable"

Wifi signal is strong this unit is located close to the router.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: bh_sonoff03
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: xxx
  password: xxx

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "BH Sonoff 03"
    password: "xxxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    pin: GPIO12
    name: "Stairs Light"
    id: stairs_light_relay

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14 #Yellow wire on jack (Tip of Pin)
      inverted: True
    id: stairs_lights_top_sw
    name: "Stairs Light Top Switch"
    on_state:
      then:
        - switch.toggle: stairs_light_relay
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

Nothing in the Homeassistant logs for this component.

```

**Additional information and things you've tried:**

GazTheGeek, 2021-07-18:
I still have this issue. Anybody got any ideas?

glmnet, 2021-07-18:
How well can you reproduce this issue? Does it happens always when you restart ha? Is it happening to several nodes or only one?

glmnet, 2021-07-18:
If it is happening always then that's a good thing. You need to increase logger verbosity on the ESPHome component (ha side, we assume issue is there) then something related to the node should show up. Compare with another node what is the first message logged on a worker node and let's start finding what goes differently. 

It's happening to me too but it's random in my case. However using the reload integration in my case fixes the issue. So this might be a different problem. 

For ha logging you'll need to modify your configuration yaml file and include the logger there. https://www.home-assistant.io/integrations/logger/. I believe it should be `home assistant.components.esphome: debug`

GazTheGeek, 2021-07-18:
Yesterday I deleted the ESPHome integration before I could restart the ESPHome devices were found by HA so I re configured them all. The problem appears to be fixed. I've done several restarts since and the devices are all working fine.

closed 2021-07-18 by GazTheGeek