esphome/issues #2078: Command line --port flag not working

DutchmanNL, 2021-05-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A issue is related to command line

**ESPHome version (latest production, beta, dev branch)**

latest production 1.18.0

**Affected component:**

command line

**Description of problem:**
The flag port is not working to have the dashboard running at a different port than 6052

`esphome config/ dashboard`
starts the dashboard as expected at 6052

`esphome config/ dashboard --port 6053`
Fails with : 
`usage: esphome [-h] [-v] [-q] [-s key value]
               [configuration ...] {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard,vscode,update-all} ...
esphome: error: argument command: invalid choice: '6053' (choose from 'config', 'compile', 'upload', 'logs', 'run', 'clean-mqtt', 'wizard', 'mqtt-fingerprint', 'version', 'clean', 'dashboard', 'vscode', 'update-all')`

**Problem-relevant YAML-configuration entries:**
N/A

**Logs (if applicable):**

```
usage: esphome [-h] [-v] [-q] [-s key value]
               [configuration ...] {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard,vscode,update-all} ...
esphome: error: argument command: invalid choice: '6053' (choose from 'config', 'compile', 'upload', 'logs', 'run', 'clean-mqtt', 'wizard', 'mqtt-fingerprint', 'version', 'clean', 'dashboard', 'vscode', 'update-all')

```

**Additional information and things you've tried:**

DutchmanNL, 2021-09-26:
Avoid stall

bdraco, 2023-11-13:
Please use `--device` instead of `--port`

https://github.com/esphome/esphome/blob/45276cc24452b7ea1a7a0c6f88ed11ca38a4a1e6/esphome/dashboard/dashboard.py#L409

closed 2023-11-13 by bdraco