esphome/issues #2079: DHT sensor consistently reads nan state from D1 mini

Bartdmm, 2021-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I'm using Home Assistant (formerly, confusingly, hass.io) as a VM, with ESPHome running as an add-on.

**ESP (ESP32/ESP8266, Board/Sonoff):**

I'm using a Wemos D1 Mini v3.0.0, which is ESP8266-based.

**ESPHome version (latest production, beta, dev branch)**

The ESPHome version I am running is 1.18.0.

**Affected component:**

The component that's giving me trouble is a [DHT shield](https://nl.aliexpress.com/item/32805120516.html?spm=a2g0s.9042311.0.0.11064c4dECFnYK) for the D1 mini, so the relevant component on the ESPHome site would be [this one](https://esphome.io/components/sensor/dht.html).

**Description of problem:**
The D1 Mini can't get a reading from the DHT sensor. Both temperature and humidity consistently read as a 'nan' state.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: thermometer
  platform: ESP8266
  board: d1_mini

sensor:
  - platform: dht
    model: dht22_type2
    pin: D4
    temperature:
      name: "Temperatuur"
    humidity:
      name: "Luchtvochtigheid"
    update_interval: 60s

```

**Logs (if applicable):**

```
[02:17:30][I][app:105]: ESPHome version 1.18.0 compiled on May 28 2021, 01:05:14

[02:17:30][C][logger:189]: Logger:
[02:17:30][C][logger:190]:   Level: VERY_VERBOSE
[02:17:30][C][logger:191]:   Log Baud Rate: 115200
[02:17:30][C][logger:192]:   Hardware UART: UART0
[02:17:30][C][dht:017]: DHT:
[02:17:30][C][dht:018]:   Pin: GPIO2 (Mode: INPUT)
[02:17:30][C][dht:024]:   Model: DHT22 (or equivalent)
[02:17:30][C][dht:027]:   Update Interval: 60.0s
[02:17:30][C][dht:029]:   Temperature 'Temperatuur'
[02:17:30][C][dht:029]:     Device Class: 'temperature'
[02:17:30][C][dht:029]:     Unit of Measurement: '°C'
[02:17:30][C][dht:029]:     Accuracy Decimals: 1
[02:17:30][C][dht:030]:   Humidity 'Luchtvochtigheid'
[02:17:30][C][dht:030]:     Device Class: 'humidity'
[02:17:30][C][dht:030]:     Unit of Measurement: '%'
[02:17:30][C][dht:030]:     Accuracy Decimals: 0

[02:17:35][VV][scheduler:152]: Running interval 'update' with interval=60000 last_execution=4264293 (now=4324293)
[02:17:35][W][dht:162]: Requesting data from DHT failed!
[02:17:35][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[02:17:35][V][sensor:013]: 'Temperatuur': Received new state nan
[02:17:35][D][sensor:099]: 'Temperatuur': Sending state nan °C with 1 decimals of accuracy
[02:17:35][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 882975355
  state: nan
  missing_state: NO
}
[02:17:35][V][sensor:013]: 'Luchtvochtigheid': Received new state nan
[02:17:35][D][sensor:099]: 'Luchtvochtigheid': Sending state nan % with 0 decimals of accuracy
[02:17:35][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 3170564867
  state: nan
  missing_state: NO
}

```

**Additional information and things you've tried:**

Assuming the DHT shield might be faulty somehow, I have replaced it with another of the same type. This yielded the exact same result as the first.

Having read the note on the DHT sensor component page, and also [this PR and its comments](https://github.com/esphome/esphome/pull/926), I have tried different model values. These were the auto-detect, DHT22, AM2302 and DHT22_Type2. All gave the same results.

Of course the obvious things to consider are the connection between the D1 Mini and the DHT sensor, and the pull-up resistor. the resistor is part of the DHT shield, specifically made to be used with the D1 mini, so I have to assume that's good. The 'wiring' consists of the supplied pins soldered on to the boards (male on the DHT shield, long pin female on the D1 mini). I've had others doublecheck the soldering, all concur it looks fine.

The configured pin number (D4) is definitely correct, as that pin is marked on the board, the board specifically states pin D4 is the one to use, and also D4 is the only pin that has a trace going to it.

I've run out of variables to consider, as far as I can tell. What's going wrong here? Any help would be appreciated.

agners, 2021-06-11:
To me it seems that the ESP tries to initiate communication but the DHT22 does not answer.  The message `Requesting data from DHT failed!` indicates that the pin just stays high (the DHT does not initiate communication).

From the picture it does seem that D4 is the right pin. Probably obvious, but make sure to match the D4 of the shield with the D4 of the D1.

Other than that, I don't have an idea right now.

agners, 2021-06-11:
I don't think it makes a difference but you can try to use the GPIO number directly (GPIO2 in your case).

Bartdmm, 2021-06-13:
Thank you, @agners, for your analysis and suggestion. It is good advice to make sure to match up the ports on the boards, and I have indeed made sure to check this repeatedly; they match.

As you suspected already, changing the pin identification to GPIO2 explicitly, rather than through the D4 alias, unfortunately made no difference.

Could the design of the DHT shield simply be incorrect? It seems unlikely, given how simple they are, but signs do seem to point in that direction.

Fusseldieb, 2021-06-14:
I've personally had sensors that were labeled wrong. Black was positive and red negative, but only ONE sensor. Clearly wrong. Connected them backwards as a last resort and it started working.

Bartdmm, 2021-06-16:
I found out that one of the sensors in the batch I'd ordered is different. The plastic housing for the sensor itself is of a much poorer finish, but the PCB itself looks much better made. So I've tried that one.. and it does work.

So I guess the conclusion should be that I have bought a batch of faulty sensor shields. That'll teach me for trying to do it on the cheap.

Thanks for the help and suggestions. I think we can close this issue now.

closed 2021-06-16 by Bartdmm