esphome/issues #2080: D1 Mini Pro fails OTA uploads with Program failed with code 1

mloebl, 2021-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**  D1 Mini Pro

**ESPHome version (latest production, beta, dev branch)** v1.18.0

**Affected component:**

OTA 

**Description of problem:**
D1 Mini Pro fails OTA uploads gets to 100%, then fails with `Program failed with code 1` .  No problems uploading via USB

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  disp_name: airquality03
  mqtt_broker: !secret mqtt_broker
  mqtt_username: !secret esphome_airquality03_username
  mqtt_password: !secret esphome_airquality03_password

esphome:
  name: ${disp_name}
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: !secret esphome_wifi_ssid
  password: !secret esphome_wifi_password
  domain: !secret esphome_domain

# Enable logging
logger:
  baud_rate: 0
#  level: very_verbose

# Enable Home Assistant API
api:
  password: !secret esphome_airquality03_password

ota:
  password: !secret esphome_airquality03_password

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

status_led:
  pin: GPIO2

```

**Logs (if applicable):**
```
INFO Reading configuration /config/esphome/airquality03.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/airquality03
Processing airquality03 (board: d1_mini_pro; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/airquality03/.pioenvs/airquality03/firmware.elf
Checking size /data/airquality03/.pioenvs/airquality03/firmware.elf
RAM:   [====      ]  42.6% (used 34860 bytes from 81920 bytes)
Flash: [===       ]  34.8% (used 363212 bytes from 1044464 bytes)
========================= [SUCCESS] Took 1.24 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of airquality03
INFO  -> 192.168.5.194
INFO Uploading /data/airquality03/.pioenvs/airquality03/firmware.bin (367360 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.
```

**Additional information and things you've tried:**

I've removed all sensor info down to just the status LED and Wifi, not sure what's causing it.  This board on Amazon that I purchased, https://www.amazon.com/gp/product/B08HH96RJY , an off brand.

mloebl, 2021-09-27:
No change with Current version: 2021.9.1

nuttytree, 2021-10-06:
I have seen this issue also, changing the board type to d1_mini resolved the issue, not sure if that change has any other negative affects but it worked for me.

mloebl, 2021-10-06:
@nuttytree  Thank you! That worked for me, guessing coming from upstream them from platformio or, something in this clone maybe isn't a true d1_mini_pro.? I can close for now as a pretty reasonable work around, thanks again!