esphome/issues #2081: First measurement using dallas component is too early (before wifi & mqtt is connected)

TobbeJ, 2021-05-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome as addon to home assistant that is running as a vm using the official vm image.

**ESP (ESP32/ESP8266, Board/Sonoff):**

sparkfun esp32 thing

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

dallas & deep sleep

**Description of problem:**
I'm working on getting a esp32 setup to run off batteries using deep sleep most of the time.
To get this working it is best if the device can do one measurement then right away go to sleep.
This part i got mostly figured out.

The issue i have is that the first temperature measurement of my ds18b20 happens a bit too early, before wifi and mqtt is connected so i am forced to wait one full polling intervall to complete the first working measurement and publish to mqtt.
This can also easily be observed in the log too

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Logs (if applicable):**

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[I][logger:170]: Log initialized
[C][status_led:014]: Setting up Status LED...
[C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[D][binary_sensor:036]: 'greenhouse temp published': Sending state OFF
[D][binary_sensor:036]: 'greenhouse rssi published': Sending state OFF
[D][binary_sensor:034]: 'greenhouse status': Sending initial state OFF
[C][dallas.sensor:032]: Setting up DallasComponent...
[C][wifi:033]: Setting up WiFi...
[D][wifi:324]: Starting scan...
[D][dallas.sensor:153]: 'greenhouse Temperature #1': Got Temperature=25.4°C
[D][sensor:099]: 'greenhouse Temperature #1': Sending state 25.43750 °C with 1 decimals of accuracy
[D][wifi:339]: Found networks:
[I][wifi:385]: - 'xxxxxxx' (xx:xx:xx:xx:xx:xx) ▂▄▆█
[D][wifi:386]:     Channel: 10
[D][wifi:387]:     RSSI: -44 dB
[I][wifi:194]: WiFi Connecting to 'xxxxxxx'...
[I][wifi:457]: WiFi Connected!
[C][wifi:303]:   SSID: 'xxxxxxx'
[C][wifi:304]:   IP Address: 192.168.x.x
[C][wifi:306]:   BSSID: xx:xx:xx:xx:xx:xx
[C][wifi:307]:   Hostname: 'test1'
[C][wifi:311]:   Signal strength: -45 dB ▂▄▆█
[C][wifi:315]:   Channel: 10
[C][wifi:316]:   Subnet: 255.255.255.0
[C][wifi:317]:   Gateway: 192.168.x.x
[C][wifi:318]:   DNS1: 192.168.x.x
[C][wifi:319]:   DNS2: 0.0.0.0
[D][wifi:466]: Disabling AP...
[C][ota:029]: Over-The-Air Updates:
[C][ota:030]:   Address: x.y.z:3232
[C][ota:032]:   Using Password.
[C][api:022]: Setting up Home Assistant API server...
[C][mqtt:025]: Setting up MQTT...
[D][mqtt:111]: Resolving MQTT broker IP address...
[D][mqtt:144]: Resolved broker IP address to x.x.x.x
[I][mqtt:170]: Connecting to MQTT...
[I][mqtt:210]: MQTT Connected!
[I][app:059]: setup() finished successfully!
[D][sensor:099]: 'greenhouse WiFi RSSI': Sending state -42.00000 dBm with 0 decimals of accuracy
[D][binary_sensor:036]: 'greenhouse rssi published': Sending state ON
[D][binary_sensor:036]: 'greenhouse status': Sending state ON
[I][app:105]: ESPHome version 1.18.0 compiled on May 29 2021, 17:15:20
```
Note that the first temp reading is retrieved long before wifi and mqtt is connected.
Can something be done to fix this?

**Additional information and things you've tried:**

glmnet, 2021-05-29:
What you really want to do?

glmnet, 2021-05-29:
There is no on connected for Mqtt but you could poll https://esphome.io/components/mqtt.html#mqtt-connected-condition then invoke update on Dallas so you get a reading as soon as Mqtt connects then go to sleep or whatever

TobbeJ, 2021-05-29:
Take exactly one measurement of 2 or more sensors and then go to sleep as soon as they have been sent to mqtt.
Something i have figured out except that the dallas temp sensors first measurement happens way too soon before any kind of network is up and running so that measurement is never sent over mqtt which means i must wait one complete polling intervall for it.
and that means wasting time keeping the device awake (on battery)

TobbeJ, 2021-06-05:
Let me explain a bit more how i set things up.

Each sensor have a matching binary template sensor that defaults to OFF.
Then each sensor got an on_value trigger that checks if mqtt is connected and if so, sets its template sensor to ON and executes a script.
the script then checks if ALL of the binary template sensors is set to ON and if so, go to sleep for some time.
the point of all this is to get exactly one reading from each sensor and the immediately go to sleep to not waste any time on battery.
also i have a debug output in the script just before it goes to sleep that prints out how long it was awake for.

this all works well, except that first read of the dallas temp sensor happens long before wifi is connected and mqtt is connected (see long posted earlier).
this results in a full polling intervals delay of the temperature reading and in this case (if i leave the defaults alone) 60 seconds of delay where device must be awake.

yes as a workaround i can reduce the update_intervall of the dallas temp sensor but when you have multiple sensors and an mqtt server that i cant just spam data to it gets a bit tricky.

best would be if first sensor readings is done after wifi is up.

glmnet, 2021-06-05:
Your logic don't seem fine to me. 

You're trying that the Dallas update when it is convenient for you but that's is very arbitrary, it's not intuitive when it really should happens. Anybody else can come with a similar issue ask for something slightly different and that would break your setup. 

If you want to have full control of when it updates you can disable automatic updates by setting update interval to never then use the component.update action.

closed 2021-10-17 by stale[bot]