esphome/issues #2083: BLE_client arg handling typo

gitlemat, 2021-05-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.): PIP over OS X**

**ESP (ESP32/ESP8266, Board/Sonoff): ESP32**

**ESPHome version (latest production, beta, dev branch): v18.0**

**Affected component: BLE_client**

**Description of problem:**
The event handling routine has apparently typo and evaluates gattc_if instead of esp_gattc_if. This generates that the some events are digested for all the instantiations and not just the correct one.

```
  if (event != ESP_GATTC_REG_EVT && esp_gattc_if != ESP_GATT_IF_NONE && gattc_if != this-&gt;gattc_if)    return;
```
I think should be:
```
  if (event != ESP_GATTC_REG_EVT && esp_gattc_if != ESP_GATT_IF_NONE && esp_gattc_if != this-&gt;gattc_if)    return;
```

**Problem-relevant YAML-configuration entries:**
```yaml
ble_client:
  - mac_address: ff:ff:c0:00:60:6b
    id: BLE_Luz_Azotea3
  - mac_address: ff:ff:ab:0b:f1:da 
    id: BLE_Luz_Azotea4
  
light:
  - platform: ble_triones_light
    ble_client_id: BLE_Luz_Azotea3
    color_interlock: true
    name: Luz_BLE_Atico_3
    id: Luz_BLE_Atico_3
    internal: false
    default_transition_length: 0s
  - platform: ble_triones_light
    ble_client_id: BLE_Luz_Azotea4
    color_interlock: true
    name: Luz_BLE_Atico_4
    id: Luz_BLE_Atico_4
    internal: false
    default_transition_length: 0s

```

**Logs (if applicable):**

```
No descriptor found for notify of handle 0x03

```

**Additional information and things you've tried:**

With the proposed change it works correctly.

torfbolt, 2021-06-01:
@jesserockz is this fix valid?

jesserockz, 2021-06-01:
Yes, I have created a PR above. Thanks

closed 2021-06-01 by jesserockz