esphome/issues #2084: Inkbird IBS-TH1 Mini integration not reading values

haplm, 2021-05-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io, latest version

**ESP (ESP32/ESP8266, Board/Sonoff):**

Couple different boards built around ESP32-WROOM

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

https://esphome.io/components/sensor/inkbird_ibsth1_mini.html

**Description of problem:**
Inkbird sensor configured in YAML, MAC address located and correctly set in the configuration. I can see 'sps' device being picked-up by the esp32_ble_tracker, but no values are read. I assume that the problem lays in the fact that the 'Manufacturer specific' response looks different compared to the description in the code, which is:
// 5) the manufacturer datas should contain a 16-bit uuid amd a 7-byte data vector  
// 6) the 7-byte data component should have data[2] == 0 and data[6] == 8

This is what I'm getting when I scan public beacons from my phone:
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: hall_bt_scanner
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "yyy"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hall Bt Scanner Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:

sensor:
  - platform: inkbird_ibsth1_mini
    mac_address: 10:08:2C:1F:38:6D
    temperature:
      name: "Inkbird Mini Temperature"
    humidity:
      name: "Inkbird Mini Humidity"
    battery_level:
      name: "Inkbird Mini Battery Level"
```

**Logs (if applicable):**

```
[14:45:07][I][app:105]: ESPHome version 1.18.0 compiled on May 30 2021, 13:36:48
[14:45:07][C][wifi:443]: WiFi:
[14:45:07][C][wifi:303]:   SSID: [redacted]
[14:45:07][C][wifi:304]:   IP Address: 192.168.1.157
[14:45:07][C][wifi:306]:   BSSID: [redacted]
[14:45:07][C][wifi:307]:   Hostname: 'hall_bt_scanner'
[14:45:07][C][wifi:311]:   Signal strength: -48 dB ▂▄▆█
[14:45:07][C][wifi:315]:   Channel: 3
[14:45:07][C][wifi:316]:   Subnet: 255.255.255.0
[14:45:07][C][wifi:317]:   Gateway: 192.168.1.1
[14:45:07][C][wifi:318]:   DNS1: 192.168.1.1
[14:45:07][C][wifi:319]:   DNS2: 0.0.0.0
[14:45:07][C][logger:189]: Logger:
[14:45:07][C][logger:190]:   Level: DEBUG
[14:45:07][C][logger:191]:   Log Baud Rate: 115200
[14:45:07][C][logger:192]:   Hardware UART: UART0
[14:45:07][C][esp32_ble_tracker:606]: BLE Tracker:
[14:45:07][C][esp32_ble_tracker:607]:   Scan Duration: 300 s
[14:45:07][C][esp32_ble_tracker:608]:   Scan Interval: 320.0 ms
[14:45:07][C][esp32_ble_tracker:609]:   Scan Window: 30.0 ms
[14:45:07][C][esp32_ble_tracker:610]:   Scan Type: ACTIVE
[14:45:07][C][inkbird_ibsth1_mini:012]: Inkbird IBS TH1 MINI
[14:45:07][C][inkbird_ibsth1_mini:013]:   Temperature 'Inkbird Mini Temperature'
[14:45:07][C][inkbird_ibsth1_mini:013]:     Device Class: 'temperature'
[14:45:07][C][inkbird_ibsth1_mini:013]:     Unit of Measurement: '°C'
[14:45:07][C][inkbird_ibsth1_mini:013]:     Accuracy Decimals: 1
[14:45:07][C][inkbird_ibsth1_mini:014]:   Humidity 'Inkbird Mini Humidity'
[14:45:07][C][inkbird_ibsth1_mini:014]:     Device Class: 'humidity'
[14:45:07][C][inkbird_ibsth1_mini:014]:     Unit of Measurement: '%'
[14:45:07][C][inkbird_ibsth1_mini:014]:     Accuracy Decimals: 1
[14:45:07][C][inkbird_ibsth1_mini:015]:   Battery Level 'Inkbird Mini Battery Level'
[14:45:07][C][inkbird_ibsth1_mini:015]:     Device Class: 'battery'
[14:45:07][C][inkbird_ibsth1_mini:015]:     Unit of Measurement: '%'
[14:45:07][C][inkbird_ibsth1_mini:015]:     Accuracy Decimals: 0
[14:45:07][C][captive_portal:169]: Captive Portal:
[14:45:07][C][ota:029]: Over-The-Air Updates:
[14:45:07][C][ota:030]:   Address: hall_bt_scanner.local:3232
[14:45:07][C][api:095]: API Server:
[14:45:07][C][api:096]:   Address: hall_bt_scanner.local:6053
```

**Additional information and things you've tried:**

Wubarius, 2021-06-04:
Hi,

I have the same problem with my esp32 and Inkbird.
I noticed that it works without an external sensor cable on the Inkbird.
[12:22:11][D][sensor:099]: 'Temperature': Sending state 27.34000 °C with 1 decimals of accuracy
[12:22:11][D][sensor:099]: 'Luftfeuchtigkeit': Sending state 60.00000 % with 1 decimals of accuracy
[12:22:11][D][sensor:099]: 'Batterie Sensor': Sending state 94.00000 % with 0 decimals of accuracy

As soon as I plug in the temperature sensor again, no more data comes in. 
It just comes:
[12:21:07][D][esp32_ble_tracker:620]: Found device 49:42:06:00:43:E3 RSSI=-71
[12:21:07][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:21:07][D][esp32_ble_tracker:643]:   Name: 'sps'

Need help ...

haplm, 2021-06-07:
I can confirm this. When I unplug the external sensor, I'll get the readings. Apparently with the sensor plugged in the order of the bytes changes... I'm happy to help to debug this.

nuttytree, 2021-10-05:
I seem to be having issues getting data with or without the external sensor connected.  A little debugging shows get_manufacturer_datas() consistently returning an empty vector.  I have confirmed the data is being transmitted to the app on my phone so not sure why no data is getting picked up by ESPHome.  @fkirill what version of the firmware is on the devices you used when writing this integration?  My devices have version 1.7.0, wondering if something has changed in newer firmware versions.

nuttytree, 2021-10-06:
So after much trial and error I determined that the reason I wasn't getting any data was because of the scan parameters I had set trying to get the connection to HA stable.  I am now getting data but unfortunately my HA connection is back to dropping for a couple of seconds every few minutes.  Good news for everyone else is that while I don't plan to use the external sensors I should be able to get a PR up in the next day or 2 that will allow these to work with or without the external sensor.

nuttytree, 2021-10-06:
So apparently I grabbed an old version of the code for this when I was troubleshooting and this has already been resolved by esphome/esphome#1983.

fkirill, 2021-10-10:
Hey guys, I don't have a model with an external sensor. So while I understand the difficulty, I cannot really help with fixing it. Can someone with the device and programming skills have a look at it?

fkirill, 2021-10-10:
As for HA connectivity, this particular piece of code we are talking about only has to do with decoding BlueTooth signals related to the Inkbird device on ESP32 and converting them to the temperature, humidity, and battery sensor levels that can be understood by HA.
Connectivity to HA itself is a completely separate and unrelated topic and probably deserves an issue of its own.

fkirill, 2021-11-06:
Hey guys, looking at the code, where this message comes from https://github.com/esphome/esphome/blob/d536509a63189490f314e5923471828ca28ba26d/esphome/components/inkbird_ibsth1_mini/inkbird_ibsth1_mini.cpp#L33

It appears that it is not a bug. It only means that we're trying to parse the Bluetooth data that is coming from some other device (Not Inkbird) and shouldn't be processed by the Inkbird code anyway.
The logging itself is only printed at a VV logging level (super-verbose) and is only really useful for debugging.

In the process of setting up the device, you will have to establish the MAC address and set it in the yaml configuration.
https://esphome.io/components/sensor/inkbird_ibsth1_mini.html#setting-up-devices

It stands to reason that Inkbird code ignores all other BLE signals from all other devices (with different MAC addresses).

Please double-check that you set up your address properly.

Thanks.

vwt12eh8, 2022-01-11:
In my case, it seems that this one line is causing the value to fail to be retrieved.  https://github.com/esphome/esphome/blob/d536509a63189490f314e5923471828ca28ba26d/esphome/components/inkbird_ibsth1_mini/inkbird_ibsth1_mini.cpp#L58

The condition in the code is 8, but my device advertises 6.

fkirill, 2022-04-22:
So the idea is that I wanted to have as specific as possible to avoid misreading the data from the wrong device.

Can you test it locally if you change condition to `((mnf_data.data[6] == 6)  || (mnf_data.data[6] == 8))`?