esphome/issues #2085: Invalid Configuration For Deta-Grid-Connect-Smart-Fan-Speed-Controller-with-Touch-Light-Switch #101

rowie1, 2021-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS Raspberry Pi4 : core-2021.5.5

**ESP (ESP32/ESP8266, Board/Sonoff):**
 ESP8266 Deta-Grid-Connect-Smart-Fan-Speed-Controller-with-Touch-Light-Switch

**ESPHome version (latest production, beta, dev branch)**
 ESPHome version is the latest 1.18.0

**Affected component:**
 https://www.esphome-devices.com/devices/Deta-Grid-Connect-Smart-Fan-Speed-Controller-with-Touch-Light-Switch/

**Description of problem:**

I have a known working .yaml configuration file for this device, however when I copy and paste the known working file in to ESPHome in Homeassistant it is recognised as being invalid and I'm unable to compile and upload the .bin file to the fan/light controller.
I can upload the same config file to the device using the web server and the device functions as intended.
The issue ESPHome has with the config file is the fan speeds: LOW,MEDIUM & HIGH. Image attached showing the error returned from these values:

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

jkelly73, 2021-07-24:
I get this same error too.  Looking for some advice how to solve please.

Thx
Jon

rowie1, 2021-07-24:
Hi Jon

ESPHome doesn’t recognise low, med & high anymore. Replace each of these entries with the equivalent percentage value (33,66,100) in your configuration then it will validate.

Hope this helps

Cheers 

Adrian

jkelly73, 2021-07-25:
Thanks Adrian, changing the low, medium and high for 33, 66, & 100 worked like a charm!

My question now is how do you change the fan speed?  The fan ON/OFF and speed buttons doesn’t seem to behave as I would expect.  I can’t select High (100), or Medium (66), only LOW (33).  I suspect the problem lies in the timing return state routine associated with the fan_speed, here:
  - platform: template
    id: fan_speed
    type: float
    write_action:
      - then:
        - if:
            condition:
              lambda: return ((state > 0) && (state < .4));
            then:
  
As I am not a programmer and only a newbie at Homeassistant, I am in no way certain of my diagnostics!!

Any help would be greatly appreciated.

Thanks
Jon

rowie1, 2021-07-25:
Have you controlled the fan in HA yet or just attempted to operate using the physical buttons on the controller?
I think from memory I had issues with the physical buttons until I activated the fan speeds in HA.  Following this they began working correctly on the controller.

FYI I use an ifan03 custom card in Lovelace for my fans, it works well.

Cheers 

Adrian

jkelly73, 2021-07-25:
Well how strange! The buttons did not work as expected (as I mentioned above)... I have installed the ifan03 custom card - that works excellently and performs exactly as I expected / wanted and now the buttons work correctly also!?!?!?!

Just one small problem now, 2 of the 4 fan icons are missing from the lovelace card, although they are present in the 'static' folder????

[image]

Any thoughts?
Thanks
Jon

jkelly73, 2021-07-25:
> Well how strange! The buttons did not work as expected (as I mentioned above)... I have installed the ifan03 custom card - that works excellently and performs exactly as I expected / wanted and now the buttons work correctly also!?!?!?!
> 
> Just one small problem now, 2 of the 4 fan icons are missing from the lovelace card, although they are present in the 'static' folder????
> 
> [image]
> 
> Any thoughts?
> Thanks
> Jon
> 
> Update...
> Ive looked in the ifan03-card.js file and I suspect there are inconsistencies between "low" and "slow.svg" and similarly with "high" and "fast.svg".

Fixed it...
I had to rename the two svg files from slow.svg to low.svg and fast.svg to high.svg and then remove the resource and re-add it.  Simply 'update'ing the resource had no affect.
[image]

rowie1, 2021-07-25:
Yeah cool, I had to do something like that too, just couldn’t quite remember what I did. I also changed the speeds to low, med, high. I reckon I just opened the json file & edited all of  the normal & fast entries to be med & high.

closed 2021-12-07 by github-actions[bot]