esphome/issues #2086: logger component can not set log level from within ESPHome packages

lanrat, 2021-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker, but should not matter.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

logger
https://esphome.io/components/logger.html

**Description of problem:**
When using packages, it seems that I'm unable to set the log level for a specific component from within a package.

**Problem-relevant YAML-configuration entries:**
ble_package.yaml
```yaml
logger:
  logs:
    esp32_ble_tracker: INFO

esp32_ble_tracker:
```

test_node.yaml
```yaml
substitutions:
  node_name: ble

packages:
  ble: !include ble_package.yaml

esphome:
  name: ${node_name}
  platform: ESP32
  board: nodemcu-32s

```

**Logs (if applicable):**

All `DEBUG` logs entries from the `esp32_ble_tracker` are printed.

**Additional information and things you've tried:**
If I add the `logger` declaration to the main yaml (`test_node.yaml`) then the logger only logs `INFO` or higher information for `esp32_ble_tracker` as expected.

glmnet, 2021-06-02:
Can you add the output of command esphome <file> config

lanrat, 2021-06-02:
```shell
$ esphome test_node.yaml config
INFO Reading configuration test_node.yaml...
INFO Configuration is valid!
substitutions:
  node_name: ble
esphome:
  name: ble
  platform: ESP32
  board: nodemcu-32s
  arduino_version: platformio/espressif32@3.0.0
  build_path: ble
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
logger:
  logs:
    esp32_ble_tracker: INFO
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  level: DEBUG
esp32_ble_tracker:
  scan_parameters:
    duration: 5min
    interval: 320ms
    window: 30ms
    active: true
```

glmnet, 2021-06-02:
I can't reproduce this issue, the generated main.cpp is the same in both cases

lanrat, 2021-06-09:
I did some more debugging and I think I found the real underlying issue.
If the `logger:` is defined in one package, and a specific scope for it is defined in another, then they don't get merged.
Here is a better example:

test_node.yaml
```yaml
substitutions:
  node_name: ble-test

packages:
  base: !include base_package.yaml
  ble: !include ble_package.yaml

esphome:
  name: ${node_name}
  platform: ESP32
  board: nodemcu-32s
```

base_package.yaml 
```yamp

ble_package.yaml
```yaml
logger:
  logs:
    esp32_ble_tracker: INFO

esp32_ble_tracker:
```

lanrat, 2021-06-09:
I did some more debugging and I think I found the real underlying issue.
If the `logger:` is defined in one package, and a specific scope for it is defined in another, then they are not merged corectly.

Here is a better example with 2 packages:

test_node.yaml
```yaml
substitutions:
  node_name: ble-test

packages:
  base: !include base_package.yaml
  ble: !include ble_package.yaml

esphome:
  name: ${node_name}
  platform: ESP32
  board: nodemcu-32s
```

base_package.yaml 
```yaml
logger:
```

ble_package.yaml
```yaml
logger:
  logs:
    esp32_ble_tracker: INFO

esp32_ble_tracker:
```

Output of `esphome test_node.yaml config`
```yaml
INFO Reading configuration test_node.yaml...
INFO Configuration is valid!
substitutions:
  node_name: ble-test
esphome:
  name: ble-test
  platform: ESP32
  board: nodemcu-32s
  arduino_version: platformio/espressif32@3.0.0
  build_path: ble-test
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
logger:
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  level: DEBUG
  logs: {}
esp32_ble_tracker:
  scan_parameters:
    duration: 5min
    interval: 320ms
    window: 30ms
    active: true
```

Is there a correct way to do this?

michaelblight, 2022-05-31:
This is still an issue. Furthermore, if you put 'level' and/or 'logs' inside the included package file, you get an IDE error stating:
```
Component not found: level.
Component not found: logs.
```
This is when using:
```
packages:
  wifi: !include include/wifi.yaml
  device_base: !include include/m5stamp-base-esp-idf.yaml
  logger: !include include/logger.yaml
```
Where `logger.yaml` contains:
```
level: DEBUG
logs:
  esp32_ble_tracker: INFO
```