esphome/issues #2088: Docs Table of Contents is not scrollable

Chupaka, 2021-06-02:
**Affected component:**

Doc pages with many sections. For example: https://esphome.io/devices/sonoff.html

**Description of problem:**
Table of Contents in docs is not scrollable, so even having browser window maximized I can't reach out to the bottom items.

**Additional information and things you've tried:**
I can repeat it in Chrome and Safari browsers on my laptop.

On mobile device, ToC is rendered on the top of page, so it's not a problem there.

Chupaka, 2021-10-01:
The activity (as the issue is still relevant)

nepozs, 2021-11-29:
It is really important for all using low resolution displays - eg. for my laptop everything below Light Components is unavailable!
[image]

slovdahl, 2021-12-03:
FWIW, also happening on Firefox on (on macOS).

slovdahl, 2021-12-03:
This SO answer seems to have the solution: https://stackoverflow.com/a/57040610/1283519

closed 2021-12-06 by oxan