esphome/issues #2091: LCTech Dual Relay

KimHEF, 2021-06-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Hello i tried the whole day to get my LCTech Dual Relay running. But it doesn't work...
The Logger shows: 

[21:58:10][D][switch:021]: 'Relay_1' Turning ON.
[21:58:10][D][switch:045]: 'Relay_1': Sending state ON

The Board is in the Red-Light-Mode... i tried two different types.
One with GPIO and one with UART...

`esphome:
  name: esp-relay
  platform: ESP8266
  board: esp01

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

captive_portal:

# Enable Home Assistant API
api:
  password: !secret password

# Enable logging
logger:

switch:
  - platform: gpio
    pin: GPIO0
    name: "Relay_1"
    
  - platform: gpio
    pin: GPIO2
    name: "Relay_2"`

KimHEF, 2021-06-03:
> **Operating environment/Installation (Hass.io/Docker/pip/etc.):**
>
> **ESP (ESP32/ESP8266, Board/Sonoff):**
>
> **ESPHome version (latest production, beta, dev branch)**
>
> **Affected component:** [...]

Fusseldieb, 2021-06-14:
> The Board is in the Red-Light-Mode...

What do you mean by that?

Note that using GPIO0 is used to enter flash mode on ESPs. Try using other pins, if possible.