esphome/issues #2095: HTU21D returns humidity above 100%

LuckyTriple7, 2021-06-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASSOS, HASS 2021.06.02

**ESP8266 NodeMCU v3**

**ESPHome version: V1.18**

**Affected component: HTU21D Temperature+Humidity Sensor**

**Description of problem:**

HTU21D reports Humidity above 100%. 
These are unrealistic values.
Maximum humidity is 100%

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
```
sensor:
  - platform: htu21d
    temperature:
      icon: mdi:thermometer
      name: Temperatur ${devicename_short}
      id: temp
    humidity:
      icon: mdi:water-percent
      name: Luftfeuchtigkeit ${devicename_short}
      id: humi
    update_interval: 60s
```

```

**Logs (if applicable):**

```
`[08:18:28][D][htu21d:052]: Got Temperature=17.1°C Humidity=101.7%`

```

**Additional information and things you've tried:**

Fusseldieb, 2021-06-14:
From https://forums.adafruit.com/viewtopic.php?f=22&t=108435:

> The higher-than-100% output you're seeing is a side effect of the humidity sensor absorbing too much water during the fog.
> 
> The sensor assumes that its sense cap will produce a specific reading at a specific temperature and humidty level.. usually 70% humidity at 25C. A sense cap that's been exposed to too much water will produce a higher reading than expected at the reference temperature and humidity levels. Exposing the sense cap to long periods of 100% humidity (like fog) will pick up that kind of offset.
> 
> You have two options:
> 
> 1) You can declare the sensor dead and get a new one.
> 
> 2) You can try to recondition the sensor by putting it in a 120C oven for about an hour to evaporate all the water stored in the sense cap, then let the sensor sit in a container at 70% humidity and 25C temperature for 24 hours.

This isn't an issue related to ESPHome.

LuckyTriple7, 2021-06-15:
Thank you for the information. the sensor is in a greenhouse. The values above 100% occur at night. When the windows are opened in the morning, the values are correct again. I have created a filter which filters out the values above 100%.

I hope the code is correct

```
      filters:
        - lambda: |-
            float MIN_VALUE = 1;
            float MAX_VALUE = 100.0;
            if (MIN_VALUE <= x && x <= MAX_VALUE) return x;
```

Fusseldieb, 2021-06-15:
Nice to know!

It will certainly help others.

If you want, you can close this issue :)

closed 2021-06-15 by LuckyTriple7