esphome/issues #2099: Pulse Meter and Binary Sensor detect pin changes, but Pulse Counter not

MiguelAngelLV, 2021-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Manjaro Linux with ESPHome by Docker.

**ESP (ESP32/ESP8266, Board/Sonoff):** 

Lolin 32
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.18
**Affected component:**

Pulse counter

**Description of problem:**

Pulse counter always return 0 pulses. But if I change for pulse_meter o by binary sensor, I can detect the pulses.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
- platform: pulse_counter
  pin: 33
  unit_of_measurement: 'kW'
  name: 'Power Meter House'
  internal_filter: 13us
  accuracy_decimals: 2
  filters:
  - multiply: 0.015

  total:
    unit_of_measurement: 'kWh'
    name: 'Energy Meter House'
    accuracy_decimals: 3
    filters:
      - multiply: 0.00025

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

[07:54:09][D][pulse_counter:159]: 'Power Meter House': Retrieved counter: 0.00 pulses/min
[07:54:10][D][sensor:099]: 'Power Meter House': Sending state 0.00000 kW with 2 decimals of accuracy
[07:54:10][D][pulse_counter:164]: 'Power Meter House': Total : 0 pulses
[07:54:10][D][sensor:099]: 'Energy Meter House': Sending state 0.00000 kWh with 3 decimals of accuracy
[07:55:09][D][pulse_counter:159]: 'Power Meter House': Retrieved counter: 0.00 pulses/min
[07:55:10][D][sensor:099]: 'Power Meter House': Sending state 0.00000 kW with 2 decimals of accuracy
[07:55:10][D][pulse_counter:164]: 'Power Meter House': Total : 0 pulses
[07:55:10][D][sensor:099]: 'Energy Meter House': Sending state 0.00000 kWh with 3 decimals of accuracy
[07:56:09][D][pulse_counter:159]: 'Power Meter House': Retrieved counter: 0.00 pulses/min
[07:56:10][D][sensor:099]: 'Power Meter House': Sending state 0.00000 kW with 2 decimals of accuracy
[07:56:10][D][pulse_counter:164]: 'Power Meter House': Total : 0 pulses
[07:56:10][D][sensor:099]: 'Energy Meter House': Sending state 0.00000 kWh with 3 decimals of accuracy

```

**Additional information and things you've tried:**

MiguelAngelLV, 2021-06-07:
I change the resistor for other and it work now... But is weird that not working before with pulse counter and but works with pulse meter...

Fusseldieb, 2021-06-14:
You have probably forgotten to pullup or even pulldown the sensor. 

Some integrations might activate the pullup/pulldown for you, others don't.

You can specify if you want pullup/pulldown when setting up the pin, refer to: https://esphome.io/components/binary_sensor/gpio.html#activating-internal-pullups

By using a resistor, like you said, you probably got enough current flowing to properly pull it up/down.

MiguelAngelLV, 2021-06-14:
Maybe.

I think that would be interesting add to doc witch integration configure pullup/pulldown and how configure manually

closed 2021-06-14 by MiguelAngelLV