esphome/issues #2101: climate_ir_lg "Light Off" changes state to Off

marciogranzotto, 2021-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome Add-On on Home Assistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCUv2 with custom PCB

**ESPHome version (latest production, beta, dev branch)**

v1.18.0

**Affected component:**

`climate_ir_lg`
https://esphome.io/components/climate/ir_climate.html#climate-ir-lg-climate

**Description of problem:**
My remote control has a "Light Off" button, which turns off the LED display of the AC unit, but it does not turn off the AC.
When pressing that button, the state on ESPHome changes to Off.
Logs below.

Also, it would be great to be able to send that command from ESPHome.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  room: bedroom
  name: "Bedroom AC"

esphome:
  name: bedroom_ac
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom Ac Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_receiver:
  id: receiver
  pin:
    number: D1
    inverted: true
  dump: raw

remote_transmitter:
  pin: D2
  carrier_duty_percent: 50%

sensor:
  - platform: homeassistant
    name: "temp sensor"
    internal: true
    id: bedroom_temperature
    entity_id: sensor.temperature_sensor

climate:
  name: "${name}"
  platform: climate_ir_lg
  receiver_id: receiver
  sensor: bedroom_temperature
  header_high: 3265us # AC Units from LG in Brazil use these timings
  header_low: 9856us
```

**Logs (if applicable):**

```
///////// Actual OFF button:
[14:59:49][D][climate.climate_ir_lg:124]: Decoded 0x88C0051
[14:59:49][D][climate:262]: 'Bedroom AC' - Sending state:
[14:59:49][D][climate:265]:   Mode: OFF
[14:59:49][D][climate:270]:   Fan Mode: HIGH
[14:59:49][D][climate:273]:   Swing Mode: OFF
[14:59:49][D][climate:276]:   Current Temperature: nan°C
[14:59:49][D][climate:282]:   Target Temperature: 21.00°C

///////// ON button:
[14:59:51][D][climate.climate_ir_lg:124]: Decoded 0x880064A
[14:59:51][D][climate:262]: 'Bedroom AC' - Sending state:
[14:59:51][D][climate:265]:   Mode: COOL
[14:59:51][D][climate:270]:   Fan Mode: HIGH
[14:59:51][D][climate:273]:   Swing Mode: OFF
[14:59:51][D][climate:276]:   Current Temperature: nan°C
[14:59:51][D][climate:282]:   Target Temperature: 21.00°C

///////// LIGHT OFF button:
[14:59:53][D][climate.climate_ir_lg:124]: Decoded 0x88C00A6
[14:59:53][D][climate:262]: 'Bedroom AC' - Sending state:
[14:59:53][D][climate:265]:   Mode: OFF
[14:59:53][D][climate:270]:   Fan Mode: HIGH
[14:59:53][D][climate:273]:   Swing Mode: OFF
[14:59:53][D][climate:276]:   Current Temperature: nan°C
[14:59:53][D][climate:282]:   Target Temperature: 21.00°C
```

**Additional information and things you've tried:**

marciogranzotto, 2021-10-16:
still an issue

jbrandek, 2023-06-15:
Same problem

Einixx, 2024-10-26:
Hi, 
same issue with Gree aircon.
Did you find a workaround?
Cheers