esphome/issues #2102: DHT Sensor Halving Temperature Value

gb53smith, 2021-06-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266, NodeMCU board
Home Assistant 2021.5.5 Add On running on Raspberry PI 3B

**ESPHome version (latest production, beta, dev branch)**

ESPHome version  v1.18
**Affected component:**

DHT/AM2302 sensor.

**Description of problem:**
Occasionally, the temperature value sent to HA is half the correct value.  To workaround I have added a lambda filter to double the values in error.

**Problem-relevant YAML-configuration entries:**
```yaml

logger:
  level: NONE

substitutions:
  node_name: plants
  device_verbose_name: "Basement Plants"
  
packages:
  nodemcu: !include common/nodemcu.yaml
  
switch:
  - platform: gpio
    pin: 14
    name: "Plant Light"
  - platform: gpio
    pin: 12
    name: "Plant Heater"
  - platform: gpio
    pin: 13
    name: "Plant Dehumidifier"

dallas:
  - pin: GPIO10
    update_interval: 10s

sensor:
  - platform: dht
    pin: 2
    temperature:
      name: "Plants Temperature"
      filters:
      - delta: 0.2
      - lambda: !lambda if ( x < 12.0 ) { return x * 2.0; } else { return x; }
    humidity:
      name: "Plants Humidity"
      filters:
      - delta: 1
    update_interval: 10s
    model: am2302
  - platform: dallas
    address: 0x1F3C01B556A19A28
    name: "Plants Soil Temperature"
    filters:
      - delta: 0.2

#### common/nodemcu.yaml
esphome:
  name: ${node_name}
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "GreenMountain"
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Plants Fallback Hotspot"
    password: !secret fallback_password

captive_portal:
 
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_password

ota:
  password: !secret ota_password
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I know its hard to believe, so here is a picture:

[image]

Originally, I thought that the logging was interfering with the measurement, but I turned logging OFF.
The humidity value from the same sensor is stable.  The temperature value comes from the third and fourth bytes transmitted from the sensor.  Looks like the temperature binary is shifted once to the right, thereby halving is value.
Occasionally, there is a checksum error in the log for the DHT sensor but these errors don't correspond with the half value errors.

This is my first project using ESP Home. I previously used  PlatformIO.  The same hardware works well with PlatformIO developed code.

gb53smith, 2021-06-08:
Another workaround, called a "max difference" filter.
```
     - lambda: |-
            float MAX_DIFFERENCE = 5.0;  // adjust this!
            static float last_value = NAN;
            if (isnan(last_value) || std::abs(x - last_value) < MAX_DIFFERENCE)
              return last_value = x;
            else
              return {};
```
I would prefer that the bug was found and fixed.

agners, 2021-06-11:
@gb53smith do you see CRC errors in the log (`Checksum invalid`..)?

The only thing I see is that the init timing for the MCU defaults to 800us, however, the datasheet of the AM2302 you are using requires 1ms according to datasheet. Can you try `DHT22_TYPE2` as model? It sets the initial timing to 2ms, which might work better.

This fix sets the MCU init timing to 1ms: https://github.com/esphome/esphome/pull/1901

gb53smith, 2021-06-11:
I saw occasional checksum errors too but they did not seem to correspond to times when the temperature values was halved.
I have implemented a custom component based on:
```
https://github.com/Diaoul/DHTNew
```
It is working so far but requires testing overnight.  Interesting learning exercise but kind of defeats the simplicity
that ESPhome is supposed to offer.
I will try the DHT22_TYPE2 solution and report back tomorrow.

PS.
I found a great variety of libraries on PlatformIO for the DHT sensor. Some are better than others, I guess.
For the DHT component,  ESPHome coders seem to have copied one library's source code 
and added more statements to integrate with  ESPHome.   This makes it difficult to switch in another DHT library.
My custom component approach allows me to try another version of a DHT library just my changing the library: option.

gb53smith, 2021-06-12:
@agners I tried using
 ```
model: DHT22_TYPE2
```
but  the issue with half temperature values is NOT fixed as seen in this picture:
[image]

I used my custom DHT22 sensor overnight without errors and switched back to the Esphome DHT sensor about 9:00 this morning.  There is something else wrong with the Esphome DHT sensor, as looking into the DHTNew library, I found that the start delay was also only 800 us.  Maybe there is some interference created by running the Dallas sensor at the same time?
Someone should look into the DHTNew code and determine if the critical serial to 5 byte conversion section can be copied to the Esphome DHT sensor.
Here is my custom sensor code:
```
esphome:
  name: plants
  platform: ESP8266
  board: nodemcuv2
  includes:
    - my_dht.h
  libraries:
    - "diaoul/DHTNew @ 1.0.0"

wifi:
  ssid: "GreenMountain"
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Plants Fallback Hotspot"
    password: !secret fallback_password

captive_portal:
 
# Enable logging
logger:
  level: NONE
  esp8266_store_log_strings_in_flash: False

# Enable Home Assistant API
api:
  password: !secret ota_password

ota:
  password: !secret ota_password
  
switch:
  - platform: gpio
    pin: 14
    name: "Plant Light"
  - platform: gpio
    pin: 12
    name: "Plant Heater"
  - platform: gpio
    pin: 13
    name: "Plant Dehumidifier"

dallas:
  - pin: GPIO10
    update_interval: 10s
    
sensor:
  - platform: custom
    lambda: |-
      auto my_dht = new MyDHT();
      App.register_component(my_dht);
      return {my_dht->temperature_sensor, my_dht->humidity_sensor};
    sensors:
    - name: "Plants Temperature"
      unit_of_measurement: °C
      accuracy_decimals: 1
      filters:
      - delta: 0.2
    - name: "Plants Humidity"
      unit_of_measurement: "%"
      filters:
      - delta: 1
  - platform: dallas
    address: 0x1F3C01B556A19A28
    name: "Plants Soil Temperature"
    filters:
      - delta: 0.2

```
```
#include "esphome.h"
#include "esphome/core/log.h"
#include "esphome/core/helpers.h"
#include "DHTNew.h"

class MyDHT : public PollingComponent, public Sensor {
 public:
  Sensor *temperature_sensor = new Sensor();
  Sensor *humidity_sensor = new Sensor();
  // constructor.  Update interval set here to 10s
  MyDHT() : PollingComponent(10000) {}

  float get_setup_priority() const override { return esphome::setup_priority::DATA; }
  // Pin and model type set here
  DHT dht_dht{2, DHT_MODEL_DHT22};
  void setup() override {
    // This will be called by App.setup()
   
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.
    float temperature = dht_dht.readTemperature();
    ESP_LOGD("custom", "The value of my temperature is: %f", temperature);
    temperature_sensor->publish_state(temperature);
    float humidity = dht_dht.readHumidity();
    ESP_LOGD("custom", "The value of my humidity is: %f", humidity);
    humidity_sensor->publish_state(humidity);
  }
};
```

agners, 2021-06-17:
> but the issue with half temperature values is NOT fixed as seen in this picture:

Hm, DHT22_TYPE2 sets the wait time to 2ms, so it seems not to be the start delay then.

> Maybe there is some interference created by running the Dallas sensor at the same time?

This should not be the case as all interrupts re locked during the period of reading the sensor. But would be cool if you could try without the Dallas sensor, just to make sure...

What pull-up values are you using?

gb53smith, 2021-06-17:
I have re-flashed without the Dallas sensor and with the DHT22_TYPE2 model.
I am using the Keyes AM2302 DHT pictured [here](http://cactus.io/hookups/sensors/temperature-humidity/dht22/hookup-arduino-to-keyes-dht22-temp-humidity-module).
It has a 4K7 pullup on board.  It is wired to an ESP8266 using 3 feet of cat-5 cable.  
cat-5 cable has 50 pF/m so the RC delay added would be only about 0.25 us.
It did not take long for the error to re-appear:
[image]

The DHT sensor has been difficult to support by Arduino.   That why there are so many versions on PlatformIO.  
I don't like how components are maintained in Esphome.  See my feature request [here](https://github.com/esphome/feature-requests/issues/1257).  The basic idea is to wrap a PlatformIO library and leave bugs fixes to the library originator.

agners, 2021-06-17:
> I don't like how components are maintained in Esphome. See my feature request here. The basic idea is to wrap a PlatformIO library and leave bugs fixes to the library originator.

Well, in ESPHome the components are tighter integrated then with PlatformIO: There is a standardized interface, e.g. sensors have a state etc, and all components adhere to it. For that to work, when you want to change the interface, all components need be adjusted. That is much simpler if you have all under a single roof, with a single repository and version number.

Furthermore, when you link multiple components in a single firmware, things my interfere with each other. ESPHome takes care of that to a certain degree. A lot of issues can come from this situation. Then you end up in the situation where the original PlatformIO library author just says: "Doesn't work with ESPHome? Well, I don't know, works here when I have it running in a standalone application"...

gb53smith, 2021-06-17:
I was just thinking that the wrapper approach worked well with my custom DHT example.  Can't you still add a standardized interface in a wrapper class?   As for interference, the DHTNew library only includes Arduino.h which hopefully can be relied on across all components.
The use of namespace should prevent other conflicts.

Take a look at the read method in ```"diaoul/DHTNew @ 1.0.0"```.  Maybe it does a better job at serial to byte conversion.  Could code fragment that be copied into the Esphome code?  It measures the pulses and blocks interrupt in a different manner.

gb53smith, 2021-06-19:
I have re-written my custom component without using the DHTNew library component.  I copied the read method from DHTNew.
This too solves my issue.  To fix the Esphome DHT component, copy the section labelled "critical code".  I also increased the start delay from 800 to 1000 us.  Probably doesn't make any difference but it conforms to the AM2302 datasheet recommendation.
My code can be found [plants.yaml](https://github.com/gb53smith/Public/blob/master/plants.yaml) and [my_dht.h](url) on GitHub.
As a bonus, I put the AM2302 module my the freezer and test the negative temperature performance.  As the photo shows my code fixes issue#1744 too.
[image]

closed 2021-06-22 by OttoWinter

agners, 2021-06-22:
@gb53smith the merged patch also changes read timing a bit. Can you try if that it makes a difference? It should be possible to test with the new `external_components` method:

```yaml
external_components:
  - source: github://esphome/esphome@dev
    components: [dht]
```

I did look at `DHTNew`, I can't really see a big difference. The library uses `pulseIn` instead of manually parsing the pulses. But otherwise a lot of things are the same (in both cases interrupts are disabled and the pin is set as input pin with pull-up). Timing is slightly different, they use 50us (vs. 40us in ESPHome) as tie breaker between a one and a zero. I created a branch, you can test it with this `external_components`:

```yaml
external_components:
  - source: github://agners/esphome@improve-dht-data-polarity-detection
    components: [dht]
```

gb53smith, 2021-06-24:
**@agners**  I tested each version for an overnight duration.

Both SOLVED this issue.  

The external_components is a great alternative to the custom component.
I plan to do an experiment starting, from the dev version, which sweeps the starting delay up and down with each sensor read.
I have a theory that any delay from 800 to 20000 us should work.
In the dev version, Otto has made a few other changes than just providing a 1000 us delay for the AM2302.  Those other changes could be the "real" solution.