esphome/issues #2103: BLE tracker race condition during OTA update

npawelek, 2021-06-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

ble_tracker

**Description of problem:**

Unable to perform an OTA update with the provided config. The stack trace is consistent and happens at different percentages every execution. Flashing via USB using the same config and version of ESPHome is successful.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: ble_tracker
  friendly_name: BLE Tracker

esphome:
  name: $device_name
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_not_ssid
  password: !secret wifi_not_password
  manual_ip:
    static_ip: 192.168.10.9
    gateway: 192.168.10.1
    subnet: 255.255.255.0
    dns1: 192.168.0.10
  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret fallback_ap_password

captive_portal:

logger:

api:
  password: !secret api_password

ota:
  password: !secret ota_password

esp32_ble_tracker:
  scan_parameters:
    window: 160ms

text_sensor:
  - platform: version
    name: ${friendly_name} ESPHome Version
    icon: "mdi:unfold-more-vertical"
    hide_timestamp: True
  - platform: template
    name:  ${friendly_name} Uptime
    id: uptime_human
    icon: mdi:clock-start

sensor:
  # Uptime sensor.
  - platform: uptime
[... 502 lines cut ...]
WARNING Decoded 0x401caeee: std::terminate() at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:112
WARNING Decoded 0x401caf9b: __cxa_throw at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:87
WARNING Decoded 0x401cb02e: operator new(unsigned int) at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/new_op.cc:54
WARNING Decoded 0x400d79f9: esphome::esp32_ble_tracker::ESP32BLETracker::gap_event_handler(esp_gap_ble_cb_event_t, esp_ble_gap_cb_param_t*) at /Users/redacted/esphome/ble_tracker/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:312
WARNING Decoded 0x4015af5d: btc_gap_ble_cb_to_app at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:1032
 (inlined by) btc_gap_ble_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gap/btc_gap_ble.c:898
WARNING Decoded 0x401560ce: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:110
WARNING Decoded 0x4008e8e1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[17:09:55]
[17:09:55]Rebooting...
[17:09:55]ets Jun  8 2016 00:22:57
[17:09:55]
[17:09:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[17:09:55]configsip: 0, SPIWP:0xee
[17:09:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[17:09:55]mode:DIO, clock div:2
[17:09:55]load:0x3fff0018,len:4
[17:09:55]load:0x3fff001c,len:1044
[17:09:55]load:0x40078000,len:8896
[17:09:55]load:0x40080400,len:5828
[17:09:55]entry 0x400806ac
```

**Additional information and things you've tried:**
N/A

Misiu, 2021-06-08:
@npawelek that's a known issue with `esp32_ble_tracker`
There is a PR that will add OTA callbacks (https://github.com/esphome/esphome/pull/1816), this way we will be able to disable `esp32_ble_tracker` when OTA is in progress.
@mmakaay please correct me if I'm wrong, but your change should help in this case.

mmakaay, 2021-06-08:
I'll correct you on that one. I tried to disable BLE with my working OTA callbacks on my dev environment, but I wasn't able to get OTA working in combination with LTE. 
I do have my hopes up for new coexist features in esp-idf v4 though, although making use of those features in the higher level libraries that  we use (ESPHome itself, Arduino) is not especially straight forward. So it won't be a snap fit solution for combining BLE with WiFi.

For now, I make use of two ESP's therefore, when I need both WiFi and BLE support, making the two ESP's communicate over for example I2C.

Misiu, 2021-06-08:
@mmakaay thank you for the comment.
I also was hoping that your PR will fix the problem once and for all.
Guess we have to wait a bit more.

npawelek, 2021-07-25:
After manually upgrading the ble_tracker past 1.18.0, OTA updates appear to be working again.

closed 2021-07-25 by npawelek