esphome/issues #2106: GPIO binary sensor filter delayed_on not triggering if time parameter > 2s

Oderik, 2021-06-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU ESP8266

**ESPHome version (latest production, beta, dev branch)**
v1.18.0

**Affected component:**
https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**
"High" time parameter value like `1min` on filter `delayed_on` prevents the GPIO binary sensor from publishing `on` / `true`. A parameter of `2s` works fine.

It is very unlikely that the filter prevents transition to `on` because of actual bouncing. With a low `delayed_on` parameter the state is stable enough so that the other parameters can be expected to work fine.

Maybe related to #1931 ?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  board: nodemcu
  on_boot:
    - lambda: |-
        id(sensor_d5).publish_state(true);
        id(sensor_d6).publish_state(true);
        id(sensor_d7).publish_state(true);

binary_sensor:
  - platform: gpio
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: True
    name: "$node_name light D5"
    id: sensor_d5
    filters:
      - delayed_on: 1min
  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
      inverted: True
    name: "$node_name light D6"
    id: sensor_d6
    filters:
      - delayed_on: 1min
  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: True
    name: "$node_name light D7"
    id: sensor_d7
    filters:
      - delayed_on: 1min
```

**Logs (if applicable):**
No applicable logs recorded.

**Additional information and things you've tried:**
Same behavior for `5min`, `1min` and `60s`. `2s` works fine!

jesserockz, 2021-06-13:
You are sure that the gpio is staying on for the whole time? If the pin flips during the time it resets the `delayed_on` timer

Oderik, 2021-06-15:
I am pretty sure. When I use no filter at all, the signal is stable (unless during a limited transition phase - that's what I need the filter for).

Please correct me if I am mistaking, but an `off` reading that would cancel the `delayed_on` would be a published value without any filters, correct? Or to put it the other way round: I assume the `delayed_on` does not react on some separate interrupt mechanism that would not be noticeable by the sensor but only on the downstream values that would lead to a publish.

As soon as I find the time I will connect one of the pins to 3V3 instead of the sensor just to make sure.

Oderik, 2021-06-18:
@jesserockz sorry for the delay. I added another gpio binary sensor configured as non inverted pullup:
```
  - platform: gpio
    pin:
      number: D2
      mode: INPUT_PULLUP
      inverted: False
    name: "$node_name light D2"
    id: sensor_d2
    filters:
      - delayed_on: 30s
```

 I did not connect anything, so the input should constantly read `on`. Instead, it constantly reads `off`.

closed 2021-11-01 by github-actions[bot]