esphome/issues #2111: Connection error with  RC522 Rfid reader

raidnet-ms, 2021-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io with latest version.
**ESP (ESP32/ESP8266, Board/Sonoff):**

Node MCU
**ESPHome version (latest production, beta, dev branch)**

v 1.18 stable
**Affected component:**

rc522

**Description of problem:**
I get an error in loop with the YAML provided in the documentation. You can see the log below.

**Problem-relevant YAML-configuration entries:**
```yaml

spi:
  clk_pin: D5
  miso_pin: D6
  mosi_pin: D7

rc522_spi: # or rc522_i2c
  cs_pin: D8
  update_interval: 1s

  on_tag:
    then:
    - logger.log: "Tag letto."
    - text_sensor.template.publish:
        id: rfid_tag
        state: "Tag letto"
    - display.page.show: nfc
    - homeassistant.tag_scanned: -|!lambda 'return x;'

```

**Logs (if applicable):**

```
[13:42:18][W][rc522:392]: Communication with the MFRC522 might be down, reset in -211
[13:42:18][I][rc522_spi:013]: SPI Setup
[13:42:18][I][rc522:288]: Soft reset...
[13:42:18][I][rc522:296]: Device online.

```

**Additional information and things you've tried:**

Tried with different pin. The harware is working properly with arduino.

glmnet, 2021-06-09:
can you try with the reset pin connected?

raidnet-ms, 2021-06-09:
Hi @glmnet , 
I just tryied with reset pin on D0 but the log is the same :-(
[https://ibb.co/z2dV1jR](url)

glmnet, 2021-06-09:
Are you sure your wiring/device is working? 

Did it work with other code already?

Did you check all your jumper wires are working?

Can you send a pic of your setup?

raidnet-ms, 2021-06-10:
Hi,
with arduino it worked properly.
This is the RC522 [module](https://it.aliexpress.com/item/32672212005.html?spm=a2g0s.9042311.0.0.27424c4d5Z6Xv8)
SDA is supposed to be SS (cs_pin) as far as I know. Correct me if I'm wrong please.

RC522  |  code             | node mcu
PIN       | pin                | pin
NAME  | name             | name
SDA     - > cs_pin           (D8 on nodemcu)
SCK      - > clk_pin         (D5 on nodemcu)
Miso     -> miso_pin      (D6 on nodemcu)
Mosi     -> mosi_pin      (D7 on nodemcu)
Rst        - > reset_pin     (D0 on nodemcu)

3.3                    ->          3.3
Gnd                  ->          ground

As far as I understand the problem is with cs_pin.

Later I will attach a picture.

I do also have a display attached using I2C on D1 and D2.

raidnet-ms, 2021-06-10:
this is the log if might help
```
[16:19:24][C][i2c:028]: I2C Bus:
[16:19:24][C][i2c:029]:   SDA Pin: GPIO4
[16:19:24][C][i2c:030]:   SCL Pin: GPIO5
[16:19:24][C][i2c:031]:   Frequency: 50000 Hz
[16:19:24][I][i2c:033]: Scanning i2c bus for active devices...
[16:19:24][I][i2c:040]: Found i2c device at address 0x3C
[16:19:24][D][homeassistant.text_sensor:016]: 'alarm_control_panel.antifurto': Got state 'disarmed'
[16:19:24][D][text_sensor:015]: 'Alarm State': Sending state 'disarmed'
[16:19:24][C][spi:081]: SPI bus:
[16:19:24][C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[16:19:24][C][spi:083]:   MISO Pin: GPIO12 (Mode: INPUT)
[16:19:24][C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[16:19:24][C][spi:085]:   Using HW SPI: YES

```

Fusseldieb, 2021-06-14:
Did you get it working?

raidnet-ms, 2021-06-15:
> Did you get it working?

Not the rc522. I was able to make some scan with the rdm6300 using uart.

Tge rc522 gives this error.

raidnet-ms, 2021-06-16:
OK solved.

had a damn PIN used in a far away included yaml :-|
sorry.
The component works great with SPI.

closed 2021-06-16 by raidnet-ms