esphome/issues #2112: Device is unstable when esp32_ble_tracker is enabled with some other components.

lanrat, 2021-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

esp32_ble_tracker
https://esphome.io/components/esp32_ble_tracker.html

**Description of problem:**

When enabling `esp32_ble_tracker` in addition to some other components, the ESP becomes unstable and crashes/reboots constantly. Sometimes it never finishes booting and gets stuck in a boot loop. It appears to be reproducible ~ 8/10 times.

I've encountered this issue when combining `esp32_ble_tracker` with `web` and `fastled_clockless`, however it does not seem to be specific to any component other than `esp32_ble_tracker`

Running `esp32_ble_tracker` on its own with no other sensors enabled at the same time sees to be stable.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ble-test
  platform: ESP32
  board: nodemcu-32s

logger:
  logs:
    esp32_ble_tracker: INFO

esp32_ble_tracker:

sensor:
  - platform: ble_rssi
    mac_address: "A4:C1:38:XX:XX:XX"
    name: "Kitchen BLE RSSI"

  - platform: atc_mithermometer
    mac_address: "A4:C1:38:XX:XX:XX"
    temperature:
      name: "Kitchen Temperature"
    humidity:
      name: "Kitchen Humidity"
    battery_level:
      name: "Kitchen LYWSD03MMC Battery Level"
    battery_voltage:
      name: "Kitchen LYWSD03MMC Battery Voltage"

  - platform: ble_rssi
    mac_address: "A4:C1:38:XX:XX:XX"
    name: "Bathroom BLE RSSI"

  - platform: atc_mithermometer
    mac_address: "A4:C1:38:XX:XX:XX"
    temperature:
      name: "Bathroom Temperature"
    humidity:
      name: "Bathroom Humidity"
    battery_level:
      name: "Bathroom LYWSD03MMC Battery Level"
    battery_voltage:
      name: "Bathroom LYWSD03MMC Battery Voltage"

  - platform: ble_rssi
[... 143 lines cut ...]
WARNING Decoded 0x400dfb7b: esphome::web_server::WebServer::sensor_json[abi:cxx11](esphome::sensor::Sensor*, float) at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/web_server/web_server.cpp:68
WARNING Decoded 0x400dfbc7: esphome::web_server::WebServer::on_sensor_update(esphome::sensor::Sensor*, float) at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/web_server/web_server.cpp:68
WARNING Decoded 0x401f01c1: std::_Function_handler<void (float), esphome::Controller::setup_controller()::{lambda(float)#2}>::_M_invoke(std::_Any_data const&, float&&) at /home/lanrat/code/esphome/test/ble-test/src/esphome/core/controller.cpp:29
 (inlined by) _M_invoke at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400de6bd: std::function<void (float)>::operator()(float) const at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/sensor/sensor.cpp:92
WARNING Decoded 0x400de749: esphome::CallbackManager<void (float)>::call(float) at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/sensor/sensor.cpp:92
 (inlined by) esphome::sensor::Sensor::internal_send_state_to_frontend(float) at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/sensor/sensor.cpp:100
WARNING Decoded 0x400de7ad: esphome::sensor::Sensor::publish_state(float) at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/sensor/sensor.cpp:92
WARNING Decoded 0x400e4271: esphome::wifi_signal::WiFiSignalSensor::update() at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/wifi_signal/wifi_signal_sensor.h:13
WARNING Decoded 0x400e4281: non-virtual thunk to esphome::wifi_signal::WiFiSignalSensor::update()
WARNING Decoded 0x401f00d9: std::_Function_handler<void (), esphome::PollingComponent::call_setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/lanrat/code/esphome/test/ble-test/src/esphome/core/component.cpp:163
 (inlined by) _M_invoke at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x40081152: std::function<void ()>::operator()() const at /home/lanrat/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
 (inlined by) esphome::Scheduler::call() at /home/lanrat/code/esphome/test/ble-test/src/esphome/core/scheduler.cpp:158
WARNING Decoded 0x400e43bd: esphome::Application::loop() at /home/lanrat/code/esphome/test/ble-test/src/esphome/core/application.cpp:70
WARNING Decoded 0x400e6bfe: loop() at /home/lanrat/code/esphome/test/ble-test/src/esphome/components/light/light_output.h:18
WARNING Decoded 0x400f4f15: loopTask(void*) at /home/lanrat/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x4008ed79: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[08:11:52]
[08:11:52]Rebooting...
[08:11:52]ets Jul 29 2019 12:21:46
```

**Additional information and things you've tried:**
I've tried multiple different ESP32 boards and all seem to have the same behavior. I've also tried adding/removing different components and the only constant to reproduce this issue seems to be `esp32_ble_tracker`.

Fusseldieb, 2021-06-14:
Afaik the ESPHome BLE is in fact unstable, it is reported on multiple threads across the Home Assistant Forum.

People were suggesting to change the interval on which the ESP scans the BLE (if I record it right). This didn't solve the issue completely, but it became better apparently.

I think the Bluetooth and Wifi stack clash onto each other and interfere with itself or something like that. Just a guess tho.

etamtlosz, 2021-06-17:
With 1.19 the issue persists. I don't know if it is just for me, but I have a feeling that before 1.18 things were better.

lanrat, 2021-06-17:
With v1.18.0 if I combined BLE with almost any other sensor he ESP would become unstable, so I've needed to dedicate a single ESP for all my BLE sensors.

with v1.19.0 enabling BLE with nothing else causes the EPS to boot loop constantly. I've had to move my BLE ESP back to v1.18.0 so that it works.

Ideally, I'd like BLE to work on all my ESP32 nodes so that I can have the EPS closest to the sensor record the data, for now with just a single node I miss some of the BLE broadcasts.

MrAlucardDante, 2021-06-21:
I have 2 ESP32 that were running fine on 1.18, I just updated to 1.19.2 and now I get the [Errno 111] Connection refused, even though I can see both connected to my wifi.

If I remove esp32_ble_tracker and all the ble sensors, the ESP32 runs fine.

Since I'm using those ESPs to connect to my bluetooth temperature sensors, they are now useless.

Maybe this is due to the fact that esp32_ble_server is now in a separate package.

Is there a way to downgrade to 1.18 ? I'm not using the HA addon, I installed esphome in a python venv. If I downgrade my install of ESPHome, will it compile the firmware in 1.18 ?

lanrat, 2021-06-27:
I tried upgrading to 1.19.4, and found the BLE stack to be more stable than 1.18.

With 1.19.4 the ESP32 crashes around every 16 hours instead of boot looping in 1.19.0, but its still nowhere near as stable as 1.18.

oetelaar24, 2021-08-06:
Hi, i'm experiencing much the same trying to combine the BLE tracker and camera on a ESP32_CAM.
I've found that there is a feature request for improving the memory usage of BLE, by switching to nimBLE. https://github.com/esphome/feature-requests/issues/810
However it seems like this is not really making much progress at the moment.

formatBCE, 2021-09-05:
Got EspHome Home Assistant add-on 2021.8.2.
BLE presence nodes, dedicated only for checking 4 BLE tags, are pretty unstable. They recover fast, but become unavailable for 1-3 seconds each minute or so.
I wonder if this is the right place to track possible fix, and if the fix is actually available.
I believe, that BT and WiFi are interferencing on ESP32, and this might cause instability.
Also I found, that nodes with enough power (e.g. connected to 2A charger) are more stable (although I still see that gaps there), than node, connected to 500 mA USB.

formatBCE, 2021-09-28:
With custom code, that does BLE scanning and sends updates to MQTT, it works flawlessly. Looks like i will have to stick with that solution instead of ESPHome, although it's not so convenient.
I transferred scanning logic with same scanning interval and window to custom ESPHome sensor, but it is still unusable - ESP32 becomes unavailable often. 
So i guess, this is something with how ESPHome is using WiFi - i suppose, it needs consistent connection to Home Assistant API, and BLE does interrupt it.

chatziko, 2022-02-05:
I'm experiencing the same issue. I tried to add `esp32_ble_tracker` to a board running [this project](https://github.com/chatziko/audioengine-a5-controller) that uses lots of different components, and I was getting restarts.

Something I noticed that might be helpful: when I was monitoring the logs over wifi, the restarts were easily reproducable. Triggering several of the board's components (eg the IR receiver) would lead to a restart after just a few seconds. But __without monitoring the logs__, the board was quite stable. Maybe the wifi activity produced by the logs is to blame.

Removing `esp32_ble_tracker` the board is very stable, no matter how many logs are produced.

aderusha, 2022-02-25:
Confirming similar issues here.  Adding `esp32_ble_tracker` to an otherwise-working configuration results in regular crashes.  The only configuration where I can get `esp32_ble_tracker` to work reliably is to have it running on its own dedicated device.

RichardvanGestel, 2022-04-06:
Confirming similar issues here, esp32_ble_tracker makes the esp32 unstable. I am using xiaomi_hhccjcy01. It worked fine until the winter. Somewhere in the winter it went wrong. Here the esp32 in HA will be red and unreachable. This is only due to esp32_ble track. I have tried several esp32 boards.

RichardvanGestel, 2022-04-06:
power_save_mode: none is nog compatible with esp_ble_tracker.... 
[image]

MrAlucardDante, 2022-04-06:
> power_save_mode: none is nog compatible with esp_ble_tracker....
> [image]

Yeah, it was compatible on 1.18 but it was removed on 1.19, like mentioned in the issue I’ve linked

mclever, 2022-04-15:
i cant figure out how to get my esp32 to work with ble tracker esphome either. whenever its enabled the wifi just keeps cutting out, lasting for a few pings and then off for a few. read one article and they pointed to cheap hardware. aren't these all cheap hardware!!! going to try a different board and see if i get better luck. I use the same boards for ble using aduino https://peyanski.com/home-assistant-switchbot-mqtt-esp32-integration/ and they have worked great.

formatBCE, 2022-04-15:
> i cant figure out how to get my esp32 to work with ble tracker esphome either. whenever its enabled the wifi just keeps cutting out, lasting for a few pings and then off for a few. read one article [...]

I used 3 different chips. Same result. And all of them work for same purpose (BLE tracking + mqtt) successfully with custom-built sketch.

efaden, 2022-04-15:
> 

Can you post one of your sketches?  I'm having the same issue as the above.  Without ble everything works perfectly, once enabled the WiFi becomes totally unstable.  Tried with both Arduino and IDF frameworks.

formatBCE, 2022-04-15:
> > 
> 
> Can you post one of your sketches?  I'm having the same issue as the above.  Without ble everything works perfectly, once enabled the WiFi becomes totally unstable.  Tried with both Arduino and IDF frameworks.

Unfortunately, that file was lost a week ago, when my hard drive suddenly burned to dead. I didn't consider it very important, so I don't think there's backup for it.
But I will check.

formatBCE, 2022-04-16:
Nah, I don't have the sketch, unfortunately.
Will have to re-create it. 
Shouldn't be too hard, as on top of it I made BLE gateway for Chinese Airbnk smart locks - it's far more complex stuff.

RichardvanGestel, 2022-04-21:
Also with esphome version 2022.4.0 the problem is not solved. When ESP32-BLE-TRACKER: is added, the esp32 stops working.

RichardvanGestel, 2022-04-21:
a miracle?  I've been running a copy for a few hours.. with the xiaomi-flower.. to be continued

etamtlosz, 2022-04-22:
I have a feeling that it depends of the device (not ESP32) and type of data used. For example I was trying to log my OralB toothbrush usage, and every time I was brushing my teeth, the esp32 just crashed/rebooted/became unresponsive.

RichardvanGestel, 2022-04-22:
15 uur heeft het apparaat prima gewerkt. Toen heb ik het apparaat uit- en aangezet. De gegevens, temperatuur, vochtgehalte etc. komen weer binnen... Maar het apparaat is offline in HA ...  Foutmelding: "WARNING Can't connect to ESPHome API for esp32-xiaomi.local: Error resolving IP address: [Errno -2] Name or service not known" Via wifi kan ik dus niet meer bij het apparaatje. 
Als ik opnieuw installeer met behulp van ESPHome-Flasher dan werkt het allemaal weer...

RichardvanGestel, 2022-04-22:
No, unfortunately... I can't get it stable anymore...
I'm going on holiday for a few days now... are you looking for a solution in the meantime?

mclever, 2022-04-22:
I ended up using openmqttgateway and it works great https://docs.openmqttgateway.com/

RichardvanGestel, 2022-04-26:
> I ended up using openmqttgateway and it works great https://docs.openmqttgateway.com/

Well, that's a great tip. It works perfectly!  Thank you! Thank you! Thank you!

mgrl, 2022-04-30:
Have the same issue as described here. Anybody found a workaround?

formatBCE, 2022-05-06:
Right, OpenMQTTGateway does work for me too. However, for now i still use my own firmware for it - too much to change with new firmware.

RichardvanGestel, 2022-05-18:
After the update to ESPHome version: 2022.5.0 I can no longer reproduce the problems.
HHCCJCY01 MiFlora works as before.

RichardvanGestel, 2022-05-18:
yes, here it works!

TheHomieFox, 2022-05-18:
Sorry to say it does not change anything to me. MiFlora devices work well (as before) but BLE presence turns to Unavailable and comes back randomly. Pinging the host results in lost packets quite frequently as well, so no changes at all with 2022.5.0.

rohrsh, 2022-09-10:
> 

I have similar times: maybe 5 minutes for either on to off or for off to on. Is there any sort of SLA or timeout for this?

formatBCE, 2022-09-10:
I created my own integration for BLE presence. Can be set up as custom ESPHome component, or via custom firmware flashing.
 Works with rooms, creates proper device_tracker entities, and requires beacon MAC or UUID.
- detecting "home" time: under 5 seconds
- detecting "not_home: adjustable from 1 to 10 minutes
- additional artificial combined device_trackers in case of need to rely on more that one tracker (e.g. keys and wallet) with adjustable logic (and/or).
Check out:
- HA part (via HACS): https://github.com/formatBCE/Format-BLE-Tracker
- ESP part: https://github.com/formatBCE/ESP32_BLE_presense

trunneml, 2022-09-25:
Having the same issue here. I have the feeling that it's an out of memory problem.
When I add a lot of string based components on my NSPanel it even won't boot. If i reduce components the device is getting more stable. When I add components with a lot of string depending lambdas it dies when HA connection begins.

pavelw, 2023-05-23:
Esphome Version: 2023.4.3
using 2 stock xiaomi_lywsd03mmc sensors,
wifi enabled

it crashed regularly after around 500 seconds.

WORKAROUND for me was to disable the logger:

```
logger:
  level: NONE
```

andrewjswan, 2023-08-03:
Maybe it's really worth switching to a new BLE stack and solving all the current BT problems? Do not keep a permanent connection to the device (then we can bypass the limitation of 3 connections), but connect "on demand", well, or depending on the settings. Reduce memory consumption, improve stability?
There are more and more devices, and many BLU devices require an active connection, but the current stack is more of a lottery than a stable operation.
Here are some good reviews about NimBLE and apparently there is a version for both Arduino and ESP-IDF.
https://github.com/esphome/feature-requests/issues/810

evoncken, 2024-06-23:
The issue still persists with ESPhome 2024.6.1 - it appears to have actually regressed from 2024.5.x.

I have multiple Panasonic airco’s with an esp32 control board. Today, after updating these devices to ESPhome 2024.6.1, they became unresponsive. The only change in the sketch was the new “platform: esphome” option for “ota:”. The devices all had an uptime of approx. 14 days before the update.

Cleaned all build files, installed and the device went offline almost immediately. Had to use a usb cable to reflash (without esp32_ble_tracker). The issue is 90-100% reproducible. Sometimes it takes longer before hanging.

It really seems like something changed between ESPhome 2024.5.x and 2024.6.x that is causing severe instability.

jrentsch, 2024-08-12:
@evoncken I thought I was going crazy, but I have been having the same problems since June as well. I have multiple Inkbird and Govee sensors that are now barely reachable. The only way new values are scanned is if each ESP32 reboots. I have tried every configuration I could find and nothing seems to be the fix.

At a loss of where to go.

MrAlucardDante, 2024-08-12:
> @evoncken I thought I was going crazy, but I have been having the same problems since June as well. I have multiple Inkbird and Govee sensors that are now barely reachable. The only way new values [...]

I have noticed that too lately. I have 2 ESP32 that pull data from xiaomi miija thermometers. Only one of them stops tracking data after some time. Rebooting it fixex the issue.

They both have the same config apart from a neopixelbus led strip on the "faulty" one

poseidogithub, 2024-09-22:
same problem here, as soon as I load BLE_TRACKER component... systems become unestable. easiet way to check, just launch a continous ping to the device and load the webpage... hit reload, and you lose a couple of pings (device rebooting)