esphome/issues #2113: Pulse_counter unable to use IO extender pins

MichaelM100, 2021-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS 5.13
supervisor-2021.05.4
core-2021.6.2

running on a vmware virtual server

**ESP (ESP32/ESP8266, Board/Sonoff):**
wemos D1 Mini and D1 mini pro

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
The issue is defining the pin for the pulse_counter. How do I setup a the pulse_counter to use a pin on a mcp23017 (I2C) IO extender.
Worked this through on the community forum (Topic = Pulse-counter on MCP23017 io extender) - recommended I try here

--- Error --- when validating
sensor.pulse_counter: [source /config/esphome/mcp2307.yaml:282]
  platform: pulse_counter
  pin:  [source /config/esphome/mcp2307.yaml:284]
    
    [mcp23017] is an invalid option for [pin]. Please check the indentation.
    mcp23017: mcp23017_z1 [source /config/esphome/mcp2307.yaml:284]
    number: 15
    mode: INPUT
    inverted: False
  name: pulse_counter_mcp
  id: pulse_counter_mcp
  force_update: False
  update_interval: 5s
  icon: mdi:pulse
  unit_of_measurement: pulses/min
  count_mode: 
    rising_edge: INCREMENT
    falling_edge: DISABLE
  accuracy_decimals: 2
  internal_filter: 13us

```

**Problem-relevant YAML-configuration entries:**
```
----  pulse_counter setup -----
sensor:  
# Waterflow pulse counter
  - platform: pulse_counter
    pin:
      mcp23017: mcp23017_z1
      number: 15
      mode: INPUT
      inverted: false
    name: "pulse_counter_mcp"
    id: "pulse_counter_mcp"
    force_update: false
    update_interval: 5s
    icon: mdi:pulse
    unit_of_measurement: pulses/min
    count_mode:
      rising_edge: INCREMENT
      falling_edge: DISABLE
    accuracy_decimals: 2
    internal_filter: 13us

----  MCP23017 setup ----

i2c:
  - id: bus_a
    # sda: GPI04
    # scl: GPI05
    scan: True

mcp23017:
  - id: mcp23017_z1
    address: 0x27

--- Binary_sensor setup that works 
--- I used the same format for the pulse_counter to use a pin on the
--- IO extender but get the above error

  - platform: gpio
    name: "MCP23017_z1_14"
    id: mcp23017_z1_14
    pin:
      mcp23017: mcp23017_z1
      number: 14
      mode: INPUT
      inverted: false
```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->

```
--- Error --- when validating
sensor.pulse_counter: [source /config/esphome/mcp2307.yaml:282]
  platform: pulse_counter
  pin:  [source /config/esphome/mcp2307.yaml:284]
    
    [mcp23017] is an invalid option for [pin]. Please check the indentation.
    mcp23017: mcp23017_z1 [source /config/esphome/mcp2307.yaml:284]
    number: 15
    mode: INPUT
    inverted: False
  name: pulse_counter_mcp
  id: pulse_counter_mcp
  force_update: False
  update_interval: 5s
  icon: mdi:pulse
  unit_of_measurement: pulses/min
  count_mode: 
    rising_edge: INCREMENT
    falling_edge: DISABLE
  accuracy_decimals: 2
  internal_filter: 13us

```

**Additional information and things you've tried:**

we tried 
mcp23xxx: mcp23017_z1

when this line is commented out the code validates fine but the pulse-counter is not using the designated pin on the io extender.

When trying to define the IO extender pin to use the code does not validate because the pulse-counter pin definition does not support the following line

mcp23017: mcp23017_z1

as is supported in the pin definition of a binary_sensor.

In short the pulse-counter pin definition needs to be extended to support pulse-counters connected to IO extenders similar to the way binary_sensors do.

maybe valid to do similar exercise for all devices that have a pin definition.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

glmnet, 2021-06-09:
That's actually not possible. 

Counting pulses at high speed requieres either specialized hardware (a function on the chip itself, like the esp32 has built in) or at least interrupt support on the  io pin. The mcp cannot do that hence ESPHome validation prevents you to trying to do it.

jesserockz, 2021-06-10:
Well, mcp23017 does actually support interrupts, and can notify esphome now, but would still need a bit of internal work for it to work with the pulse counter.

So right now you cant do this. And the validation on the pins is there so that users don't flash something and then create an issue saying its not counting correctly.

MichaelM100, 2021-06-10:
Great - thank you for the in-sight. May just up date the pulse-counter docs that due to technical constraints the pin needs to be defined on the MCU.

We can close this.

closed 2021-06-10 by jesserockz