esphome/issues #2114: ESPHome v1.8.0 requires Python >3.7, but running v3.6.9? HassOS states v3.8.9 installed.

DayRunner131, 2021-06-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant running on HassOS

## System Health

version | core-2021.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.8.9
os_name | Linux
os_version | 5.4.83-v8
arch | aarch64
timezone | America/New_York

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.12.4
Stage | running
Available Repositories | 890
Installed Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 5.13
-- | --
update_channel | stable
supervisor_version | supervisor-2021.05.4
docker_version | 19.03.15
disk_total | 219.4 GB
disk_used | 25.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Duck DNS (1.12.5), File editor (5.3.1), Mosquitto broker (6.0.1), NGINX Home Assistant SSL proxy (3.0.1), ESPHome (dev) (dev), Terminal & SSH (9.1.3), SQLite Web (3.1.0), deCONZ (6.9.0), Z-Wave JS to MQTT (0.18.1), Log Viewer (0.11.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 3
views | 5
mode | storage

</details>

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**
Attempting to update from v1.17.0 -> v1.18.0

**Affected component:**

**Description of problem:**

Attempting to update add-on ESPHome Dev v1.18.0. In the logs, it states that it requires Python >v3.7, <v4.0 and that the current installation is v.3.6.9. Will not install at the moment due to this failed test.

Looking at the configuration info, it appears Python v3.8.9 is installed through HassOS.

**Problem-relevant YAML-configuration entries:**
```N/A

```

**Logs (if applicable):**

```
[14:51:27] INFO: Installing esphome version 'v1.18.0' (https://github.com/esphome/esphome/archive/v1.18.0.zip)... 
Collecting https://github.com/esphome/esphome/archive/v1.18.0.zip 
Downloading https://github.com/esphome/esphome/archive/v1.18.0.zip (2.1MB) esphome requires Python '>=3.7,<4.0' but the running Python is 3.6.9 
[14:51:35] FATAL: Failed installing esphome pinned version.

Supervisor Logs:
21-06-09 18:42:38 WARNING (SyncWorker_3) [supervisor.docker.interface] Can't grep logs from 15ef4d2f/aarch64-addon-esphome-dev: 409 Client Error for http+docker://localhost/v1.40/containers/97d0190f2d1aae7d8a74f5a85903116d5bc72c679913b2df0570b261e4a1f946/logs?stderr=1&stdout=1&timestamps=0&follow=0&tail=100: Conflict ("can not get logs from container which is dead or marked for removal")
21-06-09 18:42:38 INFO (SyncWorker_6) [supervisor.docker.addon] Starting Docker add-on 15ef4d2f/aarch64-addon-esphome-dev with version dev
21-06-09 18:42:39 INFO (MainThread) [supervisor.api.middleware_security] /host/info access from 15ef4d2f_esphome-dev
21-06-09 18:43:59 INFO (SyncWorker_4) [supervisor.docker.interface] Stopping addon_15ef4d2f_esphome-dev application
21-06-09 18:44:00 INFO (SyncWorker_4) [supervisor.docker.interface] Cleaning addon_15ef4d2f_esphome-dev application

```

**Additional information and things you've tried:**

Tried host reboots, no affect. Opened HA Supervisor bug report, sent to report bug here.
https://github.com/home-assistant/supervisor/issues/2945

glmnet, 2021-06-09:
Did you try rebuilding the ESPHome container?

DayRunner131, 2021-06-09:
> Did you try rebuilding the ESPHome container?

Not sure if I follow. This is an HA add-on.

glmnet, 2021-06-09:
[image]

Rebuild button here

DayRunner131, 2021-06-09:
That worked! Awesome and thank you.

closed 2021-06-09 by DayRunner131