esphome/issues #2115: White channel behaviour for lights is inconsistent and broken

oxan, 2021-06-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** n/a
**ESP (ESP32/ESP8266, Board/Sonoff):** n/a
**ESPHome version (latest production, beta, dev branch):** dev
**Affected component:** light core, https://esphome.io/components/light/index.html
**Description of problem:**

I'm looking into some of the issues with the white channel in the light core component (#1086, #1475, #1943, #2040, #2046, esphome/feature-requests#663, esphome/esphome#1728, esphome/esphome#1518), and adding support for HA's new color modes to the ESPHome integration, and it turns out to be a complete mess. Before diving into fixing this, I'd like some input on the architecture we'd like to have.

In short, the current problems are:
* For addressable lights, the white channel value [isn't scaled by the brightness value](https://github.com/esphome/esphome/blob/360effcb72eefd9eb05060506b84d315125b5138/esphome/components/light/addressable_light.h#L68-L70), but for non-addressable lights [it is scaled](https://github.com/esphome/esphome/blob/360effcb72eefd9eb05060506b84d315125b5138/esphome/components/light/light_color_values.h#L196).
* When setting the RGB channels and the white channel, the values are normalized to the max of all 4 channels. This means that if the color is set to all zeros (the only way to turn on only the white channel), the magnitude of the white channel is ignored. It also means that if the white channel has a higher magnitude than the color channels, this will dim the color (which it won't do if the white channel isn't present).
* The color normalization means that ESPHome currently completely ignores the "color brightness" slider that is presented in the HA UI.
* When the color interlock is enabled, the white channel control in HA becomes totally useless and is just a binary switch between white/color (but presented as a slider). 

Personally, I think the best option is to adapt to HA's model (a master brightness slider that controls both color and white channels, a normalized color picker, and independent color/white brightness sliders). In practice this would mean that we drop the color normalization (so the color brightness is encoded in the RGB channels), and we always scale all channels by the master brightness. For lights with the color interlock enabled, we can present them as RGB (not RGBW/RGBWW) lights in HA, and switch to the white channel when the color is set to white. This eliminates the independent color/white brightness sliders, which don't make sense with the interlock.

Obviously this will be a breaking change, but I don't see a way to fix this mess without a breaking change one way or the other.

Tagging some people that might have input: @OttoWinter @jesserockz @frenck

OttoWinter, 2021-06-10:
The current model (sans newer stuff like color interlock) _were_ based on the HA model :P just HA some time ago updated the light model :) Particularly the `white_value` model is _exactly_ because back then there was no "color brightness" - if brightness was set to 0% in the frontend the light would turn of (iirc), so we were forced to define `brightness` = color brightness so that you _could_ set "only white lights on".

> Personally, I think the best option is to adapt to HA's model (a master brightness slider that controls both color and white channels, a normalized color picker, and independent color/white [...]

I don't see how you come to that conclusion - as you said, HA contains:

- master brightness slider
- _normalized_ color picker
- color brightness slider
- white brightness slider

so based on that the logical representation would be `(master_brightness,r,g,b,color_brightness,white_brightness)` where:

- `r_channel = master_brightness * color_brightness * r`, ...
- `white_channel = master_brightness * white_brightness`

The major advantage of having color normalized is that you can treat (color) brightness as an additional property, and can for example easily transition based on that property.

I'm open to changing it, for the rgb encoding though I don't really see why having brightness encoded in color is a good thing (IMHO it _feels_ cleaner to have brightness and color be separate things)

oxan, 2021-06-10:
> The current model (sans newer stuff like color interlock) were based on the HA model :P

Ah, that could be, I'm not too familar with the old HA model. 

> based on that the logical representation would be (master_brightness,r,g,b,color_brightness,white_brightness) where

Yeah, that's a sensible representation too. I was inclined to encode brightness in RGB to keep (API) changes to a minimum (and it would probably simplify the code), but this works too and I agree it's probably cleaner. 

However, the new HA color model doesn't provide normalized color + color brightness, just color with brightness encoded into it. But that's easy to calculate back of course.

oxan, 2021-06-11:
> so we were forced to define `brightness` = color brightness so that you _could_ set "only white lights on".

Note btw that this is not how ESPHome currently (since esphome/esphome@11069085) works for non-addressable lights: for those brightness is a "master brightness" that also [applies to the white channel](https://github.com/esphome/esphome/blob/360effcb72eefd9eb05060506b84d315125b5138/esphome/components/light/light_color_values.h#L196).

OttoWinter, 2021-06-14:
> Note btw that this is not how ESPHome currently (since esphome/esphome@1106908) works for non-addressable lights

Yeah that's probably a bug. Anyway, thanks for working on this!

> Yeah, that's a sensible representation too. I was inclined to encode brightness in RGB to keep (API) changes to a minimum (and it would probably simplify the code), but this works too and I agree it's probably cleaner.

I actually think it would be less of a breaking change that way - the current `brightness` would effectively be the new `color_brightness`, and `master_brightness` would be a new property. Then the protobuf wire format would even be 100% compatible because we can just rename the `brightness` field to `color_brightness` without changing semantics.

OttoWinter, 2021-06-14:
(Related: at some point, we also need to rethink the `color_temperature` model. I see HA has changed the model around that, and by the looks of it is semi-incomaptible with our current one; but I have to say the docs there are not so good concering what RGBWW/color_temp modes really mean etc; but that's for another time to figure out :smiley:  )

oxan, 2021-06-14:
> Yeah that's probably a bug. Anyway, thanks for working on this!

Yeah, I agree that PR probably shouldn't have been merged, though I can understand the confusion - it took me a while to understand the current model as well. Once we've settled on something consistent, I'll write documentation for it too, so hopefully we can avoid this in the future. 

> I actually think it would be less of a breaking change that way - the current `brightness` would effectively be the new `color_brightness`, and `master_brightness` would be a new property. Then the [...]

I think that's just a choice of whether we want to change behaviour for addressable or non-addressable lights. Currently `brightness` acts as `master_brightness` for non-addressable lights and `color_brightness` for addressable lights. I personally don't care either way, though I do like having `master_brightness` explicitly named that way.

> (Related: at some point, we also need to rethink the `color_temperature` model. I see HA has changed the model around that, and by the looks of it is semi-incomaptible with our current one; but I [...]

Yeah, we need to look at that too, though I think changes there are smaller in scope. As I understand it at the moment (but I'm not 100% sure), for non-RGB lights there's still only a color temperature input in HA, so that's fine. For RGBWW lights, you either have RGB picker + color brightness + cold white + warm white input, or a color temperature input. The second one is fine with our current model, but the first one might need some changes, as ESPHome doesn't support explicitly setting CW and WW. We might be able to get away with just converting the CW+WW input to a brightness and temperature, but we've to make sure that gives sensible behaviour with and without the constant-brightness mode (I haven't thought about that yet).

OttoWinter, 2021-06-14:
Oh wait, I was confused. I thought the current `brightness` is effectively `color_brightness`, but as you said (except for addressable lights) it's effectively `master_brightness`. Sorry for the confusion.

OttoWinter, 2021-06-14:
> We might be able to get away with just converting the CW+WW input to a brightness and temperature, but we've to make sure that gives sensible behaviour with and without the constant-brightness mode (I haven't thought about that yet).

I'm not a fan of that approach to be honest. It feels very hacky and will start to break down as you said with constant-brightness mode (or for any light output that chooses to interpret color_temperature a different way, like maybe converting it to RGB values to change the tone like with night light feature in many devices these days). Then a user would _expect_ the cw/ww leds in an RGBWW string to change with the slider in the HA frontend, but in reality the conversion to color_temp+conversion back in ESPHome leads to issues. Especially for light where things are complicated enough, I really don't want hackish approaches.

oxan, 2021-06-14:
> Oh wait, I was confused. I thought the current `brightness` is effectively `color_brightness`, but as you said (except for addressable lights) it's effectively `master_brightness`. Sorry for the confusion.

Yeah, it is confusing ;) Especially since the current `brightness` _was_ effectively `color_brightness` pre-v1.15. 

Anyway, what has your preference: making current `brightness` always be `master_brightness` and adding `color_brightness` (changing behaviour for addressable lights), or making current `brightness` always be `color_brightness` and adding `master_brightness` (changing behaviour for non-addressable lights)? 

> I'm not a fan of that approach to be honest. It feels very hacky and will start to break down as you said with constant-brightness mode (or for any light output that chooses to interpret [...]

Yeah, I agree it's probably better to expose CW+WW directly. That should be a purely additive change though, no need to change any current behaviour.

emontnemery, 2021-06-22:
One comment on this:
> the logical representation would be (master_brightness,r,g,b,color_brightness,white_brightness)

Home assistant's UI shows a color picker for RGBW lights with the following controls:
- Master brightness
- Hue+Saturation wheel
- Color brightness
- White brightness

However, the output of that is not `(master_brightness,r,g,b,color_brightness,white_brightness)`, the output is `(master_brightness,r,g,b,white)`
 
Similarly, home assistant's UI shows a color picker for RGBWW lights with the following controls:
- Master brightness
- Hue+Saturation wheel
- Color brightness
- Color temperature slider
- White brightness

The output of that `(master_brightness,r,g,b,cold_white,warm_white)`

This is explained here, https://developers.home-assistant.io/docs/core/entity/light#color-modes, although the documentation can probably be improved.

OttoWinter, 2021-06-22:
> However, the output of that is not (master_brightness,r,g,b,color_brightness,white_brightness), the output is (master_brightness,r,g,b,white)

Yes the backend gets passed the data that way. My thought is this: It makes sense to separate color from brightness (IMHO), for example this is exactly why the frontend exposes color hue/sat separately from color brightness.

Plus using a new property would be way less of a breaking change, as the color normalization with actions for example wouldn't change. And the HA side can always extract brightness from the rgb value it gets passed.

> Similarly, home assistant's UI shows a color picker for RGBWW lights with the following controls:

If I'm reading the frontend code here right, I don't think that's the case. See below there are separate sliders for cold/white brightness. 

https://github.com/home-assistant/frontend/blob/4fbc31d0b06d4d7c794381d09b17205748a2a01b/src/dialogs/more-info/controls/more-info-light.ts#L178-L203

Only if the light is in `color_temp` mode does a color temperature slider get shown (and it looks like that slider is mutually exclusive with color select, so you can _either_ have color temp selection _or_ color hue/sat).

emontnemery, 2021-06-22:
You're absolutely right about colour picker for RGBWW-lights, I updated my comment.

> My thought is this: It makes sense to separate color from brightness (IMHO), for example this is exactly why the frontend exposes color hue/sat separately from color brightness.

Sure, it's fine to normalize the color and send the color brightness separately. However, it's slightly weird IMHO to use RGB for that since it does hold brightness information, HSV is a more fitting model.

If you enable the `demo` integration you will get multiple types of lights,  including RGBW and RGBWW lights.

> Only if the light is in color_temp mode does a color temperature slider get shown (and it looks like that slider is mutually exclusive with color select, so you can either have color temp selection or color hue/sat).

Yeah, that's right.

oxan, 2021-06-22:
There's one compelling reason to separate color from color brightness information: if a light is dimmed to low levels, and then turned up again, it doesn't lose its color. 

> Sure, it's fine to normalize the color and send the color brightness separately. However, it's slightly weird IMHO to use RGB for that since it does hold brightness information, HSV is a more fitting model.

HSV probably _is_ a slightly more fitting model, but I don't think it's worth making the switch, given that everything we interact with uses RGB.

oxan, 2021-06-24:
With esphome/esphome#1895 almost done, I see two major issues in the light platform remaining. These are not really related to the white channel issues I started this issue with, but since we already started discussion about this above, let's continue here. 

**Color interlock**
The UI/UX in HA for RGBW[W] lights with a color interlock is (at least) counterintuitive. What happens on the ESPHome currently makes sense imo: if you request white, the color leds are disabled; if you request a color, white is disabled; if you request both, you get white (for historic reasons). This last part is the problem: HA presents these lights with a color picker, color brightness and white brightness slider, and always sends the full state (brightness + color + white), so you need to set the white slider to zero to actually get a color to show up.

Ideally, HA would have some notion of the color interlock and allow the user to choose between "color" and "white" mode. I haven't dived too deep into the color modes (maybe @emontnemery can correct me if I'm wrong), but it seems that if the light is labeled as supporting `COLOR_MODE_RGB` and `COLOR_MODE_WHITE` (or `COLOR_MODE_COLOR_TEMP` for RGBWW) we get exactly that behaviour. 

**Color temperature**
For RGBWW lights, HA gives two options: either a color temperature slider, or a color picker + color brightness + cold white + warm white slider†. The color temperature slider is pretty straightforward (I _think_ the idea is that RGB leds should be turned off when this is used), but the CW+WW sliders are problematic for ESPHome right now -- we don't support directly setting them at all.

This is even more complicated by the `constant_brightness` option ESPHome has, which limits the sum of CW+WW brightness to 100%. There's also the problem that the current light model in ESPHome supports lights that are RGB lights with a single color temperature channel (i.e. that don't know any notion of CW/WW lights at all). Controlling the white channel with CW+WW values is problematic for them, we'd need to convert that back to a color temperature at some point. However, no such lights are currently implemented in ESPHome.

† I find this an interesting choice -- why not allow setting the color temperature alongside RGB, or setting the CW+WW channels separately from the RGB channels (okay, you can do that by setting color brightness to zero, but that's not really convenient)? Anyway, not really relevant here.

**Solutions**
I see the following possible solutions to these problems:

1. Expose the `color_interlock` and `constant_brightness` setting in the ESPHome API, implement the color-mode logic in HA, and convert CW+WW to white+CT in HA before giving it to ESPHome. This is certainly the solution with the least changes in ESPHome, but as discussed above it's a bit hacky and requires duplication of logic in HA.
2. Expose the `color_interlock` setting in the ESPHome API, and implement all color-mode logic in HA. Add an option to set the CW and WW channels explicitly to the ESPHome API, and implement conversion logic to white+CT in ESPHome (probably in `LightCall` validation). This is less hacky as all logic is grouped together in ESPHome and there's no duplication, but still has the problem that user input is converted from CW+WW to white+CT and back for CW+WW lights. 
3. Expose the `color_interlock` setting in the ESPHome API, and implement all color-mode logic in HA. Add an option to set the CW and WW channels explicitly to the ESPHome API, and store them as CW and WW values in ESPHome. Convert all calls that set a white+CT to CW+WW values in ESPHome. This is basically the same as the previous option, but does a roundtrip conversion for lights with a white+CT output instead. 
4. Expose the `color_interlock` setting in the ESPHome API, and implement all color-mode logic in HA. Add an option to set the CW and WW channels explicitly to the ESPHome API, and store them as CW and WW values in ESPHome, along with the white+CT values currently stored. Let the light choose from which representation it reads. Implement conversion when setting either representation to write the other representation. This has the advantage that there is never roundtrip conversion, but state is duplicated (just two floats though). 
5. Implement the same color modes as HA in ESPHome. Lights have a list of supported color modes, calls can change the color mode a light is in, and the representation depends on the currently active color mode. This will consolidate all logic in ESPHome, but won't be a perfect match for all our outputs (e.g. the CWWW output will be in white+CT mode and need to convert). It will also be a pretty significant refactor of ESPHome to achieve, and that'll probably break both forward and backward compatibility (i.e. ESPHome and HA need to be upgraded in lockstep). 
6. Implement our own color mode system. Color modes would be white, CT, CWWW, RGB, RGBW, RGBWW. Same advantages/disadvantages as previous option, except this would be a match for all outputs.
7. Anything else I didn't think of?

I think my personal preference goes to option 4. 1 is easy but it's hacky and will probably require changes again when HA adds some features. 2 and 3 will inevitably lead to roundtrip conversions for some lights. 5 and 6 will arguably clean up some parts of the codebase, but it's a lot of changes and having different representations at different times will complicate stuff as well.

I'd appreciate some input.

OttoWinter, 2021-06-24:
Personally, I'm most in favor of 5/6. I really don't like exposing `color_interlock` & especially `constant_brightness` to HA, because then we'll be duplicating the (rather complicated) color interlock logic everywhere.

IMO, `color_interlock` should really just should be a concern of the light output, because it's a logic attribute rather than describing "what this entity can do". Color mode is quite a bit better there, because then the light output can say "I support these two modes, you can switch between them".

Another thing I'm unhappy about with our current API is this: the way `LightColorValues` is converted to `rgb` etc is all defined in `LightColorValues`; so in effect we get all sorts of logic thrown into `LightColorValues` (like `constant_brightness`) to make it work with all use cases. Instead, what _should_ happen is that the `LightOutput`s just get passed the `LightColorValues` and they're in charge of converting it to the physical representation (kind of like HA). So if you have some super complex formula how color_temperature gets converted to the cwww channels, you can implement it.

> I find this an interesting choice -- why not allow setting the color temperature alongside RGB, or setting the CW+WW channels separately from the RGB channels 

Yeah, I also find that very confusing. Why have it be so different for RGBWW and CWWW when they're really just the same thing.

> that'll probably break both forward and backward compatibility

I don't think this has to be true. We already know what API version the remote peer is using, and so for example HA can convert the old-style to color_mode, same for the other direction (but other direction may not be necessary, in practice the HA side is almost always more up to date - has worked for example already with covers).

> It will also be a pretty significant refactor of ESPHome to achieve

Probably true, but I have a feeling it might not be that bad. I can spare a little time for it this weekend, and even more the one after that.

oxan, 2021-06-24:
> Personally, I'm most in favor of 5/6. I really don't like exposing `color_interlock` & especially `constant_brightness` to HA, because then we'll be duplicating the (rather complicated) color interlock logic everywhere.

We don't have to implement the whole color interlock logic in HA in 1/2/3/4, only change the supported color modes based on whether it's enabled/disabled. The state change and validation is still handled by ESPHome, and that's where all the complications are.

> IMO, `color_interlock` should really just should be a concern of the light output, because it's a logic attribute rather than describing "what this entity can do". Color mode is quite a bit better there, because then the light output can say "I support these two modes, you can switch between them".

Well, `color_interlock` can never be _only_ handled in the light output, because it has ramifications all the way up the chain to how the UI is displayed. So you need to expose that information in some way. I agree that color modes are a more generic concept to do that, but it also expands the problem space: instead of only dealing with an interlock between RGB and white, you can have all kinds of color mode combinations. Just looking at the HA light frontend dialog, it's littered with conditions like this: `supportsTemp && ((!supportsColor && !supportsWhite) || this._mode === LightColorModes.COLOR_TEMP)`.

`color_brightness` on the other hand really is purely an output concern, and as long as the light gets a CW+WW value it can do the calculation itself.

> Another thing I'm unhappy about with our current API is this: the way `LightColorValues` is converted to `rgb` etc is all defined in `LightColorValues`; so in effect we get all sorts of logic thrown [...]

That's currently already possible, as `LightOutput.write_state` gets passed a `LightState` and with it the `LightColorValues`. For example addressable lights already don't use these methods. The only reason for these methods in `LightColorValues` is that multiple outputs use the same calculation, so they're deduplicated there. We can move them to a separate helper class and clean-up their parameters somewhat, but I don't see how you can get around having these calculations implemented somewhere.

There's a small advantage to having them in one place as well, it makes sure they're consistent. If different lights implement the calculation themselves, that creates the opportunity for them to behave differently (as we saw with addressable and non-addressable lights interpreting `brightness` differently). 

> I don't think this has to be true. We already know what API version the remote peer is using, and so for example HA can convert the old-style to color_mode, same for the other direction (but other [...]

True, it can be kept compatible, but for that we'll need to implement lots of conversion logic. In my experience that kind of logic generally isn't well-tested and bitrots quickly (though if it's just for compatibility reasons it might not matter much).

I also think we need both directions: newer HA with old ESPHome will need to convert from color modes to the old-style, while ESPHome itself will need to convert from old-style to color modes, as users will have lots of old-style `light.turn_on` actions in their config. I don't think forcing everyone to update their YAML is a good idea.

> Probably true, but I have a feeling it might not be that bad. I can spare a little time for it this weekend, and even more the one after that.

If you've got time I'll be curious to see what you come up with ;)

OttoWinter, 2021-06-24:
> instead of only dealing with an interlock between RGB and white, you can have all kinds of color mode combinations.

not sure I'm following here. there's only one active color mode at a time, depending on the color mode different properties can/can't be set. So if I'm not missing something obvious it should really just be "is this property allowed for this color mode", and would even clean up the `LightCall` logic.

> while ESPHome itself will need to convert from old-style to color modes, as users will have lots of old-style light.turn_on actions in their config. 

Yes, there are `light.turn_on` actions, but my feeling is that almost all of them will not need updating. Most of the time you just say "turn this light on/off/toggle" without many parameters. Next step would be "set this RGB light to this color", but only a small minority have something that would have breaking changes.

----

btw, while we have you here. Could I ask you some things about the HA Light Entity architecture, @emontnemery ?

1. Are `COLOR_MODE_HS/RGB/XY` essentially the same thing? (just exposed through different properties for integration convenience) Or is there a semantic difference?
2. What is `COLOR_MODE_WHITE`? I tried the "Entrance Color + White Light" demo light, but the UI is no different from a HS light, and the backend just gets passed `hs_color` (nothing with `white`). So what is the difference here to "normal" HS lights?
3. Maybe connected to 2.: Is there some way to "switch" between white/color mode? Like I have an RGBW strip. Sometimes I want to use only the RGB channels of it, sometimes only the white channel (and RGB off for accurate white light). How can this be modeled in the HA light model?

oxan, 2021-06-24:
> not sure I'm following here. there's only one active color mode at a time, depending on the color mode different properties can/can't be set. So if I'm not missing something obvious it should really [...]

You're right. I was thinking about having to handle transitions between different color modes, but since it'll be explicit what mode is requested that'll be as simple as writing zero to the other channels.

emontnemery, 2021-06-30:
> Are COLOR_MODE_HS/RGB/XY essentially the same thing? (just exposed through different properties for integration convenience) Or is there a semantic difference?
HS and XY are the same, although there might be a difference in addressable color space, I am not sure.
RGB differs in the sense that it allows for non-normalized colors, meaning an RGB color holds brightness information whereas an HS or XY color doesn't.

> What is COLOR_MODE_WHITE? I tried the "Entrance Color + White Light" demo light, but the UI is no different from a HS light, and the backend just gets passed hs_color (nothing with white). So what is the difference here to "normal" HS lights?

COLOR_MODE_WHITE flags that a light which can be either in color mode or in white mode with fixed color temperature is in white mode. More details here: https://developers.home-assistant.io/docs/core/entity/light#color-modes

Frontend does support this by showing a toggle between color and white:
[image]

> Maybe connected to 2.: Is there some way to "switch" between white/color mode? Like I have an RGBW strip. Sometimes I want to use only the RGB channels of it, sometimes only the white channel (and RGB off for accurate white light). How can this be modeled in the HA light model?

You can do that by setting `supported_color_modes = ["hs", "white"]`, for example.

oxan, 2021-07-06:
> Probably true, but I have a feeling it might not be that bad. I can spare a little time for it this weekend, and even more the one after that.

Hi @OttoWinter, did you get anywhere with this? I'd love it if we can get the light color control sorted out for the upcoming v1.20 / HA 2021.8 releases.

oxan, 2021-07-08:
I reopened this for the [color temperature / color interlock](https://github.com/esphome/issues/issues/2115#issuecomment-867643293) changes.

oxan, 2021-07-12:
I started with a color mode implementation, see esphome/esphome#2012.

WeekendWarrior1, 2021-07-20:
Hi @oxan, is https://github.com/esphome/esphome/commit/f9797825ad03d2e692290ef4f337a58cad47fd62 considered an intermediate piece of code?
Asking because this commit seemed to break the white channel on addressable RGBW SK6812 strips for me, but I am unsure if maybe that's also just the HA UI is no longer set up in the right way to correctly set the white channel.

If this is expected on the dev branch, please ignore, else, if you're curious, the issues I am seeing are:
The white LED is only set (on) during the transition length of a light state change, as soon as the light finishes it's transition, the white LED flicks off.
```yaml
light:
  - platform: neopixelbus
    type: RGBW
    pin: GPIO17
    variant: SK6812
    num_leds: 3
    name: "Test AC LED Strip"
```

WeekendWarrior1, 2021-07-20:
I set gamma correction to 0 but was still facing the issue.
To elaborate on the issue, regardless of what level I have set the white channel, the white LED only lights for the length of the transition, then it turns off. eg, in the picture below, I have set the white channel to max, and it lights at the maximum intensity for the default transition length (1.0s), before turning off:
[image]

This is the offending line (reverted it and white channel worked as expected):
https://github.com/esphome/esphome/commit/f9797825ad03d2e692290ef4f337a58cad47fd62#diff-3b81c3b82af5aae01e880f1f5b17d55754abcc67dd56e012dd47955284cb91e9R32

Though when I say "works as expected", it works how the RGBW control used to, which didn't really make that much sense:
- The ``Brightness`` slider controls the overall intensity of the RGB channels
- The ``Color brightness`` slider does nothing, and if you move it, it will reset to max next time you change the light state in any way
- The ``White brightness`` controls the white channel's intensity.

The reason I thought this may have been an intermediate/interim change is because I thought it might end up being split into two tabs like discussed above: https://github.com/esphome/issues/issues/2115#issuecomment-871145791

oxan, 2021-07-20:
Please open a new issue, I'd like to keep this one focused on architectural discussion and not implementation problems.

closed 2021-07-29 by OttoWinter