esphome/issues #2120: Cannot get CT Clamp Current Sensor to work

pkozul, 2021-06-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 (Wemos D1 Mini)

**ESPHome version (latest production, beta, dev branch)**
Version: 1.18.0 (latest production)

**Affected component:**
https://esphome.io/components/sensor/ct_clamp.html

**Description of problem:**
Cannot get the CT Clamp Current Sensor to work. I have a standard setup using the YHDC SCT-013-000 current sensor following this [guide](https://learn.openenergymonitor.org/electricity-monitoring/ct-sensors/interface-with-arduino).

I have configured my Wemos D1 Mini using the example from the ESPHome [CT Clamp Current Sensor](https://esphome.io/components/sensor/ct_clamp.html) docs.

I am trying to monitor the power consumption of a heater which is rated at 1500 W (6.5 A). Using ESPHome, I get roughly the same readings, regardless of whether the heater is switched ON or OFF (see logs below).

As a sanity check, I tried this [custom component](https://github.com/esphome/feature-requests/issues/96#issuecomment-488794040) provided by @animavitis, and simply changed the component to return current (RMS) instead of voltage. The results looked correct (see logs below), although some calibration would be required.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wemos_d1_mini
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "XXXXXXX"
  password: "XXXXXXXXX"

logger:
api:
ota:

# Example configuration entry
sensor:
  - platform: ct_clamp
    sensor: adc_sensor
    name: "Measured Current"
    update_interval: 60s

  # Example source sensor
  - platform: adc
    pin: A0
    id: adc_sensor
```

**Logs (if applicable):**
Below are the logs when using the ESPHome CT Clamp Current Sensor, which doesn't appear to show much difference between OFF and ON.
```
HEATER OFF

[19:28:59][D][ct_clamp:051]: 'Measured Current' - Raw Value: 0.00A
[19:28:59][D][sensor:099]: 'Measured Current': Sending state 0.00137 A with 2 decimals of accuracy
[19:29:01][D][adc:058]: 'adc_sensor': Got voltage=0.56V
[19:29:01][D][sensor:099]: 'adc_sensor': Sending state 0.56250 V with 2 decimals of accuracy

HEATER ON

[19:29:59][D][ct_clamp:051]: 'Measured Current' - Raw Value: 0.02A
[19:29:59][D][sensor:099]: 'Measured Current': Sending state 0.02464 A with 2 decimals of accuracy
[19:30:01][D][adc:058]: 'adc_sensor': Got voltage=0.54V
[19:30:01][D][sensor:099]: 'adc_sensor': Sending state 0.54492 V with 2 decimals of accuracy

[19:30:59][D][ct_clamp:051]: 'Measured Current' - Raw Value: 0.03A
[19:30:59][D][sensor:099]: 'Measured Current': Sending state 0.03254 A with 2 decimals of accuracy
[19:31:01][D][adc:058]: 'adc_sensor': Got voltage=0.52V
[19:31:01][D][sensor:099]: 'adc_sensor': Sending state 0.51660 V with 2 decimals of accuracy

[19:31:59][D][ct_clamp:051]: 'Measured Current' - Raw Value: 0.02A
[19:31:59][D][sensor:099]: 'Measured Current': Sending state 0.01632 A with 2 decimals of accuracy
[19:32:01][D][adc:058]: 'adc_sensor': Got voltage=0.52V
[19:32:01][D][sensor:099]: 'adc_sensor': Sending state 0.51660 V with 2 decimals of accuracy
```

Below are the logs when using the [custom component](https://github.com/esphome/feature-requests/issues/96#issuecomment-488794040) provided by @animavitis, which appears to show the expected difference between OFF and ON.
```
HEATER OFF

[19:38:25][D][sensor:099]: 'Power Meter': Sending state 0.16242 A with 4 decimals of accuracy

HEATER ON

[19:39:40][D][sensor:099]: 'Power Meter': Sending state 7.50545 A with 4 decimals of accuracy
```

**Additional information and things you've tried:**
I am 99% certain I was able to successfully use the ESPHome CT Clamp Current Sensor about a year ago. I am quite certain that nothing has changed in terms of the 'hardware' I am using. Could it be that a code change in ESPHome has caused this regression?

jesserockz, 2021-06-13:
You need to add filters to the CT clamp sensor to scale the readings to the correct amperage.
Here is my yaml suited to my house, clamp and resistors.

```yaml
sensor:

  - platform: adc
    id: analog
    pin: GPIO34
    attenuation: 11db
    update_interval: never

  - platform: ct_clamp
    name: CT Measured Current
    id: current
    sensor: analog
    sample_duration: 200ms
    update_interval: 1000ms
    filters:
      - calibrate_linear:
          - 0 -> 0
          - 0.47 -> 19.23

```

pkozul, 2021-06-14:
@jesserockz 

Thanks for the tip. Yes, I was not using the `calibrate_linear` filter properly. All good.

closed 2021-06-14 by pkozul