esphome/issues #2121: WaveShare e-paper 5.83in V2 display not working

TheStork239, 2021-06-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Waveshare driver board. (I also tried with ESP8266 Waveshare driver board resulting in same issue.)

**ESPHome version (latest production, beta, dev branch)**

v1.18

**Affected component:**

WaveShare e-paper display
https://esphome.io/components/display/waveshare_epaper.html

**Description of problem:**
WaveShare 5.83in black & white display does not work with ESPHome, with the display not refreshing at all, and remaining blank (white), with only a black edge appearing around entire screen when connecting to the ESP32 device.  The display has a "V2" sticker on it; could it be that a new model config is needed for that too, as the new 5.83in V2 is now 648x480 resolution rather than 600x448 like the original 5.83in model was?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: c-epaper
  platform: ESP32
  board: esp32dev
  on_boot:
    priority: -10
    then:
      - delay: 10s
      - component.update: epaperdisplay

wifi:
  ssid: "removed"
  password: "removed"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "removed"

font:

  - file: 'fonts/Roboto-Regular.ttf'
    id: main_sensor_unit
    size: 20
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'ä', 'ö', '/','º','µ','³']

spi:
  clk_pin: 13   #SCK
  mosi_pin: 14  #DIN

display:
  - platform: waveshare_epaper
    id: epaperdisplay
    cs_pin: 15
    dc_pin: 27
    update_interval: 600s
    busy_pin: 25
    reset_pin: 26
    model: 5.83in
    lambda: |-

      it.print(40, 40, id(main_sensor_unit), "TEST");

```

**Logs (if applicable):**

```
No relevant log messages / errors.

```

**Additional information and things you've tried:**
I have also tried the display on ESPHome using a Waveshare ESP8266 driver board, but with no better success.

jesserockz, 2021-06-13:
Yes looking at the codebase for waveshare, the new versions need a few changes to the code to get it working

TheStork239, 2021-10-21:
The issue is still there and not fixed.  It seems there are issues also with the recent 7.5in Waveshare display hardware versions too (issue #2216).  Are any of our Waveshare code base experts looking at reviewing how to keep the latest display versions working with ESPHome?
(Apologies, I looked at the waveshare code and my programming skills aren’t good enough to identify and implement fixes needed)

tejasitraj, 2022-01-03:
+1 on this issue. It seems the 5.83in display is not correctly supported by ESPHome. When I set the model: parameter to 4.20in, a part of the display works perfectly. But when I set the model: parameter to 5.83in, the display does not respond at all. @TheStork239 @jesserockz are there any updates on this? Thank you!!
Happy to post code or images if required.

TheStork239, 2022-04-18:
It looks like at least Waveshare's own example Arduino code has a set for 5.83in V2, separate from the "V1".  I wonder if that means that ESPHome's waveshare model parameter "5.83in" is simply not going to work with the V2 e-paper display.
https://github.com/waveshare/e-Paper/tree/master/Arduino/epd5in83_V2

I've also tried changing the reset_duration parameter to 2ms, instead of default 200ms (as that apparently helps with some display revisions), but that did not change the situation at all - the display does not seem to do much apart from have a black edge around the display area.

closed 2023-05-01 by jesserockz