esphome/issues #2123: MiFlora (Xiaomi BLE) battery level unavailable

austwhite, 2021-06-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant / Home Assistant Operating System in VM on VirtualBox
**ESP (ESP32/ESP8266, Board/Sonoff):**

Generic ESP32 device
**ESPHome version (latest production, beta, dev branch)**

Latest Production via Home Assistant addon
**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html

**Description of problem:**
This was initially reported in 2019 under #107 and closed with the statement:
Closing as won't fix - I received my MiFlora and was able to confirm that the device does not send battery level data anymore (not even in a new packet type).

However, the device definitely does send the battery level data as both the Home Assistant MiFlora integration and the FlowerCare app from Xiaomi can accurately retrieve the battery level data from the device that ESPHome cannot.
Can this issue be revisited and looked into further?

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: xiaomi_hhccjcy01
    mac_address: 'xx:xx:xx:xx:xx:xx'
    battery_level:
      name: 'Plant Battery Level'

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I have confirmed that the battery level is shown correctly in the Home Assistant MiFlora component, but due to location of the server the connection is flakey.

jesserockz, 2021-06-14:
So the issue is not that the device does not have battery, but that ESPHome only watches for the BLE advertising data which does not contain the battery level any longer. This is not something that can be changed directly, but you can use the [ble_client](https://next.esphome.io/components/sensor/ble_client.html) to connect and read that value. Keep in mind that a connection to the device (as HA and FlowerCare do) drains the battery faster than the ESPHome component that only listens.

austwhite, 2021-06-14:
> So the issue is not that the device does not have battery data, but that ESPHome only watches for the BLE advertising data which does not contain the battery level any longer. This is not something that can be changed directly, but you can use the [ble_client](https://next.esphome.io/components/sensor/ble_client.html) to connect and read that value. Keep in mind that a connection to the device (as HA and FlowerCare do) drains the battery faster than the ESPHome component that only listens.

This is curious as Home Assistant's component also connects using Bluetooth BLE, using the BLE mac address.  I am not sure of the mechanism for FlowerCare.
Can you explain that further?

jesserockz, 2021-06-14:
Sure,

There are 2 ways to get data from certain BLE sensors.
1. Connect to the sensor and read the characteristic data. (MiFlora app and HA Integration)
2. Listen to the data that is broadcast by the device into the air, BLE Advertising (ESPHome)

All of the Xiaomi devices so far do both of these, but this means that if you can just listen for the broadcast data, you do not need to connect to the device.

It just so happens that updated MiFlora devices do not broadcast the battery data inside the advertisement any longer.

austwhite, 2021-06-14:
Ahh.  Makes sense now.  So MiFlora, for whatever reason, decided you needed to connect and ask for the battery level, even though all the other data is just broadcast randomly to the world around it.

Sounds stupid on Xiaomi's part, but it is what it is.

I'll close this off.   Thanks @jesserockz

closed 2021-06-14 by austwhite