esphome/issues #2124: ESP32 Built-In Hall Effect Sensor Makes "button presses"

mochoandre, 2021-06-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Just testing module on it's on. Compilation through Hass.io Esphome module.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Wroom32 Nodemcu
**ESPHome version (latest production, beta, dev branch)**

v1.18.0
**Affected component:**

Relay 
Buttons

**Description of problem:**
I have 2 relays connected and as soon as I define Push Buttons in yaml and attach them to relays all goes crazy. I have phisically disconected buttons but logs still show buttons presses. 

**Find that it was the Bult-in hall sensor that makes button triggers.**

ESP32 is inside an electric connection box and goes crazy. I've find that if I take it out when I aproximate my finger, WITHOUT TOUCHING ANYTHING, it makes like I was pressing the push buttons (No buttons phisically connected). 

How can I disable the built-in hall sensor or configure it so it does not simulate button presses and activate my relays

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32-wc
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "wifi"
  password: "wifipass"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.4.111
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.0.0
    dns1: 8.8.8.8
    dns2: 192.168.1.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-Wc Fallback Hotspot"
    password: "sggdfhadtha24dxH"
    
web_server:
  port: 80
  
captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: restart
    name: "Reboot $hostname"
    
  - platform: gpio
    name: "Aquecedor WC"
    icon: "mdi: radiator"
    pin:
      number: GPIO23
      inverted: yes
    id: relay_1
    restore_mode: restore_default_off
    
  - platform: gpio
    name: "Luz WC"
    icon: "mdi: dome-light"
    pin:
      number: GPIO22
      inverted: yes
    id: relay_2
    restore_mode: restore_default_off
    
sensor:
  - platform: esp32_hall
    name: "ESP32 Hall Sensor WC"
    update_interval: 60s
    
binary_sensor:
  - platform: gpio
    name: "Butão Dois WC"
    pin: GPIO34
    id: switch_1
#    on_press:
#      then:
#        - switch.toggle:
#            id: relay_2

  - platform: gpio
    name: "Butão Luz WC"
    pin:
      number: GPIO35
      mode: INPUT_PULLUP
      inverted: yes
    id: switch_2
    on_press:
      then:
        - switch.toggle:
            id: relay_2
    filters:
      - delayed_on: 10ms
#    on_release:
#      then:
#        - switch.turn_off:
#            id: relay_1

```

**Logs (if applicable):**

```
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:33][D][switch:029]: 'Luz WC' Toggling ON.
[01:57:33][D][switch:045]: 'Luz WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:33][D][switch:029]: 'Luz WC' Toggling OFF.
[01:57:33][D][switch:045]: 'Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:33][D][switch:029]: 'Luz WC' Toggling ON.
[01:57:33][D][switch:045]: 'Luz WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:33][D][switch:029]: 'Luz WC' Toggling OFF.
[01:57:33][D][switch:045]: 'Luz WC': Sending state OFF
[01:57:33][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:34][D][switch:029]: 'Luz WC' Toggling ON.
[01:57:34][D][switch:045]: 'Luz WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:34][D][switch:029]: 'Luz WC' Toggling OFF.
[01:57:34][D][switch:045]: 'Luz WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state OFF
[01:57:34][D][binary_sensor:036]: 'Butão Luz WC': Sending state ON
[01:57:34][D][switch:029]: 'Luz WC' Toggling ON.
[01:57:34][D][switch:045]: 'Luz WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state ON
[01:57:34][D][binary_sensor:036]: 'Butão Dois WC': Sending state OFF

```

**Additional information and things you've tried:**

jesserockz, 2021-06-14:
GPIO34 to GPIO39 do not have internal pullups so you need an external pullup/down circuit. Without that they are left floating and this is exactly what happens.

mochoandre, 2021-06-14:
That solved the issue. I had this pins working with external pullup circuit  and I got lost during trasition to esphome. Thank you very much

closed 2021-06-14 by jesserockz

mochoandre, 2021-06-14:
Sorry to get back to this issue but I Still have problems and did a search and can not find solution because I still have problems of random Ghost Button Presses . When I turn on ESP32 all seems ok for after some time of being on I start to get Ghost  Button Presses. 

GPIO19 and GPIO21 Should be safe Right?
GND is connected to both buttons and GPIO's to other end of button.

Thanks and regards

`esphome:
  name: esp32-wc
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "wifi"
  password: "wifipass"
  manual_ip:
    static_ip: 192.168.4.139
    gateway: 192.168.1.1
    subnet: 255.255.0.0
    dns1: 8.8.8.8
    dns2: 192.168.1.1

  ap:
    ssid: "Esp32-Wc Fallback Hotspot"
    password: "dYIWJF624dxH"
    
web_server:
  port: 80
  
captive_portal:

logger:

api:

ota:

switch:
  - platform: restart
    name: "Reboot ESP32-WC"
    
  - platform: gpio
    name: "Aquecedor WC"
    icon: "mdi: radiator"
    pin:
      number: GPIO23
      inverted: yes
    id: relay_1
    restore_mode: restore_default_off
    on_turn_on:
    - delay: 480s # or whatever max on time you want
    - switch.turn_off: relay_1
    
  - platform: gpio
    name: "Luz WC"
    icon: "mdi: dome-light"
    pin:
      number: GPIO22
      inverted: yes
    id: relay_2
    restore_mode: restore_default_off
    

binary_sensor:
  - platform: gpio
    name: "Butão Dois WC"
    pin:
      number: GPIO21
      mode: INPUT_PULLUP
#      inverted: yes
    id: switch_1
#    on_press:
#      then:
#        - switch.toggle:
#            id: relay_1

  - platform: gpio
    name: "Butão Luz WC"
    pin:
      number: GPIO19
      mode: INPUT_PULLUP
#      inverted: yes
    id: switch_2
    on_press:
      then:
        - switch.toggle:
            id: relay_2
    filters:
      - delayed_on: 40ms
#    on_release:
#      then:
#        - switch.turn_off:
#            id: relay_1
  - platform: gpio
    pin: GPIO02
    name: "Movimento WC1"
    device_class: motion
    filters:
      - delayed_off: 20s
      
  - platform: gpio
    pin: GPIO04
    name: "Movimento WC2"
    device_class: motion
    filters:
      - delayed_off: 20s
      
  - platform: gpio
    pin: GPIO14
    name: "Movimento Quarto WC3"
    device_class: motion
    filters:
      - delayed_off: 20s
      
  - platform: gpio
    pin: GPIO27
    name: "Movimento Quarto WC4"
    device_class: motion
    filters:
      - delayed_off: 20s

esp32_ble_tracker:
      
sensor:
  - platform: dht
    pin: GPIO15
    model: DHT11
    temperature:
      name: "Temperatura ESP32 WC"
    humidity:
      name: "Humiddade ESP32 WC"
    update_interval: 20s
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:A1:EF:00"
    bindkey: "8381d6fbff2f435507c428931cd1dc5c"
    temperature:
      name: "Temperatura Quarto 3MMC-WC"
    humidity:
      name: "Humidade Quarto 3MMC-WC"
    battery_level:
      name: "Bateria Quarto 3MMC-WC"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:FB:FF:3B"
    bindkey: "dd8de071087ba0d9ff466eae57333967"
    temperature:
      name: "Temperatura Ecritorio 3MMC-WC"
    humidity:
      name: "Humidade Escritorio 3MMC-WC"
    battery_level:
      name: "Bateria Escritorio 3MMC-WC"
      
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:DF:44:89"
    bindkey: "6d5d5a854074d80b4c1408aae373a0db"
    temperature:
      name: "Temperatura WC 3MMC-WC"
    humidity:
      name: "Humidade WC 3MMC-WC"
    battery_level:
      name: "Bateria WC 3MMC-WC"

`