esphome/issues #2126: esp32 esp8266 show different ir code

leoncoolmoon, 2021-06-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I install by pip running on x86 cpu
**ESP (ESP32/ESP8266, Board/Sonoff):**

I usb both esp32 and esp8266, the problem happened on esp8266
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.18.0
**Affected component:**

The affected component is fujitsu climate_ir decoding of ir receiver
https://esphome.io/components/climate/ir_climate.html?highlight=ir%20remote
**Description of problem:**
I use the same code for esp8266 and esp32. Use the same remote same setting but the ir code is different.
the ir receiver on esp8266 gpio0 and esp32 gpio1
**Problem-relevant YAML-configuration entries:**
```yaml esp8266
remote_receiver:
  id: rmt_rcvr
  dump: raw 
  pin:
    number: GPIO0
    inverted: True
    mode: INPUT_PULLUP
  # high 55% tolerance is recommended for some remote control units
#  tolerance: 55%

remote_transmitter:
  pin: GPIO16
  carrier_duty_percent: 50%

climate:
  - platform: fujitsu_general # adjust to match your AC unit!
    name: "rmt_Fujitsu_AC"
    sensor: HA_BD0BT_T
    receiver_id: rmt_rcvr
```
```yaml esp32
  
remote_receiver:
  id: rcvr
  dump: raw 
  pin:
    number: GPIO1
    inverted: True
    mode: INPUT_PULLUP
  # high 55% tolerance is recommended for some remote control units
#  tolerance: 55%

      
remote_transmitter:
  pin: GPIO2
  carrier_duty_percent: 50%

climate:
  - platform: fujitsu_general # adjust to match your AC unit!
    name: "Bedroom0 AC"
    receiver_id: rcvr
    sensor: "HA_BD1BT_T"
```
**Logs (if applicable):**

```
LOG for esp32
07:37:38][D][remote.raw:028]: Received Raw: 3330, -1583, 468, -287, 540, -288, 538, -1176, 468, -286, 540, -1175, 468, -287, 539, -287, 540, -286, 540, -1175, 468, -1175, 468, -290, 537, -287, 539, -287, 541, -1174, 467, -1175, 470, -285, 540, -287, 540, -285, 541, -286, 540, -285, 
[07:37:38][D][remote.raw:028]:   541, -289, 538, -288, 537, -290, 537, -287, 539, -287, 539, -287, 540, -285, 542, -285, 539, -1175, 469, -287, 540, -285, 541, -288, 539, -285, 542, -284, 541, -286, 539, -287, 539, -1175, 469, -286, 540, -286, 540, -286, 540, -287, 540, -1175, 468, 
[07:37:38][D][remote.raw:041]:   -288, 539, -286, 540, -286, 541, -285, 541, -285, 540, -288, 539, -1175, 470, -283, 542, -1174, 469, -1175, 469, -1174, 469, -1175, 467, -1176, 468, -1176, 468
[07:37:45][D][esp32_camera:159]: Got Image: len=10240
[07:37:50][D][remote.raw:041]: Received Raw: 228
[07:37:55][D][esp32_camera:159]: Got Image: len=10028
```

```
LOG for esp8266
07:34:48][D][remote.raw:028]: Received Raw: 3347, -1564, 488, -251, 574, -251, 573, -1158, 486, -253, 572, -1158, 486, -254, 573, -252, 573, -254, 574, -1155, 486, -1156, 488, -250, 575, -251, 576, -251, 573, -1156, 488, -1155, 488, -252, 575, -252, 573, -253, 573, -252, 574, -253, 
[07:34:48][D][remote.raw:028]:   574, -249, 575, -253, 574, -251, 575, -250, 575, -252, 574, -252, 574, -253, 573, -253, 573, -1156, 488, -251, 573, -255, 572, -252, 574, -252, 573, -253, 572, -255, 572, -253, 574, -1156, 486, -253, 573, -254, 573, -252, 574, -252, 573, -1156, 487, 
[07:34:48][D][remote.raw:041]:   -252, 574, -252, 574, -252, 574, -252, 573, -254, 572, -254, 573, -1157, 485, -254, 574, -1155, 486, -1155, 489, -1155, 487, -1156, 489, -1155, 486, -1157, 486
[07:35:10][D][homeassistant.sensor:019]: 'sensor.ble_temperature_bd0bt_th': Got state 24.40
[07:35:10][D][sensor:099]: 'HA_BD0BT_T': Sending state 24.40000  with 1 decimals of accuracy
[07:35:10][D][climate:262]: 'rmt_Fujitsu_AC' - Sending state:
[07:35:10][D][climate:265]:   Mode: OFF
[07:35:10][D][climate:270]:   Fan Mode: AUTO
[07:35:10][D][climate:273]:   Swing Mode: OFF
[07:35:10][D][climate:276]:   Current Temperature: 24.40°C
[07:35:10][D][climate:282]:   Target Temperature: 22.00°C
[07:35:14][D][sensor:099]: 'rmt_WiFi Signal Sensor': Sending state -62.00000 dBm with 0 decimals of accuracy

```
**Additional information and things you've tried:**

Fusseldieb, 2021-06-15:
These are raw **timings**. In other words, how many us the IR should be on and off. The timing of your average IR remote is never 100% precise, neither is the receiving end (ESP). 
The devices (like your AC or TV) are made to tolerate this just fine, it's intended behaviour.

If you compare the readings of the ESP8266 and the ESP32:

```
ESP32..: 3330, -1583, 468, -287, 540
ESP8266: 3347, -1564, 488, -251, 574
```

you'll see they're pretty damn close. That's because they are.
The difference is 30us, which is only 0,03 milliseconds.

Does your AC trigger on both? If yes, that's just fine, not a bug. These are **timings**, not codes.

OttoWinter, 2021-06-18:
Yes, as @Fusseldieb mentioned. These are raw timings - some of the difference might come from the ESPs, but most of it will come from the remote itself - those are usually really cheap microcontrollers with inaccurate clocks (because they don't _need_ to be accurate for their task)

closed 2021-06-18 by OttoWinter