esphome/issues #2128: ST7735 appears to cause ESP8266 to fail to connect to Wifi

JamiePhonic, 2021-06-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 2021.6.4
ESP8266 (NodeMCU v3 & Wemos B1 Mini)

ESPHome V1.18.0 (Stable)

**Affected component:**

ST7735: https://www.esphome.io/components/display/st7735.html

**Description of problem:**
I cant quite figure out why, but when using (essentially) the same YAML config with the ST7735 display and with a PCF8574, the code will comple just fine on both, and will run on the ESP8266, however the one using the ST7735 will constantly fail to connect to Wi-Fi with: `reason='Auth Expired'`

The Lambda code isnt exactly "pretty" but it does work just fine with the PCF8574. (i've added comments to explain what it's doing)
It "runs" with the ST7735 (the time and date will show on the TFT and update as expected) but the device will never connect to Wi-Fi

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: lcd_test
  device_pretty_name: LCD Test

esphome:
  name: ${device_name}
  platform: ESP8266
  board: nodemcuv2

#Inherit the wifi config from common/wifi.yaml  
<<: !include common/wifi.yaml

# Enable logging
logger:

# Enable Home Assistant API
api:

# Over the air updates
ota:

font:
  - file: "fonts/arial.ttf"
    id: arial
    size: 18

globals:
  - id: display_width
    type: int
    restore_value: no
    initial_value: '20'
  - id: Li
    type: int
    restore_value: no
    initial_value: '0'
  - id: Le
    type: int
    restore_value: no
    initial_value: '0'
    
text_sensor:
  #Get this devices current IP address
  - platform: wifi_info
    ip_address:
      id: my_ip
  #Query HA for data from the Media text sensors
  - platform: homeassistant
    entity_id: media_player.bedroom
    id: bedroom_media_state
  - platform: homeassistant
    entity_id: sensor.bedroom_media_type
[... 211 lines cut ...]
[17:40:03][C][st7735:235]: Setting up ST7735...
[17:40:04][D][st7735:247]:   START
[17:40:04][C][st7735:372]: ST7735
[17:40:04][C][st7735:372]:   Rotations: 90 °
[17:40:04][C][st7735:372]:   Dimensions: 160px x 128px
[17:40:04][C][st7735:373]:   Model: ST7735 BLACKTAB
[17:40:04][C][st7735:374]:   CS Pin: GPIO2 (Mode: OUTPUT)
[17:40:04][C][st7735:375]:   DC Pin: GPIO4 (Mode: OUTPUT)
[17:40:04][C][st7735:376]:   Reset Pin: GPIO5 (Mode: OUTPUT)
[17:40:04][D][st7735:377]:   Buffer Size: 20480
[17:40:04][D][st7735:378]:   Height: 160
[17:40:04][D][st7735:379]:   Width: 128
[17:40:04][D][st7735:380]:   ColStart: 0
[17:40:04][D][st7735:381]:   RowStart: 0
[17:40:04][C][st7735:382]:   Update Interval: 1.0s
[17:40:04][D][st7735:249]:   END
[17:40:05][C][wifi:033]: Setting up WiFi...
[17:40:05][I][wifi:194]: WiFi Connecting to 'REDACTED'...
[17:40:10][W][wifi_esp8266:444]: Event: Disconnected ssid='REDACTED' bssid=8A:2A:A8:14:2A:C3 reason='Auth Expired'
[17:40:10][W][wifi:485]: Error while connecting to network.
[17:40:10][W][wifi:522]: Restarting WiFi adapter...
[17:40:10][I][wifi:194]: WiFi Connecting to 'REDACTED'...
```

**Additional information and things you've tried:**

JamiePhonic, 2021-07-03:
Update: I can confirm this exact same yaml config runs without issues on the ESP-Wroom-32 (Just updating the platform and board)

adamcodemaker, 2021-09-18:
I also have similar problems on nodemcu v3
On ESP-Wroom-32 it runs ok.

duchenpaul, 2021-09-25:
I can confirm this bug, here is my configure:
```yaml
esphome:
  name: $device_name
  platform: ESP8266
  board: $board_model
  build_path: build/$device_name
  # esp8266_restore_from_flash: yes

substitutions:
  board_model: nodemcuv2
  device_name: helloworld
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_fast_connect: "true"
  wifi_reboot_timeout: 600s
  ota_password: !secret ota_password
  api_reboot_timeout: 600s

wifi:
  ssid: $wifi_ssid
  password: $wifi_password
  #  manual_ip:
  #    static_ip: 192.168.178.230
  #    gateway: 192.168.178.1
  #    subnet: 255.255.255.0
  #    dns1: 1.1.1.1
  #    dns2: 1.2.2.1
  #----------------------------------------------------
  domain: .local
  reboot_timeout: $wifi_reboot_timeout
  power_save_mode: none
  fast_connect: $wifi_fast_connect
  ap:
    ssid: $device_name
    password: $wifi_password
    ap_timeout: 30s

captive_portal:

#--------------------------------------------
web_server:
  port: 80
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
#----------------------------------------------------

# Enable logging
logger:
  # hardware_uart: UART0
  level: DEBUG
  baud_rate: 115200

status_led:
  pin:
    number: GPIO2
    inverted: True

api:
  password: $ota_password

ota:
  password: $ota_password

spi:
  clk_pin: D5
  mosi_pin: D6
  # miso_pin: D6

display:
  - platform: st7735
    cs_pin: D8
    dc_pin: D7
    reset_pin: D3

    model: "INITR_18BLACKTAB"
    device_width: 128
    device_height: 160
    col_start: 0
    row_start: 0
    eight_bit_color: true
    lambda: |-
      it.fill(Color::BLACK);
      it.rectangle(0,  0, it.get_width()/2, it.get_height()/2, id(my_red));

color:
  - id: my_red
    red: 100%
    green: 3%
    blue: 5%
```

grego1981, 2021-12-28:
Same behaviour with ESP8266  nodemcu which does not connect if an amg8833 sensor is connected of a VL53L0X or any other sensor using SDA SCL pins. The same code works if the hardware is not connected. The example config for every hardware (without ESPHome that is) work just fine to the same wifi setup.

cbulock, 2022-01-27:
Just duplicated this same issue. Using a Wemos D1 Mini and once I add the config for the ST7735, I start seeing wifi connection timeouts.

abanchieri, 2022-02-02:
I believe I have the same (or exceptionally similar) issue. 

In my case I am trying to connect a 20x4 LCD display using ESPHome.
I have tried both an ESP8266 D1 Mini and an ESP8266 NodeMCU. 
I intend to connect the display is connected through a PCF8474 module. 

- When the display is connected to the SDA and SCL pins the text on the display will update but the ESP8266 will not connect to WiFi.
- When the display is disconnected from the SDA and SCL pins the ESP8266 will connect to WiFi. 

My YAML file is as follows (set up for the NodeMCU test case)

`esphome:
  name: sprinkler
  platform: ESP8266
#  board: d1_mini
  board: nodemcuv2

substitutions:
  devicename: sprinkler
  friendly_name: sprinkler
  
wifi:
  networks:
    - ssid: !secret wifi_ssid          
      password: !secret wifi_password  
  ap:
    ssid: "AP_${devicename}"

# Enable the captive portal for inital WiFi setup
captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE
  # level: VERBOSE

# Enable Home Assistant API
api:

# Enable OTA upgrade
ota:
  password: !secret ESPHome_OTA_password
  
time:
  - platform: homeassistant
    id: homeassistant_time  
  
binary_sensor:
  - platform: status
    name: "${friendly_name} Status"

i2c:
  sda: D2
  scl: D1
  
display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27 
    lambda: |-
      it.strftime(2,0, "%H:%M:%S", id(homeassistant_time).now());`

blounolito, 2022-07-26:
I have the exact same problem,

- If I let SDA / SCL pins connected, WiFi will struggle to connect (sometimes does not connect at all, and will switch to AP,
- If I disconnect SCL, WiFi connects just fine,

It looks like screens / I2C devices mess up with the WiFi connexion.

Some logs ;
`[23:41:52][W][wifi:572]: Restarting WiFi adapter...
[23:41:53][I][wifi:248]: WiFi Connecting to 'mine'...
[23:41:55][W][wifi_esp8266:482]: Event: Disconnected ssid='mine' bssid=MAC reason='Auth Expired'
[23:41:55][W][wifi:536]: Error while connecting to network.
[23:41:55][W][wifi:572]: Restarting WiFi adapter...
[23:42:00][D][wifi:373]: Starting scan...
[23:42:07][D][wifi:388]: Found networks:
[23:42:07][I][wifi:431]: - ...
	WiFi networks list
[23:42:07][I][wifi:248]: WiFi Connecting to 'mine'...
[23:42:09][W][wifi_esp8266:482]: Event: Disconnected ssid='mine' bssid=MAC reason='Auth Expired'
[23:42:09][W][wifi:536]: Error while connecting to network.
[23:42:09][W][wifi:572]: Restarting WiFi adapter...
[23:42:10][I][wifi:248]: WiFi Connecting to 'mine'...`

Board ; ESP8266 (Nodemcu)
Configuration ; the one in this cookbook, **exactly the same** (delta fonts used and sensors names) : https://esphome.io/cookbook/display_time_temp_oled.html

I have other similar boards with and without using I2C, they are running fine. The most complex of them exploit I2C to use an Arduino Nano as a GPIO expander, so I guess the underlying problem is in I2C + OLED / TFT.

schinckel, 2022-11-19:
Try changing your display's `update_interval`.

Setting this to 2s enabled my device to connect to my WiFi.

ssieb, 2022-11-20:
Try increasing the i2c frequency.  That has been a common solution.

ax42, 2023-03-31:
Same issue for me -- Wemos D1 with a 4x20 display via `lcd_pcf8574` gives the auth error on WiFi.

Can confirm that increasing the i2c bus speed from the default 50kHz ixes the issue:

```
i2c:
  sda: GPIO4
  scl: GPIO5
  frequency: 100kHz
```

eherranzr, 2024-08-16:
Seeing something similar in #5890
Tried increasing i2c frequency and display's update_interval but did not solve the issue

rafalkur, 2024-12-14:
Is there a solution for this issue? i was trying to find here and elsewhere, but no success. In my case i use esp32 wroom 32u and st7735s. When esp32 is with initial configuration and wifi connection, it works very reliable from networking perspective. Then, without any sensors or display connected yet im adding this line of code: 
[image]
[image]

and esp32 networking issue starts. Dropping randomly packets (ping replies get time from 20ms to 3000ms or timeout ), not possible to install anymore via wifi. 
in logs i see esp connects and disconnets:
WARNING esphome-web-221c5c @ 192.168.0.192: Connection error occurred: [Errno 104] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for esphome-web-221c5c @ 192.168.0.192
WARNING Disconnected from API

increasing the frequency in i2c: also didn't help.