esphome/issues #2130: No upload button on esphome:dev dash board

jbuszkie, 2021-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker from unraid

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.20.0-dev

**Affected component:**

Upload

**Description of problem:**

Upload button is gone and replaced with "Install"
When I click Install it pops up a pop up with three choices...  None of which do anything.
[image]

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2021-06-16:
Rebuild your install. The install wirelessly is the new OTA upload 

It should be enabled in newer build

jbuszkie, 2021-06-16:
I'm running from a docker from unraid.  I don't "build" anything. :-(
"compile" is also missing?? 1.19.0b2 is the last version that just worked.
Where is the manual compile and then download?

glmnet, 2021-06-16:
Looks like 1.19 is out today

Otherwise this might be a bug

jbuszkie, 2021-06-16:
I don't see 1.19 as released? I still see 1.18 and the "latest"  I just see 1.19.0b6..  which is broken for me as well as b5,b4, and b3
And the dev version (which also doesn't work) is 1.20-dev

jbuszkie, 2021-06-16:
Ok..  With the help of someone on the discord channel...  A full reload of the webpage (no cache - CTRL + SHIFT + R) fixed the issue.
I'll leave this open for a while in case any of the devs want to change anything so a full re-load isn't necessary.  And so other folks might be able to find this.

jbuszkie, 2021-06-16:
This is what it's supposed to look like, I guess
[image]

closed 2021-10-31 by github-actions[bot]