esphome/issues #2133: Issue with Tuya device with esphome > 1.16.2

RomRider, 2021-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP01_1m

**ESPHome version (latest production, beta, dev branch)**

Any version after 1.16.2

**Affected component:**

tuya

**Description of problem:**

Any version after `1.16.2` breaks my tuya fan.
It's stuck forever in:
```
[21:11:17][C][tuya:028]:   Configuration will be reported when setup is complete. Current init_state: 0
[21:11:17][C][tuya:029]:   If no further output is received, confirm that this is a supported Tuya device.
[21:11:18][V][tuya:317]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[04 (1)] INIT_STATE=0
[21:11:19][V][tuya:317]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[04 (1)] INIT_STATE=0
[21:11:20][V][tuya:317]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[04 (1)] INIT_STATE=0
```

While on version `1.16.2`:
```
[21:18:30][C][tuya:033]: Tuya:
[21:18:30][C][tuya:036]:   Configuration will be reported when setup is complete. Current init_state: 2
[21:18:30][C][tuya:037]:   If no further output is received, confirm that this is a supported Tuya device.
[21:18:30][D][tuya.fan:064]: Setting switch: OFF
[21:18:30][D][tuya.fan:072]: Setting oscillating: OFF
[21:18:30][D][tuya.fan:084]: Setting speed: 2
[21:18:30][D][tuya.fan:033]: MCU reported switch is: OFF
[21:18:30][D][tuya.fan:024]: MCU reported speed of: 2
[21:18:30][D][tuya.fan:064]: Setting switch: OFF
[21:18:30][D][tuya.fan:072]: Setting oscillating: OFF
[21:18:30][D][tuya.fan:084]: Setting speed: 2
[21:18:31][D][tuya.fan:064]: Setting switch: OFF
[21:18:31][D][tuya.fan:072]: Setting oscillating: OFF
[21:18:31][D][tuya.fan:084]: Setting speed: 2
[21:18:31][D][tuya.fan:041]: MCU reported oscillation is: OFF
[21:18:31][C][tuya:033]: Tuya:
[21:18:31][C][tuya:042]:   Datapoint 1: switch (value: OFF)
[21:18:31][C][tuya:048]:   Datapoint 2: enum (value: 2)
[21:18:31][C][tuya:048]:   Datapoint 3: enum (value: 0)
[21:18:31][C][tuya:042]:   Datapoint 8: switch (value: OFF)
[21:18:31][C][tuya:048]:   Datapoint 11: enum (value: 0)
[21:18:31][C][tuya:044]:   Datapoint 12: int value (value: 0)
[21:18:31][C][tuya:058]:   Product: 'NFpRxSl74TyJnhvD1.0.0'
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: '${device_id}'
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

tuya:

fan:
  - platform: 'tuya'
    name: '${device_name}'
    switch_datapoint: 1
    speed_datapoint: 2
    oscillation_datapoint: 8
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

trvrnrth, 2021-08-01:
@RomRider I you haven't already you might want to try the latest version as the issue your describe is hopefully fixed by https://github.com/esphome/esphome/pull/2065

RomRider, 2021-08-01:
Confirmed, it works now with that pull request merged! Thanks 🙏

closed 2021-08-01 by RomRider