esphome/issues #2135: ESPHome 1.19, buttons in logwindow doesn't work in Firefox

fribse, 2021-06-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant on PVE in an Intel NUC 

**ESP (ESP32/ESP8266, Board/Sonoff):**

Not related to the chip, but the ESPHome interface

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19

**Affected component:**

**Description of problem:**
After pressing 'Install' (update) over 'wireless' (OTA) it shows the logwindow in firefox (in chrome it asks for install methods)
At the bottom there are a few buttons, questionmark, EDIT STOP 'three dots'.
After doing the update, and you want to close the window, nothing happens in firefox, it works in chrome

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
[image]

balloob, 2021-06-16:
I couldn't reproduce the last time. Will try again. 

Work around is to hit ESC

glmnet, 2021-06-17:
Try another browser or clear cache

tomlut, 2021-06-17:
I can reproduce this. Particularly if an upload fails. The only way to close the window is to reload the browser.

Guyohms, 2021-06-17:
Same here on Chrome

Reverted to 1.18.0 because the ESP32 crashed  when uploaded to 1.19.0 but that will be for another issue.

nickrout, 2021-06-17:
Same in Chrome.

fribse, 2021-06-17:
It seems that doing a clear cache in firefox, and then shutting it down, and starting it up, brings the buttons back. Why does everything in Home Assistant require a clearing cache now, the amount of 'dead sessions' left in HA is staggering when you have to clear cache for every update.

balloob, 2021-06-17:
The template should add a cache busting hash to the end of the legacy `esphome.js`, but it somehow isn't working for the people reporting here. In https://github.com/esphome/esphome/pull/1930 I've rewritten the cache busting logic so this should fix it. 

What you're seeing now is certain buttons being handled by old + new code.

closed 2021-06-17 by jesserockz

tomlut, 2021-06-17:
Deleting everything in my browsing history for all time ( Firefox Settings / Privacy ) and then restarting fixed this and other issues for me.