esphome/issues #2136: No way to download .bin file in Firefox

tomlut, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant OS (update your document template Hassio has not been a thing for 18 months).

**ESP (ESP32/ESP8266, Board/Sonoff):**

Any ESP board
**ESPHome version (latest production, beta, dev branch)**

1.19
**Affected component:**

Main Interface

**Description of problem:**
There is no way to compile and download a .bin file in Firefox as this pop-up only appears when using Chrome browser. I get that the serial options are not available on firefox but there's no need to disable downloading .bin files as well.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

tht, 2021-06-17:
I think there is a broader issue here… since the update to 1.19.0 it is also no longer possible to download the bin-file for a device never flashed before – or I can't find the link ;)

Older versions allowed to `compile` and download the binary afterwards. I used this to update existing Tasmota devices OTA but now there does not seem to be a way to download a binary.

btw… using Safari 14.1.1 on macOS but I don't think this matters.

zvldz, 2021-06-17:
I was able to download successfully via firefox.

[image]

tomlut, 2021-06-17:
Was that the first time you ever attempted to flash that file?

zvldz, 2021-06-17:
I copied the existing yaml file with the new name and renamed the device in it. And was able to compile and download twice.

tomlut, 2021-06-17:
I just created a brand new file and attempted to install it. I see the pop-up for the briefest moment, then it disappears and an OTA install is attempted.

zvldz, 2021-06-17:
I am using esphome v1.20.0-dev.
Maybe for this reason it works fine for me.

tomlut, 2021-06-17:
I just blow away my entire browsing history, cookies, everything.

Now it works as expected.

Yet another example on not being able to flush the browser cache.

closed 2021-06-17 by tomlut