esphome/issues #2137: midea_ac component not works in 1.19.0

dszabolcs, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
debian

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp32

**ESPHome version (latest production, beta, dev branch)**
esphome 1.19.0

**Affected component:**
midea_ac

**Description of problem:**
The midea_ac component is not working, because the new features were added to 1.19.0, but unfortunately one pull request for aioesphome lib was not sent (https://github.com/esphome/aioesphomeapi/pull/24) and that's why the home assistant crashes when the esphome tries to connect it. And I think maybe all of the climate component may have the same issue. Please correct it as soon as possible.

**Problem-relevant YAML-configuration entries:**
irrelevant

**Logs (if applicable):**
2021-06-09 19:00:21 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 214, in on_login
    entity_infos, services = await cli.list_entities_services()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
    entities.append(cls(**kwargs))
  File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in init
    self.supported_modes = __attr_converter_supported_modes(supported_modes)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/enum.py", line 339, in call
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.8/enum.py", line 662, in new
    raise ve_exc
ValueError: 6 is not a valid ClimateMode

**Additional information and things you've tried:**

vzagorovskiy, 2021-06-17:
I have the same issue. midea_ac doesn't work on 1.19

maretodoric, 2021-06-17:
FYI, if you have a backup, you may be able to downgrade firmware on it to an earlier version.
- Extract the backup
- Extract the addon backup (15ef4d2f_esphome.tar.gz)
- Go to data/<your_project_name>/.pioenvs/<your_project_name>
- Flash firmware.bin to your device

You can also attempt an OTA flash if access HA console directly or use  HassOS SSH port 22222 Configurator addon. In that case, first create `<your_project_name>/.pioenvs/<your_project_name>` directory in `/config/esphome/` folder (or if you're directly connected to the console, `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`) and move the firmware.bin there.

- Open esphome addon shell using `docker exec -it addon_15ef4d2f_esphome /bin/bash`, if addon_15ef4d2f_esphome is not the correct name, check the name using `docker container ls`
- Run `esphome upload --device <device_ip_address> /config/esphome/<your_project_name>.yaml`

It should attempt the OTA upload using the binary file from `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`

TravisRoy, 2021-06-17:
I am also having the same issue using the Mitsubishi mini-split add on. What's odd is that I can control the unit, but I don't get any information back. Happened as soon as I installed 1.19

mithunsnair, 2021-06-17:
Same issue for my climate module after today's update to 1.19.0. 

In the ESPHome logs everything is working well. Device automation is also working. 

However the link to Home Assistant seems broken. HA Dashboard shows not available. 

Tried removing the integration in Home Assistant and adding it many times; it only detects the device correctly but zero entities come up. 

It's an esp32dev board with DHT11 temperature sensor using climate_ir_lg

jesserockz, 2021-06-17:
Ok so esphome/esphome#1471 was not as backwards compatible as we thought.

Until it is fixed you can downgrade the esphome addon using the `esphome_version` configuration in the addon settings and set it to `v1.18.0`

[image]

closed 2021-06-17 by jesserockz

jesserockz, 2021-06-18:
Fixed in [v1.19.1](https://github.com/esphome/esphome/releases/tag/v1.19.1) which should be available in ~10 mins

cianial, 2021-06-20:
Operating environment/Installation (Hass.io/Docker/pip/etc.):
Hass.io

ESP (ESP32/ESP8266, Board/NodeMCU):
esp8266

ESPHome version (latest production)
esphome 1.19.1

Affected component:
climate

I use the mitsubishiheatpump and starting from the esphome 1.19.0 I get the same error, so for me the esphome update to 1.19.1 don't fix the problem. any idea?

jesserockz, 2021-06-20:
@cianial Can you please open a new issue with more details?

bulletmark, 2021-06-21:
The component [mitsubishiheatpump](https://github.com/geoffdavis/esphome-mitsubishiheatpump) is broken in 1.19.0, 1.19.1, and 1.19.2. Has there a backwards incompatible change made that requires that component to be updated?

jesserockz, 2021-06-21:
@bulletmark https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/37

bulletmark, 2021-06-21:
@jesserockz thanks. I watch that repo so saw your issue and created [this commit](https://github.com/bulletmark/esphome-mitsubishiheatpump/commit/6e3ea4989b172c072c2232f2c41aed82ca75a36e) which seems to work for me.