esphome/issues #2138: OTA Install from Dashboard not working after update to 1.19.0, works again after restarting dashboard

Plawasan, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip on Raspbian GNU/Linux 10 (buster)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19.0
**Affected component:**

Dashboard

**Description of problem:**
After updating to 1.19.0 and restarting the dashboard process (sudo esphome dashboard esphome/ &) the first time i tried to Install an existing yaml via OTA to update it I immediately got "Terminated Process". After killing the dashboard process and starting it again with the identical command the OTA Install worked - see debug below. I have found a previous issue where this was caused by incorrect $PATH config, I don't believe that is the case here:

pi@raspberrypi:~ $ which esphome
/usr/local/bin/esphome
pi@raspberrypi:~ $ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
INFO 304 GET /static/js/vendor/ace/mode-yaml.js (192.168.1.224) 10.42ms
INFO Running command 'esphome --dashboard -q vscode esphome/ --ace'
INFO 101 GET /upload (192.168.1.224) 2.11ms
ERROR Uncaught exception GET /upload (192.168.1.224)
HTTPServerRequest(protocol='http', host='pi.lan:6052', method='GET', uri='/upload', version='HTTP/1.1', remote_ip='192.168.1.224')
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/tornado/websocket.py", line 647, in _run_callback
    result = callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 134, in decorator
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 212, in on_message
    handlers[type_](self, json_message)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 219, in handle_spawn
    command = self.build_command(json_message)
  File "/usr/local/lib/python3.7/dist-packages/esphome/dashboard/dashboard.py", line 297, in build_command
    json_message["port"],
KeyError: 'port'

pi@raspberrypi:~ $ fg
sudo esphome dashboard esphome/
^C
INFO Shutting down...

pi@raspberrypi:~ $ sudo esphome dashboard esphome/ &
[1] 1972
pi@raspberrypi:~ $ INFO Starting dashboard web server on port 6052 and configuration dir esphome/...
INFO 101 GET /ace (192.168.1.224) 5.25ms
INFO Running command 'esphome --dashboard -q vscode esphome/ --ace'
INFO 101 GET /upload (192.168.1.224) 4.08ms
INFO Running command 'esphome --dashboard run esphome/esp_soil_sensor_2.yaml --device OTA'

```

**Additional information and things you've tried:**

vzagorovskiy, 2021-06-17:
I have the same issue. 

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19.0

**Affected component:**

Dashboard

zvldz, 2021-06-17:
I have the same problem

taintedkernel, 2021-06-17:
Likewise here.  Seems to be a little intermittent

MattWestb, 2021-06-17:
I also getting it nearly all time trying updating device.
If not  the posses is starting its only closing the the "install" and opening it and its normally working.
Have updating (install) 10 device and half was not working on the first try but all is done on 2 or 3rd try.

balloob, 2021-06-17:
This problem occurs if no device is picked in the top right dropdown. This should happen automatically from the install dialog but this somehow doesn't always seem to work (still figuring out why). Instead of generating an invalid message, it will now fallback to OTA with https://github.com/esphome/esphome/pull/1930

closed 2021-06-17 by jesserockz