esphome/issues #2139: Issue with temperature controller on 1.19.0

Tsalokin, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP8266, Board/nodemcu):**

**ESPHome version 1.19.0**

**Affected component:**

https://esphome.io/components/climate/thermostat.html

**Description of problem:**
error thrown on thermostat component after update to 1.19.0
esphome then fails to connect and logs are not shown in the web-viewer
everything else works as long as the climate controller (and related) config is commented

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: autowindow
  platform: ESP8266
  board: nodemcu

wifi:
  ssid: "Sauce"
  password: "BrownWhiteRedClear"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Autowindow Fallback Hotspot"
    password: "7ZDl1qE374HK"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: homeassistant
    internal: true
    entity_id: sensor.outside_temperature
    id: out_temp
    filters:
      - lambda: !lambda |-
          return (x - 32.0) * 5.0/9.0;
  - platform: homeassistant
    internal: true
    entity_id: sensor.nva_room_temperature
    id: nva_temp
    filters:
      - lambda: !lambda |-
          return (x - 32.0) * 5.0/9.0;
    
    
switch:
- platform: gpio
  pin: D5
  interlock: &interlock [open_cover, close_cover]
  id: open_cover
- platform: gpio
  pin: D6
  interlock: *interlock
  id: close_cover

cover:
- platform: time_based
  name: "NVA Window"
  id: nva_window
  device_class: window
  open_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover
    - delay: 1s
    - switch.turn_on: open_cover
  open_duration: 12s
  close_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover
    - delay: 1s
    - switch.turn_on: close_cover
  close_duration: 15s
  stop_action:
    - switch.turn_off: open_cover
    - switch.turn_off: close_cover

climate:
  - platform: thermostat
    name: "NVA Room Thermostat"
    sensor: nva_temp
    id: nva_therm
    hysteresis: 0.5
    default_target_temperature_low: 65 °F
    default_target_temperature_high: 71 °F
    cool_action:
      - logger.log: "Cooling"
    heat_action:
      - logger.log: "Heating"
    idle_action:
      - logger.log: "Idle"
    off_mode:
      - lambda: !lambda |-
          auto call = id(nva_window).make_call();
          call.set_command_close();
          call.perform();
          
globals:
  - id: last_action
    type: int
    restore_value: no
    initial_value: '0'
  - id: current_action
    type: int
    restore_value: no
    initial_value: '0'

time:
  - platform: homeassistant
    id: homeassistant_time
    on_time:
    - seconds: 0
      minutes: /1
      then:
      - logger.log: 
          format: "ACTION: %.1d LACTION: %.1d NVA %.1f OUT %.1f"
          args: [ 'id(nva_therm).action', 'id(last_action)' , 'id(nva_therm).current_temperature', 'id(out_temp).state']
      - lambda: !lambda |-
      
          //update last action
          if(id(current_action) != id(nva_therm).action){
            id(last_action) = id(current_action);
            id(current_action) = id(nva_therm).action;
          }
      
          if(id(current_action) == CLIMATE_ACTION_COOLING ){ //if too hot inside and colder outside, open else close
            if (id(out_temp).state < id(nva_therm).current_temperature) {
              auto call = id(nva_window).make_call();
              call.set_position(0.5);
              call.perform();
            } else {
              auto call = id(nva_window).make_call();
              call.set_command_close();  
              call.perform();
            }
          }else if(id(current_action) == CLIMATE_ACTION_HEATING ){ //if too cold inside and hotter outside, open else close
            if (id(out_temp).state  > id(nva_therm).current_temperature) {
              auto call = id(nva_window).make_call();
              call.set_position(0.5);
              call.perform();
            } else {
              auto call = id(nva_window).make_call();
              call.set_command_close();
              call.perform();
            }
          // \/ to prevent temparature change when inside temperature is within requested range and outside is hotter or colder
          }else if(id(current_action) == CLIMATE_ACTION_IDLE){
            if (id(last_action) == CLIMATE_ACTION_COOLING && id(out_temp).state > id(nva_therm).target_temperature_high) {
              auto call = id(nva_window).make_call();
              call.set_command_close();
              call.perform();
            }else if(id(last_action) == CLIMATE_ACTION_HEATING && id(out_temp).state  < id(nva_therm).target_temperature_low){
              auto call = id(nva_window).make_call();
              call.set_command_close();
              call.perform();
            }
          }

```

**Logs (if applicable):**

```
2021-06-17 00:02:34 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 214, in on_login
entity_infos, services = await cli.list_entities_services()
File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
entities.append(cls(**kwargs))
File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in __init__
self.supported_modes = __attr_converter_supported_modes(supported_modes)
File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
return [ClimateMode(val) for val in value]
File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
return [ClimateMode(val) for val in value]
File "/usr/local/lib/python3.8/enum.py", line 339, in __call__
return cls.__new__(cls, value)
File "/usr/local/lib/python3.8/enum.py", line 662, in __new__
raise ve_exc
ValueError: 6 is not a valid ClimateMode
2021-06-16 23:53:25 WARNING (MainThread) [homeassistant.components.light] Updating tplink light took longer than the scheduled update interval 0:00:05
2021-06-17 00:02:34 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for autowindow (10.0.1.40): Timeout while connecting to ('10.0.1.40', 6053)

```

**Additional information and things you've tried:**

vzagorovskiy, 2021-06-17:
I have the same issue with climate component with midea_ac platform #2137

dszabolcs, 2021-06-17:
unfortunately, all of the climate components are affected.

maretodoric, 2021-06-17:
FYI, if you have a backup, you may be able to downgrade firmware on it to an earlier version.
- Extract the backup
- Extract the addon backup (15ef4d2f_esphome.tar.gz)
- Go to data/<your_project_name>/.pioenvs/<your_project_name>
- Flash firmware.bin to your device

You can also attempt an OTA flash if access HA console directly or use  HassOS SSH port 22222 Configurator addon. In that case, first create `<your_project_name>/.pioenvs/<your_project_name>` directory in `/config/esphome/` folder (or if you're directly connected to the console, `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`) and move the firmware.bin there.

- Open esphome addon shell using `docker exec -it addon_15ef4d2f_esphome /bin/bash`, if addon_15ef4d2f_esphome is not the correct name, check the name using `docker container ls`
- Run `esphome upload --device <device_ip_address> /config/esphome/<your_project_name>.yaml`

It should attempt the OTA upload using the binary file from `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`

closed 2021-06-17 by jesserockz

jesserockz, 2021-06-18:
Fixed in [v1.19.1](https://github.com/esphome/esphome/releases/tag/v1.19.1) which should be available in ~10 mins