esphome/issues #2140: esphome devices with climate components don't create entities in Home Assistant after upgrade to 1.19.0

lpomfrey, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome installed in a python virtualenv 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.19.0

**Affected component:**

- [Thermostat climate](https://esphome.io/components/climate/thermostat.html)
- [PID controller](https://esphome.io/components/climate/pid.html)

**Description of problem:**
Any esphome node with a climate component has it's entities in homeassistant become unavailable after updating to 1.19.0. When deleting and re-adding the integration the device appears with no entities attached. There is no sign of a problem in the esphome log although homeassistant complains about an invalid `ClimateMode`.

**Problem-relevant YAML-configuration entries:**
This is one of the affected configurations using a PID controller, if needed I can provide others using PID controllers and thermostat climate components that are affected in the same way.

```yaml
---
esphome:
  on_boot:
    then:
      - output.turn_off: dimmer2

dallas:
  - pin: 18
    update_interval: 5s

sensor:
  - platform: dallas
    address: "${temperature_address}"
    id: temperature
    name: "${stat_name} Temperature"
    filters:
      - exponential_moving_average:
          send_every: 2
  - platform: pid
    name: "${stat_name} PID kp"
    climate_id: stat1
    type: KP
    accuracy_decimals: 5
  - platform: pid
    name: "${stat_name} PID ki"
    climate_id: stat1
    type: KI
    accuracy_decimals: 5
  - platform: pid
    name: "${stat_name} PID kd"
    climate_id: stat1
    type: KD
    accuracy_decimals: 5
  - platform: pid
    name: "${stat_name} PID Heating Power"
    climate_id: stat1
    type: HEAT

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 16
    min_power: 0.1
    method: leading
    init_with_half_cycle: false
    zero_cross_pin:
      number: 15
      mode: INPUT_PULLUP
      inverted: true
  - platform: ac_dimmer
    id: dimmer2
    gate_pin: 17
    min_power: 0.1
    method: leading
    init_with_half_cycle: false
    zero_cross_pin:
      number: 15
      mode: INPUT_PULLUP
      inverted: true

switch:
  - platform: template
    name: "${stat_name} Autotune"
    turn_on_action:
      - climate.pid.autotune: stat1

climate:
  - platform: pid
    name: "${stat_name} Stat"
    id: stat1
    sensor: temperature
    default_target_temperature: ${default_temp}
    heat_output: dimmer1
    visual:
      min_temperature: ${min_value}
      max_temperature: ${max_value}
      temperature_step: ${value_step}
    control_parameters:
      kp: ${kp}
      ki: ${ki}
      kd: ${kd}

sun:
  <<: !include _location.yaml
  on_sunrise:
    then:
      - if:
          condition:
            - text_sensor.state:
                id: stat_mode
                state: "Auto"
          then:
            - climate.control:
                id: stat1
                mode: "AUTO"
                target_temperature: ${day_temp}
  on_sunset:
    then:
      - if:
          condition:
            - text_sensor.state:
                id: stat_mode
                state: "Auto"
          then:
            - climate.control:
                id: stat1
                mode: "AUTO"
                target_temperature: ${night_temp}

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 11
        then:
          - if:
              condition:
                - text_sensor.state:
                    id: stat_mode
                    state: "Auto"
              then:
                - climate.control:
                    id: stat1
                    mode: "AUTO"
                    target_temperature: ${peak_temp}
      - seconds: 0
        minutes: 0
        hours: 13
        then:
          - if:
              condition:
                - text_sensor.state:
                    id: stat_mode
                    state: "Auto"
              then:
                - climate.control:
                    id: stat1
                    mode: "AUTO"
                    target_temperature: ${day_temp}

text_sensor:
  - platform: homeassistant
    id: stat_mode
    name: "${stat_name} Stat Mode"
    entity_id: ${mode_entity_id}
    on_value:
      then:
        - if:
            condition:
              - text_sensor.state:
                  id: stat_mode
                  state: "Auto"
            then:
              - if:
                  condition:
                    - sun.is_below_horizon:
                  then:
                    - climate.control:
                        id: stat1
                        mode: "AUTO"
                        target_temperature: ${night_temp}
                  else:
                    - climate.control:
                        id: stat1
                        mode: "AUTO"
                        target_temperature: ${day_temp}

```

**Logs (if applicable):**

esphome log is perfectly normal, homeassistant log contains:

```
ValueError: 6 is not a valid ClimateMode
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/var/homeassistant/environment/lib/python3.8/site-packages/homeassistant/components/esphome/__init__.py", line 214, in on_login
    entity_infos, services = await cli.list_entities_services()
  File "/var/homeassistant/environment/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
    entities.append(cls(**kwargs))
  File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in __init__
    self.supported_modes = __attr_converter_supported_modes(supported_modes)
  File "/var/homeassistant/environment/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
    return [ClimateMode(val) for val in value]
  File "/var/homeassistant/environment/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
    return [ClimateMode(val) for val in value]
  File "/usr/lib/python3.8/enum.py", line 309, in __call__
    return cls.__new__(cls, value)
  File "/usr/lib/python3.8/enum.py", line 600, in __new__
    raise exc
  File "/usr/lib/python3.8/enum.py", line 584, in __new__
    result = cls._missing_(value)
  File "/usr/lib/python3.8/enum.py", line 613, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 6 is not a valid ClimateMode

```

**Additional information and things you've tried:**
Deleting and re-adding the integration doesn't work as detailed above, the only fix seems to be downgrading to esphome 1.18.0

dszabolcs, 2021-06-17:
Same issue happens in these cases:

https://github.com/esphome/issues/issues/2137
https://github.com/esphome/issues/issues/2139

maretodoric, 2021-06-17:
I can report this as well. I have a custom gree climate component and ever since the update, my climate entity is in unavailable state and following exception is shown in log file after the device is initiated:

```
2021-06-17 15:14:28 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 214, in on_login
    entity_infos, services = await cli.list_entities_services()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
    entities.append(cls(**kwargs))
  File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in __init__
    self.supported_modes = __attr_converter_supported_modes(supported_modes)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/enum.py", line 339, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.8/enum.py", line 662, in __new__
    raise ve_exc
ValueError: 6 is not a valid ClimateMode
```

All worked fine up until the upgrade.

```
esphome:
  name: mcu_dnevna_soba_transciever
  platform: ESP8266
  board: nodemcuv2
  includes: 
    - custom_components/gree/gree_ir.h
  libraries:
    - IRremoteESP8266

wifi:
  ssid: "SSID"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "MCU Dnevna Soba Transciever"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"

mqtt:
  broker: 192.168.0.201
  username: user
  password: pass
  discovery: false
  discovery_retain: false

remote_transmitter:
  pin: 
    number: D0
    inverted: false
  carrier_duty_percent: 50%
  id: "ir_dsoba_transmitter"

climate:
- platform: custom
  lambda: |-
    auto dsoba_ac = new GreeAC();
    App.register_component(dsoba_ac);
    return {dsoba_ac};
  climates:
    - name: "Dnevna Soba AC"
```

maretodoric, 2021-06-17:
FYI, if you have a backup, you may be able to downgrade firmware on it to an earlier version.
- Extract the backup
- Extract the addon backup (15ef4d2f_esphome.tar.gz)
- Go to data/<your_project_name>/.pioenvs/<your_project_name>
- Flash firmware.bin to your device

You can also attempt an OTA flash if access HA console directly or use  HassOS SSH port 22222 Configurator addon. In that case, first create `<your_project_name>/.pioenvs/<your_project_name>` directory in `/config/esphome/` folder (or if you're directly connected to the console, `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`) and move the firmware.bin there.

- Open esphome addon shell using `docker exec -it addon_15ef4d2f_esphome /bin/bash`, if addon_15ef4d2f_esphome is not the correct name, check the name using `docker container ls`
- Run `esphome upload --device <device_ip_address> /config/esphome/<your_project_name>.yaml`

It should attempt the OTA upload using the binary file from `/mnt/data/supervisor/homeassistant/esphome/<your_project_name>/.pioenvs/<your_project_name>`

closed 2021-06-17 by jesserockz

jesserockz, 2021-06-18:
Fixed in [v1.19.1](https://github.com/esphome/esphome/releases/tag/v1.19.1) which should be available in ~10 mins