esphome/issues #2141: With update to esphome v1.19 BLE Tracker on ESP32-S causes issue

baskapteijn, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HAS OS 6.0 VM

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 nodemcu-32s

**ESPHome version (latest production, beta, dev branch)**
v1.19.0

**Affected component:**
https://esphome.io/components/esp32_ble_tracker.html

**Description of problem:**
After updating from ESPHOME v1.18.0 to v1.19.0 and updating all my devices, the ESP32-S devices that use the BLE Tracker component fail to give log output over the air. They also no longer pass sensor data to MQTT.

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_tracker:
```

**Logs (if applicable):**

**Additional information and things you've tried:**
As soon as I remove the sensor and BLE Tracker component, flash them OTA (that's still possible), I get OTA log output again.

kordys, 2021-06-17:
Hi, the same applies to me.

rcmcronny, 2021-06-17:
Hi, for me the same, thank good, someone reported it already :)

thedutchrockstar, 2021-06-17:
This problem also occurs with me. I can see the device online in both my router and esphome environment. But it passes through no data. I also tried updating through USB (even though OTA seems to work) but this makes no difference.

fmeus, 2021-06-17:
I am experiencing the same issues (bye not working and the compiler messages). For now I have managed to get my esp32 (esp32doit-devkit-v1) working again by turning off the power_save_mode. 

```
wifi:
  ssid: !secret wifi_iot_ssid
  password: !secret wifi_iot_password
  # power_save_mode: none
  fast_connect: true
```

Maybe this workaround works for others as well.

fryc88, 2021-06-17:
^^ confirmed, it works again, but device is red (unavailable) in my esphome dashboard, but it sends BLE info again.

rcmcronny, 2021-06-17:
For me, commenting out the line
` power_save_mode: none`
also works ;)

playersct, 2021-06-17:
Yes, this works :)

baskapteijn, 2021-06-18:
This `power_save_mode: none` might indeed be related since I also use that key/value. Thanks for reporting guys.

thedutchrockstar, 2021-06-18:
Can also conform that commenting out the power_save_mode ensured that data was being passed through :)

Bloodpack, 2021-06-18:
confirmation
commenting out " power_save_mode: none " also works for me

AussiSG, 2021-06-18:
triple confirm :)

OttoWinter, 2021-06-18:
Likely related to https://github.com/esphome/esphome/pull/1789 - can someone test if the following works too?

```yaml
esphome:
  arduino_version: 1.0.4
```

thedutchrockstar, 2021-06-18:
esphome:
  arduino_version: 1.0.4

works!

fmeus, 2021-06-18:
I have just tried your suggestion (added arduino_version and set power_save_mode to none) on two of my esphome devices.
This seems to be working as well.

yvosound, 2021-06-19:
Same here, 

commenting out " power_save_mode: none " also works for me

baskapteijn, 2021-06-21:
Works:

esphome:
  arduino_version: 1.0.4

With ESPHOME v1.19.2.

OttoWinter, 2021-06-21:
Ok, thanks everybody for confirming it.

Now the next step: fixing it. So my question is: why do you put `power_save_mode: NONE` in your configs?

I know some time ago we recommended it for ESP8266s (for some specific HW/framework configurations it worked wonders), but I don't remember it helping for ESP32s too.

Can someone elaborate if `power_save_mode: NONE` is actually needed in some cases for ESP32s? The reason is: it makes sense that mode could somehow interfere with BLE coexistence.

agners, 2021-06-21:
I am trying to reproduce this on my end. However, when enabling `esp32_ble_tracker`, along with a `sensor` with platform `xiaomi_hhccjcy01`, I can see BLE devices found in the log, WiFi seems to work and I get sensor data from my HA system. I tested this with two different ESP32 devices.

What is the exact `esp32_ble_tracker` use case which causes problems? Is maybe a specific device required to reproduce it?

baskapteijn, 2021-06-21:
@OttoWinter I added the 'power_save_mode: NONE' to try to solve a problem where my ESP32s would no longer be reacheable after running about a day. While searching for clues I found someone hinting that this would might solve it. And in fact since then I did not see this problem occur anymore.

OttoWinter, 2021-06-21:
@agners Make sure you enable `power_save_mode: NONE` in the wifi section (that appears to be the other requirement in the comments here) - at least for me I can reproduce it like that. But haven't debugged the issue in detail.

@baskapteijn Thanks for the info! It might be that the new arduino version (`1.0.6`, the new default), fixes the other issue you saw. If possible, could you run your device with no `arduino_version` and without `power_save_mode` in your config and see if the 1 day issue still exists nowadays?

agners, 2021-06-21:
> @agners Make sure you enable `power_save_mode: NONE` in the wifi section (that appears to be the other requirement in the comments here) - at least for me I can reproduce it like that. But haven't debugged the issue in detail.

Oh I see, I somehow understood that part the other way around. Now with `NONE` in, I see the problem too. UART shows this stack trace:

```
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008f8ac: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4008fb29: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
WARNING Decoded 0x4011f00b: pm_set_sleep_type
WARNING Decoded 0x400f75f4: wifi_set_ps_process
WARNING Decoded 0x400f61ea: ieee80211_ioctl_process
WARNING Decoded 0x40121b3a: ppTask
WARNING Decoded 0x40090b9e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

agners, 2021-06-21:
It seems that no sleep mode in WiFi/Bluetooth coexist mode is really not supported anymore, there is a debug print in the log a bit further up:

```
[20:28:25][C][wifi:037]: Setting up WiFi...
[20:28:25][D][esp-idf:000]: E (1740) wifi:
[20:28:25][D][esp-idf:000]: Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!
```

That is also noted in the v3.3.5 ESP-IDF documentation (which Arduino 1.0.6 is based on)

> Disabling modem sleep entirely is not possible for Wi-Fi and Bluetooth coexist mode. 

https://docs.espressif.com/projects/esp-idf/en/v3.3.5/api-guides/wifi.html#station-sleep

However, this seems not to be the case in v3.3.4 and older ESP-IDF documentation (I think Arduino 1.0.4 is based on some v.3.3.3+ version)
https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-guides/wifi.html#station-sleep

baskapteijn, 2021-06-22:
`@baskapteijn Thanks for the info! It might be that the new arduino version (1.0.6, the new default), fixes the other issue you saw. If possible, could you run your device with no arduino_version and without power_save_mode in your config and see if the 1 day issue still exists nowadays?`

@OttoWinter I've done as requested. No news is good news. I'll let you know if the problem occurred again.

rcmcronny, 2021-06-22:
For me, my BLE Tracker works since 5 days as it should with the power_save_mode removed, so it really seems to be better in the new version. I have had the same issue, that after some time, the device was not available through wifi. That was mostly after 1-2 days. So looks good :)

OttoWinter, 2021-06-22:
Thanks @agners for debugging and spotting the breaking change! Ok then this is thankfully not a upstream regression :) Sorry I didn't spot it when reviewing the PR.

So to recap: `power_save_mode: NONE` (since framework `1.0.6` is incompatible with BLE coexistence, so we will make "BLE used together with power_save_mode: NONE" invalid.

`power_save_mode: NONE` _should_ no longer be required with the newer framework, as (hopefully) the issues that previously made the option necessary are fixed now.

agners, 2021-06-22:
> power_save_mode: NONE should no longer be required with the newer framework, as (hopefully) the issues that previously made the option necessary are fixed now.

I am using the default (which is `LIGHT`) on several ESP32 devices working since weeks stable on my network, so I agree.

What I did notice is that latency (e.g. when using ping) is higher with power save modes (with `NONE` latency is in the sub 10ms, with power supply is around 100ms). But I don't think it has a negative impact per-se, its just the way it works when a WiFi device gets into a lower power mode.

closed 2021-06-22 by OttoWinter

fi-sch, 2021-06-24:
I am using 'none' as I keep my ESP32 in a basement with really poor WiFi coverage. I installed it there over 1.5 year ago so I don't remember clearly, but I believe that changing power_save_mode from 'light' to 'none' helped me with maintaining the WiFi connection.

This breaking change (I mean 1.19) is really a pain for me, because for over a week now I don't have connection to this ESP and can't upload a new firmware with the fix. I don't need it at this moment so I procrastinated... but eventually I would need to disassemble my tightly-packed device, take out ESP32 and reflash it by plugging it into the mac.

baskapteijn, 2021-06-24:
For now you can just switch back to using Arduino 1.0.4 in your yaml file(s) even with the latest ESPHome version, from what I gather. This would solve it for a while at least.

fi-sch, 2021-06-24:
> For now you can just switch back to using Arduino 1.0.4 in your yaml file(s) even with the latest ESPHome version, from what I gather. This would solve it for a while at least.

Thanks! Will do as soon as I will be motivated enough to make it work again ;) I will try 'light' mode for a while, just to be sure that I indeed need 'none'. If so, then I will have to use Arduino 1.0.4. It's just a pity that I learned it the hard way.

agners, 2021-06-24:
> This breaking change (I mean 1.19)

Just to clarify, this is a breaking change we inherited from upstream Arduino SDK (which in turn came from a breaking change in Espressif IDF SDK). We can't really do anything about it other than staying with an old Arduino SDK version, which is not a viable solution long term.

fi-sch, 2021-06-24:
> Just to clarify, this is a breaking change we inherited from upstream Arduino SDK (which in turn came from a breaking change in Espressif IDF SDK). We can't really do anything about it other than [...]

Many thanks for explanation. Do you foresee any chance it will be "fixed" in Espressif IDF SDK anytime in the future, or should I take care of improving my WiFi coverage? :)

agners, 2021-06-24:
> Many thanks for explanation. Do you foresee any chance it will be "fixed" in Espressif IDF SDK anytime in the future, or should I take care of improving my WiFi coverage? :)

I don' think this is a bug, it seems a consequence of how Espressif designed the WiFi/Bluetooth co-existence now.

When you think about it, ESP32 uses a single antenna, and to do Bluetooth and WiFi on a single antenna you have to switch between the two modes (time division). Since you have to switch off WiFi to do Bluetooth every now and then, you might as well put WiFi into proper sleep.

I did dig a bit into the code: From what I can tell browsing through commits in v3.3.5, Espressif decoupled WiFi/Bluetooth code. While doing so, they removed the mode where WiFi modem sleep is none while using Bluetooth.

The documentation has then been updated in this commit:
https://github.com/espressif/esp-idf/commit/1e519370dd1c66acd09a5c0f783b395183805bc5

They also improved quite a bit in WiFi libs, e.g. see this list of improvements:
https://github.com/espressif/esp-idf/commit/ae246927851f30d1f437b0dd7f41110abc854657

Power saving mode must not directly lead to WiFi drop-out problems. It might be that what you have been seeing with Arduino 1.0.4 and light modem sleep was just a bug. Maybe it is fixed now, definitly worth a try since there are quite some improvements in the WiFi area.

Btw, I had a problem with coverage for one of my ESP32 as well, I solved it with a TTGO T1 which has a u.FL connector and a cheap flex antenna. I have a solid connection since then. Nothing beats a solid WiFi signal  :signal_strength:  :sweat_smile: