esphome/issues #2142: Setting Fan speed over MQTT

arctus, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.19.0

**Affected component:**

Speed FAN

**Description of problem:**
MQTT speed fan command only accepts preset values of: low, medium, high, that are converted to 33, 66 and 100 in the specified configuration below. MQTT speed fan state returns presets as well and not actual values. As I understand, the purpose of the change was to introduce ability to use actual speeds instead of presets, however it is impossible to set those using MQTT command (and get their value).

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: ledc
    pin: GPIO16
    id: gpio_16

fan:
  - platform: speed
    output: gpio_16
    name: "Recuperator"
    speed_count: 100

```

arctus, 2021-06-18:
Additionally what I found out is that speed is not restored on power failure. The Fan power is restored, however, speed is always restored to 0.00.

rcmcronny, 2021-06-18:
Hi,

my esphome speedfan device is on an older release because, i have to migrate that also, no time atm for this. 
But i found in this thread some info:
https://community.home-assistant.io/t/esphome-1-17-0-speed-fan-changes/305934/9

It seems you have to configure mqtt fan to percentage somehow, perhaps it helps you.

OttoWinter, 2021-06-18:
Correct, the fan integration for mqtt does not support speed levels yet. Pull Requests to add these are welcome!

SuperMasterPhoenix, 2021-09-06:
Unfortunately with the latest home assistant version 2021.9 the support for the legacy speed modes (low, medium, high) has been removed and only percentage values are now possible.
Am I correct that one cannot control a fan of a esphome device via MQTT anymore?

ssieb, 2022-08-02:
The mqtt fan speed problem has been fixed.

closed 2022-08-02 by ssieb