esphome/issues #2143: Custom Climate component is "Unavailable" in Home Assistant after upgrade to 1.19.0

maretodoric, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS, Raspberry Pi 3B+

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU

**ESPHome version (latest production, beta, dev branch)**

1.19.0

**Affected component:**

https://esphome.io/components/climate/custom.html

**Description of problem:**
Ever since the upgrade to 1.19.0 the entity is "Unavailable" in Home Assistant and Exception is thrown and displayed in home-assistant.log

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: mcu_dnevna_soba_transciever
  platform: ESP8266
  board: nodemcuv2
  includes: 
    - custom_components/gree/gree_ir.h
  libraries:
    - IRremoteESP8266

wifi:
  ssid: "SSID"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "MCU Dnevna Soba Transciever"
    password: "xxx"

captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"

mqtt:
  broker: 192.168.0.201
  username: user
  password: pass
  discovery: false
  discovery_retain: false

remote_transmitter:
  pin: 
    number: D0
    inverted: false
  carrier_duty_percent: 50%
  id: "ir_dsoba_transmitter"

climate:
- platform: custom
  lambda: |-
    auto dsoba_ac = new GreeAC();
[... 272 lines cut ...]
[15:27:47][C][mqtt.switch:038]: MQTT Switch 'Mi TV OK': 
[15:27:47][C][mqtt.switch:039]:   State Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_ok/state'
[15:27:47][C][mqtt.switch:039]:   Command Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_ok/command'
[15:27:47][C][mqtt.switch:038]: MQTT Switch 'Mi TV Back': 
[15:27:47][C][mqtt.switch:039]:   State Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_back/state'
[15:27:47][C][mqtt.switch:039]:   Command Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_back/command'
[15:27:47][C][mqtt.switch:038]: MQTT Switch 'Mi TV Home': 
[15:27:47][C][mqtt.switch:039]:   State Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_home/state'
[15:27:47][C][mqtt.switch:039]:   Command Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_home/command'
[15:27:47][C][mqtt.switch:038]: MQTT Switch 'Mi TV Vol. Up': 
[15:27:47][C][mqtt.switch:039]:   State Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_vol_up/state'
[15:27:47][C][mqtt.switch:039]:   Command Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_vol_up/command'
[15:27:47][C][mqtt.switch:038]: MQTT Switch 'Mi TV Vol. Down': 
[15:27:47][C][mqtt.switch:039]:   State Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_vol_down/state'
[15:27:47][C][mqtt.switch:039]:   Command Topic: 'mcu_dnevna_soba_transciever/switch/mi_tv_vol_down/command'
[15:27:47][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'ir_dsoba_button':
[15:27:47][C][mqtt.binary_sensor:019]:   State Topic: 'mcu_dnevna_soba_transciever/binary_sensor/ir_dsoba_button/state'
[15:27:47][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Mi TV Status':
[15:27:47][C][mqtt.binary_sensor:019]:   State Topic: 'mcu_dnevna_soba_transciever/binary_sensor/mi_tv_status/state'
[15:27:48][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Dnevna Soba Gree light':
[15:27:48][C][mqtt.binary_sensor:019]:   State Topic: 'mcu_dnevna_soba_transciever/binary_sensor/dnevna_soba_gree_light/state'
```

**Additional information and things you've tried:**
When i open climate entity from home assistant it's displayed as Unavailable. But when i click on "Operation" then select a mode, for example "Cool" or "Heat", AC will accept the command from ESP device and will respond, but home assistant entity is still displayed as unavailable.

maretodoric, 2021-06-17:
Seems like the same outcome as in:
#2140
#2139 
#2137

closed 2021-06-17 by jesserockz

jesserockz, 2021-06-18:
Fixed in [v1.19.1](https://github.com/esphome/esphome/releases/tag/v1.19.1) which should be available in ~10 mins

maretodoric, 2021-06-18:
Actually, it appears that it still doesn't work.
Just tried a clean build but i still get the the same error message

```
2021-06-18 09:02:41 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 214, in on_login
    entity_infos, services = await cli.list_entities_services()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 137, in list_entities_services
    entities.append(cls(**kwargs))
  File "<attrs generated init aioesphomeapi.model.ClimateInfo>", line 9, in __init__
    self.supported_modes = __attr_converter_supported_modes(supported_modes)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in _convert_climate_modes
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/model.py", line 245, in <listcomp>
    return [ClimateMode(val) for val in value]
  File "/usr/local/lib/python3.8/enum.py", line 339, in __call__
    return cls.__new__(cls, value)
  File "/usr/local/lib/python3.8/enum.py", line 662, in __new__
    raise ve_exc
ValueError: 6 is not a valid ClimateMode
```

jesserockz, 2021-06-18:
Ah yes sorry about not making that clearer for custom code.